pez2001/razer_chroma_drivers #2: installation process failed (something with SDL2)

specygamer, 2015-05-28:
I tried to install this driver (because I got the Blackwidow Chroma) and it seems that some SDL libraries are missing. When I execute the install_driver.sh file, I get this:

In file included from razer_chroma.c:1:0:
razer_chroma.h:17:22: fatal error: SDL2/SDL.h: Datei oder Verzeichnis nicht gefunden
 #include "SDL2/SDL.h"
                      ^
compilation terminated.
make[1]: *** [razer_chroma.o] Fehler 1
make: *** [librazer_chroma] Fehler 2
In file included from razer_chroma.c:1:0:
razer_chroma.h:17:22: fatal error: SDL2/SDL.h: Datei oder Verzeichnis nicht gefunden
 #include "SDL2/SDL.h"
                      ^
compilation terminated.
make[1]: *** [razer_chroma.o] Fehler 1
make: *** [librazer_chroma] Fehler 2

"Datei oder Verzeichnis nicht gefunden" is german for "File or Folder is not found"
"Fehler" is german for "Error"
(for all those guys that don't understand german ^^)

pablomenino, 2015-05-28:
Hi, what linux distribution are you using?, may be you have to install the SDL devel libraries.

On Fedora:
sudo dnf install SDL2 SDL2-devel

specygamer, 2015-05-29:
I'm using Linux Mint (so Ubuntu 14.04) :)

specygamer, 2015-05-29:
OK... I installed libsdl2-dev but if I run ./install_driver.sh, I get the same error :O

pablomenino, 2015-05-30:
On the web site for Arch Linux, the file SDL2/SDL.h is on the package sdl2
 https://www.archlinux.org/packages/extra/x86_64/sdl2/files/

Can you check if you have this file?
/usr/include/SDL2/SDL.h

If you don't have, try installing the package sdl2 (no dev package).

carstene1ns, 2015-05-30:
SDL2 is currently not used anywhere, so it is safe to remove the lines for now. If needed later, they should be system headers, not local ones (`<>` vs. `""`). Also the `Makefile` needs to be updated to link against SDL2 libraries then.

specygamer, 2015-05-30:
So and which Makefiles do I need to modify and who? I have never done something with Makefiles except runnig them. ^^

carstene1ns, 2015-05-30:
You can try removing the [four lines](https://github.com/pez2001/razer_blackwidow_chroma_driver/blob/master/lib/razer_chroma.h#L17-20) including sdl headers from ` razer_blackwidow_chroma_driver/lib/razer_chroma.h` and running `make` again.

pez2001, 2015-05-31:
i know the source is a bit messy right now.
i used sdl for early debugging.maybe some ui can be developed with it or ill just stick to dbus.
there is still a lot to do ... but at least im actively developing the driver again... check back for updates...
for now its safely to remove the sdl dependencies.if you dont want to wait for the next commit.

carstene1ns, 2015-05-31:
@pez2001: Your work is really appreciated!

specygamer, 2015-05-31:
I removed these lines but I still get errors that ld cannot find "-lSDL2_image"

RafTim, 2015-06-22:
I'm on a fresh arch-linux install. After installing relevant packages (sdl2, sdl2_gfx, sdl2_image, sdl2_mixer), I attemted to build the driver with the Install_driver.sh:

```=====================================================
::::::::::::::::\033[32m COMPILING librazer_chroma.so\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m COMPILING librazer_chroma.so [DEBUG]\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m COMPILING razer_bcd [SHARED]\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m COMPILING razer_bcd daemon\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m COMPILING razer_bcd daemon [DEBUG]\033[0m ::::::::::::::
=====================================================
In file included from razer_daemon.c:1:0:
razer_daemon.h:21:41: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_open)(struct razer_fx_render_node *render);
                                         ^
razer_daemon.h:21:41: warning: its scope is only this definition or declaration, which is probably not what you want
razer_daemon.h:22:42: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_close)(struct razer_fx_render_node *render);
                                          ^
razer_daemon.h:23:43: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_update)(struct razer_fx_render_node *render);
                                           ^
razer_daemon.h:24:46: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_key_event)(struct razer_fx_render_node *render);
                                              ^
razer_daemon.h:25:47: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_dbus_event)(struct razer_fx_render_node *render);
                                               ^
razer_daemon.h:27:58: warning: ‘struct razer_daemon’ declared inside parameter list
 typedef struct razer_effect *(*razer_effect_init)(struct razer_daemon *daemon);
                                                          ^
razer_daemon.c: In function ‘dbus_handle_messages’:
razer_daemon.c:85:11: warning: passing argument 1 of ‘printf’ from incompatible pointer type [-Wincompatible-pointer-types]
    printf(stderr, "Out Of Memory!\n");
           ^
In file included from razer_daemon.h:4:0,
                 from razer_daemon.c:1:
/usr/include/stdio.h:362:12: note: expected ‘const char * restrict’ but argument is of type ‘struct _IO_FILE *’
 extern int printf (const char *__restrict __format, ...);
            ^
razer_daemon.c:117:11: warning: passing argument 1 of ‘printf’ from incompatible pointer type [-Wincompatible-pointer-types]
    printf(stderr, "Out Of Memory!\n");
           ^
In file included from razer_daemon.h:4:0,
                 from razer_daemon.c:1:
/usr/include/stdio.h:362:12: note: expected ‘const char * restrict’ but argument is of type ‘struct _IO_FILE *’
 extern int printf (const char *__restrict __format, ...);
            ^
razer_daemon.c:167:11: warning: passing argument 1 of ‘printf’ from incompatible pointer type [-Wincompatible-pointer-types]
    printf(stderr, "Out Of Memory!\n");
           ^
In file included from razer_daemon.h:4:0,
                 from razer_daemon.c:1:
/usr/include/stdio.h:362:12: note: expected ‘const char * restrict’ but argument is of type ‘struct _IO_FILE *’
 extern int printf (const char *__restrict __format, ...);
            ^
razer_daemon.c: In function ‘daemon_run’:
razer_daemon.c:240:3: warning: implicit declaration of function ‘effect3’ [-Wimplicit-function-declaration]
   effect3(daemon);
   ^
razer_daemon.c:244:3: warning: implicit declaration of function ‘razer_frame_limiter’ [-Wimplicit-function-declaration]
   razer_frame_limiter(daemon->chroma,daemon->fps);
   ^
razer_daemon.c:245:17: warning: unused variable ‘end_ticks’ [-Wunused-variable]
[... 716 lines cut ...]
/tmp/razer_blackwidow_chroma_driver/driver/razerkbd.c:664:8: note: in expansion of macro ‘DEVICE_ATTR’
 static DEVICE_ATTR(macro_keys, S_IRUGO | S_IWUSR | S_IWGRP | S_IWOTH, razer_attr_read_macro_keys, razer_attr_write_macro_keys);
        ^
include/linux/bug.h:33:45: error: negative width in bit-field ‘<anonymous>’
 #define BUILD_BUG_ON_ZERO(e) (sizeof(struct { int:-!!(e); }))
                                             ^
include/linux/kernel.h:816:3: note: in expansion of macro ‘BUILD_BUG_ON_ZERO’
   BUILD_BUG_ON_ZERO((perms) & 2) +    \
   ^
include/linux/sysfs.h:77:12: note: in expansion of macro ‘VERIFY_OCTAL_PERMISSIONS’
    .mode = VERIFY_OCTAL_PERMISSIONS(_mode) },  \
            ^
include/linux/device.h:540:45: note: in expansion of macro ‘__ATTR’
  struct device_attribute dev_attr_##_name = __ATTR(_name, _mode, _show, _store)
                                             ^
/tmp/razer_blackwidow_chroma_driver/driver/razerkbd.c:665:8: note: in expansion of macro ‘DEVICE_ATTR’
 static DEVICE_ATTR(set_brightness, S_IRUGO | S_IWUSR | S_IWGRP | S_IWOTH, razer_attr_read_set_brightness, razer_attr_write_set_brightness);
        ^
scripts/Makefile.build:264: recipe for target '/tmp/razer_blackwidow_chroma_driver/driver/razerkbd.o' failed
make[2]: *** [/tmp/razer_blackwidow_chroma_driver/driver/razerkbd.o] Error 1
Makefile:1390: recipe for target '_module_/tmp/razer_blackwidow_chroma_driver/driver' failed
make[1]: *** [_module_/tmp/razer_blackwidow_chroma_driver/driver] Error 2
Makefile:6: recipe for target 'all' failed
make: *** [all] Error 2
```

kenimura, 2015-06-23:
I'm having almost the same output/error as RafTim on Fedora 22 x64. Here's the end of the output, I'll post the full output on request, I didn't because it's quite long as above....:
```
include/linux/bug.h:33:45: error: negative width in bit-field ‘<anonymous>’
 #define BUILD_BUG_ON_ZERO(e) (sizeof(struct { int:-!!(e); }))
                                             ^
include/linux/kernel.h:816:3: note: in expansion of macro ‘BUILD_BUG_ON_ZERO’
   BUILD_BUG_ON_ZERO((perms) & 2) +    \
   ^
include/linux/sysfs.h:77:12: note: in expansion of macro ‘VERIFY_OCTAL_PERMISSIONS’
    .mode = VERIFY_OCTAL_PERMISSIONS(_mode) },  \
            ^
include/linux/device.h:540:45: note: in expansion of macro ‘__ATTR’
  struct device_attribute dev_attr_##_name = __ATTR(_name, _mode, _show, _store)
                                             ^
/tmp/razer_blackwidow_chroma_driver-master/driver/razerkbd.c:665:8: note: in expansion of macro ‘DEVICE_ATTR’
 static DEVICE_ATTR(set_brightness, S_IRUGO | S_IWUSR | S_IWGRP | S_IWOTH, razer_attr_read_set_brightness, razer_attr_write_set_brightness);
        ^
scripts/Makefile.build:264: recipe for target '/tmp/razer_blackwidow_chroma_driver-master/driver/razerkbd.o' failed
make[2]: *** [/tmp/razer_blackwidow_chroma_driver-master/driver/razerkbd.o] Error 1
Makefile:1394: recipe for target '_module_/tmp/razer_blackwidow_chroma_driver-master/driver' failed
make[1]: *** [_module_/tmp/razer_blackwidow_chroma_driver-master/driver] Error 2
Makefile:6: recipe for target 'all' failed
```

carstene1ns, 2015-06-23:
@kenimura, @RafTim: Please checkout #3. I will try to resolve the other warnings later.

kenimura, 2015-06-24:
@carstene1ns  That compiles, and after manually signing the driver (UEFI system) and installing, it seems to work!

commit 02d13f67 (2015-06-27): Merge pull request #3 from carstene1ns/fix_driver_permissions

Fix #2.
commit 37c4b3d0 (2015-06-27): Do not make the sysfs files world-writeable.

This is needed to pass the strict permission checking for sysfs files.
References: https://lkml.org/lkml/2014/3/19/33
            https://lkml.org/lkml/2015/5/1/153

Fixes #2.
closed 2015-06-27 by pez2001