pez2001/razer_chroma_drivers #4: librazer_chroma.so: undefined reference to `atan2'

AngeloG, 2015-07-10:
I can't seem to get this to build on my machine (Ubuntu 15.04).

`razer_bcd` fails to build because it can't link against `atan2`. It's weird, because everything seems to be linking with `-lm`, so I'm kind of stumped.

```
:: COMPILING razer_bcd [SHARED]
================================
gcc razer_daemon.c  `pkg-config --cflags dbus-1` -DUSE_DBUS -Wl,-E -Wall -Wextra -W -w -Os -std=c99 -DMAJOR_VERSION=0 -DMINOR_VERSION=2 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 -lm -ldl `pkg-config --libs dbus-1`  librazer_chroma.so -o razer_bcd 
librazer_chroma.so: undefined reference to `atan2'
```

pez2001, 2015-07-11:
you could search for atan2 and remove the function ---> should work...

closed 2015-07-11 by pez2001

AngeloG, 2015-07-13:
What do you mean with "remove the function"?

It's used here: https://github.com/pez2001/razer_blackwidow_chroma_driver/blob/c2d0d08a9f4b649e8b8bf9b1332e64bd5e5a8f02/lib/razer_chroma.c#L1384

jordanking, 2015-07-18:
I had a few similar problems as the other issues, here are my fixes:
atan issue: move the linked library flags to the end of the compile lines in the make files
lsdl2 issue: sudo apt-get install libsdl2-dev and libsdl2-image-dev
cannot stat init.d/install_driver.sh: renamed install_driver.sh to install_driver (might be more elegant to fix script)

Using Ubuntu 14.04