pez2001/razer_chroma_drivers #7: Cant get this running on my linux box.

InfiniteDevelopment, 2015-07-19:
This is made for linux correct?

pez2001, 2015-07-19:
correct

pez2001, 2015-07-19:
and what is your problem if i may ask ?

kitsuneplusplus, 2015-07-22:
Hey! I'm having an issue getting this to run on my Ubuntu Mate installation. I'm pretty familiar with C, so I did like 2 hours of toying around with it, but was still unable to get it working. My problem is that on compilation, I receive an "undefined reference to atan2" error. My natural response was just in case to manually -lm all of the compilation lines, this presented more errors (which I unnecessarily fixed lol) which brought me back to the undefined atan2 reference. Any help would be much appreciated! I think the error throws on the debug files being compiled. Please help :3 I love pretty colors on my keyboard.

InfiniteDevelopment, 2015-07-26:
Same here, I am using the Ubuntu 14.04 distro, its not lighting up at all after following the build steps.

kitsuneplusplus, 2015-07-27:
hey InfiniteDevelopment, try downloading the newest update of the driver. Make sure to clean out any old stuff. Then, instead of running the original install script, run the new debian_install.sh script. It fixed it for me.

closed 2015-10-02 by pez2001