pez2001/razer_chroma_drivers #9: Can't seem to get this working in Linux mint

lanmalkieri, 2015-07-26:
Hey man, 

As I mentioned I cannot seem to figure out how to get this working. 

My keyboard is just showing some randomly colored purple keys in the middle of my keyboard. I have downloaded and ran your script and attempted to run a few of the driver scripts to get any affects working but none will do anything or make any chages to the keyboard. 

Any advice?

Thanks

pez2001, 2015-07-30:
i would at first check if everything compiled and if all files were properly installed.
check if kernel driver loaded / bound correctly and finally see if razer_bcd is running.
but to really help you i need outputs from commands you executed.

Mosie1, 2015-08-03:
Hello

i'm also having troubles in linux mint. I compiled and installed everything without errors. After restarting i tried using the scripts but there was no response. 

razer_bcd doesn't seem to be running.

pez2001, 2015-08-03:
in the source directory execute "sudo daemon/razer_bcd_debug_static" and post the output please.

Mosie1, 2015-08-04:
thank you for your reply. 
This is the output:

Starting razer blackwidow chroma daemon
opening chroma lib
error no compatible device found
razer_bcd: error initializing daemon

carstene1ns, 2015-08-04:
Can you give us the output of `lsusb -v -d 1532:` ? This should give us a list of razer usb devices.

pez2001, 2015-08-04:
it seems that the kernel driver isn't loaded/activated.
please try:
git pull
make -s clean all install
scripts/tests/activate_driver.sh
daemon/razer_bcd_debug_static

pez2001, 2015-08-04:
the driver needs access to sysfs to be able to bind itself.
check if you got a /sys directory in mint.

Mosie1, 2015-08-04:
Ok, don't know if it matters but compiling now gives me following warning 

>razer_daemon_dbus.c: In function ‘daemon_dbus_handle_messages’:
razer_daemon_dbus.c:1589:3: warning: string length ‘5061’ is greater than the length ‘4095’ ISO C99 compilers are required to support [-Woverlength-strings]
   "<!DOCTYPE node PUBLIC \"-//freedesktop//DTD D-BUS Object Introspection 1.0//EN\"\n\"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd\">\n\

executing the script scripts/tests/activate_driver.sh gives the following error  

>insmod: ERROR: could not insert module ../../driver/razerkbd.ko: File exists

And yes i have a /sys directory in mint

Mosie1, 2015-08-04:
I used the deactivate_driver.sh script before using the activate_driver.sh script and this time i didn't get the error. 

sudo daemon/razer_bcd_debug_static still gives:
>Starting razer blackwidow chroma daemon
opening chroma lib
error no compatible device found
razer_bcd: error initializing daemon

pez2001, 2015-08-04:
sudo scripts/tests/bind_driver.sh

Mosie1, 2015-08-04:
sudo scripts/tests/bind_driver.sh runs doesn't return anything.

same error from sudo daemon/razer_bcd_debug_static

Mosie1, 2015-08-04:
i've been looking into you scripts and the command ls /sys/bus/hid/devices/ | grep "1532:0203" returns 

>0003:1532:0203.0004
0003:1532:0203.0005
0003:1532:0203.0006

then ls /sys/bus/hid/devices/0003:1532:0203.0004 returns
>driver  hidraw  modalias  power  report_descriptor  subsystem  uevent

so none of these contain a folder "input" which seems to be needed for the next step

Don't know if this helps. I was just looking around.

Mosie1, 2015-08-04:
I got my keyboard to light up :D 

Problem is in the script where you execute the command "ls /sys/bus/hid/devices/$DEV/input". Since the input folder is missing the script just stops there. My guess was you use this to identify the right device? 

I managed to bind the driver and light up the keyboard by manually looking for the device's HID address (0003:1532:0203.0006) and just pasting it in the script.

pez2001, 2015-08-05:
im glad it works now for you :-) i incoporated it into the source.credits will be given.would be nice to know if it works out of the box after reboots.

Mosie1, 2015-08-05:
After rebooting the keyboard lights still work. 

Thank you for all your help!

closed 2015-08-05 by pez2001