pez2001/razer_chroma_drivers #14: makefile doesn't work with archlinux

elKokito, 2015-09-08:
`sudo make -s all install` return with `Makefile:7: recipe for target 'razer_kdb' failed`, any idea what is going wrong?

carstene1ns, 2015-09-09:
You need to provide more output, the real error is above in the build log.
A random guess: Did you install [`linux-headers`](https://www.archlinux.org/packages/core/x86_64/linux-headers/) package?

elKokito, 2015-09-09:
well I have now, error I get is
`chown: cannot access '/etc/init.d/razer_bcd': Not a directory`
and having a look at `/etc/init.d` it's a shell script, not a directory, and it's generated by
the makefile. do you need all the output from the `sudo make -S all install`?

carstene1ns, 2015-09-11:
Arch Linux does not use SysV init, so the directory does not exist. The Makefile now copies the init script to the non-existent directory and creates a file with the same name instead. If you want to use the Makefile, you need to create it yourself before.
However, note that you need to start the daemon yourself, even if the script exists.

Flennic, 2015-11-27:
Hello everybody,

I have exactly the same problem but due to missing knowledge I am unable to to solve the issue myself. It'd be great if there were a makefile which works with Arch Linux. Is there anyone around who is able to do that? If there is any way to help, I'd do that.

terrycain, 2015-11-27:
I've not forgotten about you guys. Currently debugging some usb stuffs, Have a distro based on arch in a vm so testing it is on my list of things to do.

Flennic, 2015-11-27:
Hi terrycain,

glad to hear that. Just got my Keyboard today :-)
If there is anything to help let me know!

terrycain, 2015-11-28:
See #37, has fix. Well it fixes Manjaro which seems to be an arch derivative, so try that.

closed 2016-01-13 by terrycain