pez2001/razer_chroma_drivers #15: Daemon cannot open file "mode_custom"

xaqq, 2015-09-11:
Hello,

Notes:
+ I own blackwidow chroma tournament edition.
+ I've replaced the `0x0203` product id by `0x0209` which corresponds to my keyboard.
+ I am familiar with C but not with kernel development.

After looking at the source and adding a few prints I managed to partially trace my problem.
It failed to open `/sys/bus/hid/devices/0003:1532:0209.0006/mode_custom`. This file simply doesn't exist.
I suppose the kernel module is supposed to create it so userland can talk to the driver.
The kernel module seems loaded: `lsmod  | grep razer` reports the following:

```
razerkbd               18592  0 
hid                   102264  3 hid_generic,razerkbd,usbhid
usbcore               195340  5 ehci_hcd,ehci_pci,razerkbd,usbhid,xhci_hcd
```

Do you have an idea about what could have wrong? Is there additional information that I could provide that would help resolving the issue?

Thanks,

pez2001, 2015-10-13:
did you modify the kernel driver too ? razerkbd.c & razerkbd.h

terrycain, 2016-04-05:
This should all work now?

closed 2016-05-20 by terrycain