pez2001/razer_chroma_drivers #16: write permissions to repo

jamesshawver1, 2015-09-13:
Is there any chance I could get write permission to your repo? I managed to create a profile from your driver that I think people would like to use. It essentially is a non-collision detecting version of the old game "Snake".

pez2001, 2015-10-02:
sorry for replying this late but i was on vacation.
i will look into your source (manual pull and reduce redundancies) but i will add your example and credit you for it, thats promised.
best greets and keep on the good work ;-)

tim

pez2001, 2015-10-05:
added snake, nice and funny addition, minor makefile polishing done. thanks again for your contribution, and i hope to see other stuff from you.

closed 2015-10-05 by pez2001

jamesshawver1, 2015-10-06:
not a problem :) thanks to you for doing all the heavy lifting