pez2001/razer_chroma_drivers #23: Razer DeathStalker

redaxmedia, 2015-11-04:
Hello,

can you give a roadmap about written driver for the Razer DeathStalker?
Is there a way to help coding porting the driver?

StephanieSunshine, 2015-11-05:
I doubt this tool will ever help you with your DeathStalker.  This is specifically for dealing with the chroma features of the BlackWidow.

redaxmedia, 2015-11-05:
Search for "Stretch Goal 2" inside the README.md

terrycain, 2015-11-28:
Its extremely hard to write a driver to a device you don't own, I assume that's why no progress has been made on it.

If you want to help you could try and reverse engineer the usb traffic. Not an easy task. At some point when I can be bothered I might write a page on how I did it for the Chroma and Ultimate 2013 keyboards (I assume the others that have done the same did it in the same or similar ways).

closed 2016-01-13 by terrycain

duanejeffers, 2016-01-21:
@terrycain I just got the Deathstalker Chroma keyboard, and I'm willing to help. I don't know how to reverse engineer USB traffic, but I'm willing to learn (hell, I'll probably google it when I'm done with this message). Let me know what you need from me, and I'll pass it on. :-)

terrycain, 2016-01-21:
@duanejeffers looking at the layout, in theory Razer should have been lazy and just copied the chroma firmware onto it so it *should* just work but I would rather be sure as between the 3 keyboards I've done there has been some subtle differences. I was debating on creating a wiki page on how to do it. I'll create a page on doing just that later. ;)

You'll need wireshark, sudo, virtualbox with a windows VM on it running that synapse program.

redaxmedia, 2016-01-21:
I would be happy to help out, just give me a wiki...

terrycain, 2016-01-21:
If I haven't made the Wiki page by Sunday, prod me again as I would have forgotten but I aim to do it this evening.