pez2001/razer_chroma_drivers #26: Missing /sys/bus/hid/drivers/razerkbd/bind

tnais, 2015-11-10:
First of all, thank you for the driver!

I noticed that the above file does not show up even if the kernel module seems loaded.

What could I have done wrong? I am using a 3.2.0-3-amd64 kernel.

terrycain, 2015-11-10:
What does running `lsmod | grep razerkbd` return?

Also running `ls /lib/modules/$(uname -r)/kernel/drivers/usb/misc | grep razerkbd` to ensure the driver is in the correct location?

tnais, 2015-11-10:
First of all thank you for the quick reply!

$lsmod | grep razerkbd
razerkbd               12986  0 
usbcore               128498  9 ehci_hcd,usbhid,uas,usb_storage,btusb,uvcvideo,razerkbd,ipheth

$ls /lib/modules/$(uname -r)/kernel/drivers/usb/misc | grep razerkbd
razerkbd.ko

They seem in the right place.

$cat /sys/module/razerkbd/initstate 
live

$cat /sys/module/razerkbd/refcnt 
0

terrycain, 2015-11-10:
That is most strange. What Linux distribution are you using?

tnais, 2015-11-10:
It is an AMD64 debian oldstable (the previous stable).
Kernei is a 3.2.0-3.

Are there messages in dmesg that I should search for?

Should I try to add some logging to the kernel module (I never did kernel coding, so forgive any possible blasphemy :) )?

terrycain, 2015-11-11:
You could try looking in `/var/log/syslog` and `/var/log/dmesg`.

I just installed Debian 8.2.0 onto a VM and installed the kernel driver and its worked successfully (ish). I doubt it, but it could be the driver is built for a different kernel version.

I'm not sure logging in the kernel module will be that useful, but it can be done using `printk`.

carstene1ns, 2015-11-11:
Does `lsusb` list the correct device Id?

```
$ lsusb | grep Razer
Bus 001 Device 003: ID 1532:0203 Razer USA, Ltd
```

tnais, 2015-11-11:
What do you mean With succesful(ish) :) terry win ? And what's the suggested kernel?
I will take a look to both files.

There are both the 1532:0203 an 1532:0202 IDs, possibly due the two sub connectors required to get the required current.

terrycain, 2015-11-11:
Ahh the daemon didn't start at boot but I could still bind the keyboard to the driver.

I am currently running the Ubuntu LTS on `3.13.0-66-generic`. The version shouldn't matter.

I have both USB's from my chroma plugged in and still don't get `1532:0202`, I just thought it was straight `usb` passthough to the keyboard `usb` port.

tnais, 2015-11-12:
Syslog
Nov 10 09:46:12 quigley kernel: [    3.860330] usb 2-1.5.2.4: Product: Razer BlackWidow Chroma
Nov 10 09:46:12 quigley kernel: [    3.860333] usb 2-1.5.2.4: Manufacturer: Razer
Nov 10 09:46:12 quigley kernel: [    3.861675] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.2/2-1.5.2.4/2-1.5.2.4:1.0/input/input2
Nov 10 09:46:12 quigley kernel: [    3.861846] generic-usb 0003:1532:0203.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma] on usb-0000:00:1d.0-1.5.2.4/input0
Nov 10 09:46:12 quigley kernel: [    3.864230] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.2/2-1.5.2.4/2-1.5.2.4:1.1/input/input3
Nov 10 09:46:12 quigley kernel: [    3.864401] generic-usb 0003:1532:0203.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma] on usb-0000:00:1d.0-1.5.2.4/input1
Nov 10 09:46:12 quigley kernel: [    3.866175] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.2/2-1.5.2.4/2-1.5.2.4:1.2/input/input4
Nov 10 09:46:12 quigley kernel: [    3.866396] generic-usb 0003:1532:0203.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer BlackWidow Chroma] on usb-0000:00:1d.0-1.5.2.4/input2

DMESG
[    3.860319] usb 2-1.5.2.4: New USB device found, idVendor=1532, idProduct=020
3
[    3.860325] usb 2-1.5.2.4: New USB device strings: Mfr=1, Product=2, SerialNu
mber=0
[    3.860330] usb 2-1.5.2.4: Product: Razer BlackWidow Chroma
[    3.860333] usb 2-1.5.2.4: Manufacturer: Razer
[    3.861675] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.2/2-1.5.2.4/2-1.5.2.4:1.0/input/input2
[    3.861846] generic-usb 0003:1532:0203.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma] on usb-0000:00:1d.0-1.5.2.4/input0
[    3.864230] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.2/2-1.5.2.4/2-1.5.2.4:1.1/input/input3
[    3.864401] generic-usb 0003:1532:0203.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow Chroma] on usb-0000:00:1d.0-1.5.2.4/input1
[    3.866175] input: Razer Razer BlackWidow Chroma as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.2/2-1.5.2.4/2-1.5.2.4:1.2/input/input4
[    3.866396] generic-usb 0003:1532:0203.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer BlackWidow Chroma] on usb-0000:00:1d.0-1.5.2.4/input2
[    3.954356] scsi 6:0:0:0: Direct-Access     Maxtor   Basics Desktop   0122 PQ: 0 ANSI: 4
[    3.955549] sd 6:0:0:0: Attached scsi generic sg2 type 0

tnais, 2015-11-12:
About the errata corrige:
I have a single entry in the output of lsusb.

I have the keyboard twice here

 # ls -l /sys/bus/hid/drivers/generic-usb/
total 0
lrwxrwxrwx 1 root root    0 Nov 12 16:56 0003:1532:0203.000F -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.2/2-1.5.2.4/2-1.5.2.4:1.0/0003:1532:0203.000F
lrwxrwxrwx 1 root root    0 Nov 12 16:56 0003:1532:0203.0010 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5.2/2-1.5.2.4/2-1.5.2.4:1.1/0003:1532:0203.0010
--w------- 1 root root 4096 Nov 12 16:56 bind
lrwxrwxrwx 1 root root    0 Nov 12 16:56 module -> ../../../../module/usbhid
--w------- 1 root root 4096 Nov 12 16:56 new_id
--w------- 1 root root 4096 Nov 12 16:56 uevent
--w------- 1 root root 4096 Nov 12 16:56 unbind

terrycain, 2015-11-12:
Running `ls -l /sys/bus/hid/drivers/generic-usb/`
Should list 3 files starting with `0003:1532:0203`

What does `ls -l /sys/bus/hid/drivers/hid-generic` output

tnais, 2015-11-12:
The latter command replies that hid-generic does not exist. 

BTW, I run just X, no DE.

terrycain, 2015-11-12:
Ahh sorry different naming from ubuntu -> debian. Shouldn't matter as dmesg said they are all there.

Not running a DE shouldn't matter as the driver still should have the bind.

Sounds stupid but have you tried completely removing the driver, rebooting, reinstalling then rebooting ?

tnais, 2015-11-12:
A good redet is nötig stupid. Is what goodjokes about IT people are made of. I will do the test tomorrow evening when back from an on the road shift.

pez2001, 2015-11-14:
maybe it was a bug with the input system that i fixed ?
pull and compile again please.

tnais, 2015-11-14:
I updated the coded and started having this error:

In file included from razer_chroma_controller.c:1:0:
razer_chroma_controller.h:35:38: error: unknown type name ‘DBusError’
make[1]: *** [razer_chroma_controller.o] Error 1
make[1]: Leaving directory `/home/saint/sorgenti/razer_blackwidow_chroma_driver/

It seems that on my machine dbus headers are under /usr/include/dbus1.0/dbus ... should I install some more thing?

terrycain, 2015-11-14:
The USE_DBUS macro in that file relies on `pkg-config --cflags dbus-1`. It will only include the DBUS headers if they've been found essentially.

Install `libdbus-1-dev` that should do it. You should be running DBUS if you want the daemon to do its thing. (`apt-get install -y libdbus-1-dev`).

If that doesn't fix it post the output of `pkg-config --cflags dbus-1`

tnais, 2015-11-15:
The libdbus-1-dev was already on my machine when I issued the command.

About pkg-config the output is the following:

> 16:31:28 [10] $pkg-config --cflags dbus-1
> -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include

According to the output of make all,

> 16:49:10 [25] $make
> make -C lib all
> make[1]: Entering directory `/home/saint/sorgenti/razer_blackwidow_chroma_driver/lib'
> ar -c -rs librazer_chroma.a razer_chroma.o list.o razer_string.o
> gcc -rdynamic -nostdlib -DCREATELIB -shared razer_chroma.o list.o razer_string.o -lm -o librazer_chroma.so
> gcc -Wall -Wextra -W -w -O0 -std=c99 -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 -c -o razer_chroma_controller.o razer_chroma_controller.c
> In file included from razer_chroma_controller.c:1:0:
> razer_chroma_controller.h:35:38: error: unknown type name ‘DBusError’
> make[1]: *** [razer_chroma_controller.o] Error 1
> make[1]: Leaving directory `/home/saint/sorgenti/razer_blackwidow_chroma_driver/lib'
> make: *** [librazer_chroma] Error 2

the makefile fails to identify the presence of the DBUS support and the USE_DBUS symbol remains unset.

Anyway, this is a snippet from razer_chroma_controller.h

```C
19    #ifdef USE_DBUS
20    	#include <dbus/dbus.h>
21    #endif
22
23
24    struct razer_daemon_controller
25    {
26    	int running;
27    	#ifdef USE_DBUS
28    		DBusConnection *dbus;
29    		DBusPendingCall *pending;
30    	#endif
31    };
32
33
34
35    int dc_dbus_error_check(char*message,DBusError *error);
```

This means that the include in line 20 is skipped and when the compiler arrives in line 25 the DBUSError symbol is undefined too. I think that this prevents the compilation without DBUS support.

I played it rough, i changed the Makefile in lib so that ALWAYS includes dbus support in both lib and examples and the compilation went on smoothly.

Nevertheless, despite

quigley:~# lsmod | grep raz
razerkbd               13021  0 
usbcore               128498  6 ehci_hcd,usbhid,uvcvideo,btusb,razerkbd

I don't see the named file, the daemon does not initialize and I can't use the leftmost keys (it's a long time since I last used a keyboard with function keys on the left).

tnais, 2015-11-15:
Update: I pulled the last update, once it compiled fine, then I had to bash a bit the makefile to force the
symbol definitions. But that's trivial.

I would like to understand how the thing works. Any pointer to some documentation?

pez2001, 2015-11-16:
at the moment the only documentation is in the readme.md.

tnais, 2015-11-17:
My apologies, my question was not correctly worded.

As far as I have understood (I have no kernel programming experience), the razerkbd.ko kernel module is handled by a kernel subsystem (I think the USB kernel subsystems) and this subsystem fails to "understand" that this kernel module should be readied for use by the hid. I would like to understand this subsystem workings to see if there were "hooks" to use to understand why it is not doing the right thing on my machine.

It seems to me that the module is correctly loaded (I listed it in /etc/modules after the loop module), but there's no /sys/bus/hid/drivers/razerkbd/bind directory. 

And I did not find any reference to that directory in the module source, I thought that the "failure" happened in the code that subsystem handling the razerkbd module.

terrycain, 2015-11-17:
There's not much I've found. Your best best is to just google about Linux kernel drivers.

tnais, 2015-11-17:
BTW, these are the udev rules in my machine for the razer black widow (what's the 0209 thing?) :

ACTION!="add|change", GOTO="razerkbd"

KERNEL=="razerkbd*", SUBSYSTEM=="razerkbd", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0203", TAG+="uaccess"
KERNEL=="razerkbd*", SUBSYSTEM=="razerkbd", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0209", TAG+="uaccess"

LABEL="razerkbd"

terrycain, 2015-11-17:
Pretty sure its the USB id's for the Razer Firefly

tnais, 2015-11-17:
How quaint. There is none hooked to my machine. Could wiping the udev rules help?

terrycain, 2015-11-17:
Nope shouldn't make a difference

tnais, 2015-11-17:
I see. 
Are there possibilities that we can sort my issue out?

tnais, 2015-11-17:
Another question. 

When is razer_kbd_probe()  function called?

pez2001, 2015-11-17:
probe is getting called when the device gets bound to the driver.because the hid driver grabs it first it is not being called when the device is plugged in.

tnais, 2015-11-19:
I updated to the latest stable (no kernel change) and strill the razerkbd driver is not recognized.
I got some documentation I will peruse in the near future (hopefully). Someone has any news?

tnais, 2015-11-19:
BTW, is there a way to prevent the generic hid driver to grab the keyboard? I use a laptop with an external keyboard because I hate lousy keyboards with the so called Italian layout: I want square brackets, braces and backquote!

terrycain, 2015-11-19:
udev supposedly should prevent hid from grabbing the razer keyboard but I've had no luck with that. I've seen some solutions that have a RUN clause in udev that echos the device id into hid-generic but haven't  tried that.

pez2001, 2015-11-20:
i guess you have to patch the kernel to let it skip certain ids, then
another driver would get a first hand chance to grab it. i tried something
like that in the beginning, but it would have also meant to implement the
whole low level keyboard class functions, which im now just passing letting
the class driver handle it. (saved a lot of redundant code, and also means
everybody using the keyboard can still benefit from mainstream kernel fixes)

tnais, 2015-11-23:
Terry, could you pass me some reference to the udev experiments?

Tim: how can it happen that in certain situations the kernel creates the /sys/bus/hid/driver inodes for the razer keyboard and sometimes not? I noticed that the activation script disables the generic USB binding and then enables the razer specific one therefore I suppose that, when all goes fine, you have the generic module active and the razer specific idle, both ready for the swap. I hove to have time to read my new book and understand this.

And I agree with your choice of not replicating the code and let the mainstream kernel fixes "slip in" for free.

tnais, 2015-11-26:
I tried to use this udev rule, but it did non work (it disabless the generic driver binding, but the inode for the razer does not show up.

==============================
SUBSYSTEM=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0203", MODE="0660", GROUP="plugdev"
ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0203", DRIVER="usbhid", RUN="/bin/sh -c 'echo -n $kernel >/sys/bus/usb/drivers/unbind'"
==============================

tnais, 2015-11-30:
Any news? A pointer to the udev example?

terrycain, 2015-11-30:
No news. The udev example you did was similar to my tests. The support scripts were created to bind and unbind from hid-generic as it seemed udev didnt want to work.

tnais, 2015-11-30:
Yep. The cirectory still does not appear...

terrycain, 2015-11-30:
Run `ls -la /sys/bus/hid/drivers/razerkbd`

closed 2016-01-13 by terrycain

tnais, 2016-01-14:
Sorry, but closing this ticket is inappropriate since no solution has been obtained.

tnais, 2016-01-14:
Furthermore, if I say that such directory does not appear, what should ls return (regardless of the options) other than an error?

terrycain, 2016-01-14:
The above ls should output something like
```
$ ls -la /sys/bus/hid/drivers/razerkbd
total 0
drwxr-xr-x 2 root root    0 Jan 12 12:57 ./
drwxr-xr-x 5 root root    0 Jan 11 20:09 ../
--w------- 1 root root 4.0K Jan 13 10:55 bind
lrwxrwxrwx 1 root root    0 Jan 12 12:57 module -> ../../../../module/razerkbd/
--w------- 1 root root 4.0K Jan 12 12:57 new_id
--w------- 1 root root 4.0K Jan 12 12:57 uevent
--w------- 1 root root 4.0K Jan 12 22:23 unbind
```

Have you tried uninstalling and then using ./package_for_debian and installing the deb?

tnais, 2016-05-28:
problem solved after a system upgrade. It was the former system to be broken

terrycain, 2016-05-28:
Yeah I thought as much, was still an extremely strange issue though.