pez2001/razer_chroma_drivers #28: Daemon documentation

terrycain, 2015-11-11:
Could you write some detailed documentation of the daemon and how to interact with it (specifically using DBUS). Sending messages via DBUS is easy, it would help to know what to send the daemon.

I would like to start writing a GUI, the node editor in blender shouldn't be difficult with a bit of OpenGL and GTK. From what I can gather there are render nodes and they are stored in a list and are updated in a loop.

Am I right in thinking that if one wanted to set the keyboard to static colour mode they would skip the daemon and tell the driver to do it? (From what I can see the daemon should be more than capable to do static colour)

pez2001, 2015-11-14:
hi terry,
i added some c api example (examples/controller_example)
also i added some pointers to the readme

the rendering system is inspired by the blender node editor to some extent
(nodes have two possible inputs and one output and you can add sub nodes for various computations/event handler/etc [a render chain that ends at the 'framebuffer'])
i tried to make the whole thing as simple as possible without sacrificing vast configurability options

greets
tim

pez2001, 2015-11-14:
and yes you are right , there is no static color effect yet included in the daemon ,though it should be trivial to add

terrycain, 2015-11-14:
Cheers. I'm just working on packaging up the driver etc... for Debian then I'll have a read through it.
Also added some simple options to the daemon like writing a PID file just to help with the boot scripts.

I got the daemon_controller to spit out a ton of JSON so will look into this too, thanks for the help.
Once I've got this packaging done I'll throw some effort into a GUI

pez2001, 2015-11-14:
thats great , and again thanks for all your contributions!

terrycain, 2015-11-14:
No problem. Its about time Linux got decent razer support, good job on the driver and daemon.

Looking at controller_example.c am I right in thinking it looks something like this

```
wave_effect_node─────────────┬────>mixer_node──────>framebuffer
light_blast_effect_node──────┘          │
                                        │
                                        V
                                    mouse_mixer
```

Just wondering was there any reason you chose to write the daemon & controller in C and not C++? 
I'm a big object-orientated fan is all and thought it might have been easier to do in C++ but correct me if I'm wrong.

terrycain, 2015-11-14:
Also I had some issues with the path in controller_example. I think passing the absolute path of the library to the daemon would be a better idea as the daemon will use the path relative to its working directory, which may or may not be the root of the git checkout.

closed 2015-11-17 by terrycain