pez2001/razer_chroma_drivers #37: can't install using makefile in manjaro kde

silentbirdie, 2015-11-19:
her is the output of the comand:

::\033[32m COMPILING librazer_chroma.so\033[0m
================================
::\033[32m COMPILING librazer_chroma.so [DEBUG]\033[0m
========================================
::\033[32m COMPILING razer_bcd [SHARED]\033[0m
================================
::\033[32m COMPILING razer_bcd daemon\033[0m
==============================
::\033[32m COMPILING razer_bcd daemon [DEBUG]\033[0m
======================================
/usr/bin/ld: cannot find -lrazer_chroma_debug
collect2: error: ld returned 1 exit status
Makefile:34: recipe for target 'daemon_debug' failed
make[1]: *** [daemon_debug] Error 1
Makefile:26: recipe for target 'razer_daemon' failed
make: *** [razer_daemon] Error 2

terrycain, 2015-11-20:
Haven't tried compiling it on Fedora but I'm sure some alterations will be needed. Give me a day or two to whip up a test vm and ill look into it

silentbirdie, 2015-11-20:
Thanks! It's the kde edition of manjaro

BenoitMartin2, 2015-11-27:
Hi, same to me. I can't install it on Manjaro. Thanks for your kind help mate!
Even tried this => http://pollonamid.de/index.php/en/my-devices/computer/90-how-to-get-the-razer-blackwidow-chroma-backlight-to-work-with-linux
But couldn't make it.

pablomenino, 2015-11-27:
I have the same problem on Fedora 23.

gcc -L../lib -rdynamic razer_daemon.do razer_daemon_dbus.do razer_daemon_effects.do razer_daemon_libraries.do razer_daemon_parameters.do razer_daemon_render_nodes.do razer_daemon_types.do -lrazer_chroma_debug -lm -ldl -o razer_bcd_debug_shared 

/usr/bin/ld: cannot find -lrazer_chroma_debug

[log.txt](https://github.com/pez2001/razer_blackwidow_chroma_driver/files/46158/log.txt)

terrycain, 2015-11-28:
There is a issue with the command in the lib/Makefile (and the others that use LIBDBUS) which doesn't set the LIBDBUS flag. Ensure you have dbus headers installed. Also you'll also need the kernel-headers installed for the kernel your running (aka output of `uname -a`)

I have a pull request which fixes the command which should then work for arch. This commit terrycain/razer_blackwidow_chroma_driver@32f607a contains the differences you need to get the Makefiles to work.

terrycain, 2015-11-28:
BenoitMartin2. Dont post massive errors in pull requests, its just annoying.

What files are in the lib folder after you typed make

BenoitMartin2, 2015-11-28:
Next time I'll use a paste, sry.
Here are the files in /lib => http://imgur.com/z23g32x

terrycain, 2015-11-28:
Thanks. cd into the daemon_controller and run `make`, what does that output + what files are there in there

BenoitMartin2, 2015-11-28:
Same :/   
Complete output: http://pastebin.archlinux.fr/1746199

Flennic, 2015-11-28:
Hi folks,

I did the changes to the makefiles but I still get the same problem:

http://pastebin.com/nq61YKj5
chown: cannot access ‘/etc/init.d/razer_bcd’: Not a directory

I am not sure if it'd help if I create that directory before doing make?

terrycain can you give a small introduction how to start the driver and how to echo stuff to it (if it finishes compiling one day :-)).

Thanks for all your work!

terrycain, 2015-11-28:
@BenoitMartin2 make sure you have all the libraries installed as @Fennic said. Otherwise im not sure whats wrong.

@Flennic make install wont work on arch, its designed for debian. There is a daemon start script in `install_files/init.d/razer_bcd_debian`. It wont work out of the box but what it essentially does is unbinds the chroma from the generic-hid driver and binds it to our driver and then starts the daemon.

Flennic, 2015-11-28:
Just do be curious: I executed ./activate_driver.sh and aht I got is this:

Unbinding device (0003:1532:0203.0003) from hid-generic
Binding device (0003:1532:0203.0003) to razerkbd
Bind Successful

Which looks good in my eyes. If the deamon is running now, is there a way to to send commands to it like make all keys illuminated in blue or smth. like this?

terrycain, 2015-11-28:
Unless my script is giving you false hope looks like the kernel driver is loaded.

Whats the output of `ls /sys/bus/hid/drivers/razerkbd/`?

Best make sure the driver is actually binded before trying other things

Flennic, 2015-11-28:
Output is:
ls /sys/bus/hid/drivers/razerkbd/
0003:1532:0203.0003  bind  module  new_id  uevent  unbind

How to check if the driver is actually binded?

Hell I feel like a 4 years old kid asking all those questions but I have no idea how drivers work or how close/far I am from getting it to run. I also found a folder systemd with a razer_bcd.service file. Is it possilbe if I copy all needed files manually to the correct location that it could work?

terrycain, 2015-11-28:
Run `ls /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003/`
The fact that `0003:1532:0203.0003` is there means its binded but knowing the contents makes it easier to explain

That's fine it takes a while to tame the linux kernel and its many drivers.
Yes, essentially if you copy all the required files to the specific places then things will work. If you give me a few days Im starting to populate the wiki to document all of this. (And reduce the question ;) ).

The daemon still needs a lot of work but you can manually operate the driver (once again ill make a wiki page to say what data the device driver files accept).

Flennic, 2015-11-28:
ls /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003/
country  input       mode_breath  mode_reactive  mode_wave          reset           subsystem
driver   macro_keys  mode_custom  mode_spectrum  power              set_brightness  temp_clear_row
hidraw   modalias    mode_none    mode_static    report_descriptor  set_key_row     uevent

The Wiki idea sounds really good. I really appreciate all of your hard work, sadly I can't help you that much. If I get more into this after the Wiki is up, maybe I can do some work. We will see. Just letting you know that I am really thankful.

Another question is: The deamon that already exists, does it work the same way in ubuntu and arch or are there differences? Because you said it still needs a lot of work? I guess just one color-backlight would be awesome too :-)

Take your time to do this and if there is paperwork to do (Wiki or somethink were your could need help), just let me know!

terrycain, 2015-11-28:
As sudo run this
```
echo "1" >  /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003/mode_wave
```
"should" set the keyboard into wave mode.

The wiki should be good once I've done it. The daemon works well its just not very flexible at the moment. The only issue is the differences in the layout of the filesystems between ubuntu and arch, if everything is where the daemon expect it to be it should run.

Flennic, 2015-11-28:
sudo echo "1" >  /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003/mode_wave
bash: /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003/mode_wave: Permission denied

No idea why is that.

One last question: I can see the ubuntu location for all of the files in the make make. Do you have any idea where the files for systemd go? Like /etc/init.d/racer_bcd where I have to put that? Becasue as far as I can read the makefile is works well until the point where it's trying to create /etc/init.d/ files and symlinks.

If we get to the point where everything is working as intended (yeah I know :D), is it hard to make a pacman packege for the AUR, do you have any idea how much work that is? But anyways, that's not on the list right now. But if I am bored, I am willing to try it. Could help a lot of people.

terrycain, 2015-11-28:
Run it as root
sudo su
then the echo thing.

Once I get the wiki up and running ill document everything, then feel free to make an AUR package and update the documentation to getting it working on arch

Flennic, 2015-11-28:
```
[root@pc razer_blackwidow_chroma_driver]# echo "1" >  /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003/mode_wave
```

Works now, but nothing's happening :/

terrycain, 2015-11-29:
What about
```
echo "1" >  /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003/mode_spectrum
```

Flennic, 2015-11-29:
```
echo "1" >  /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003/mode_spectrum
```
Same behaviour. Works but nothing happens.

terrycain, 2015-11-29:
Yeh the mode_static takes hexadecimal. \xFF\xFF\x0 should be yellow.

I'm going to write a page about the driver files. Then about what goes where on the filesystem and then about the daemon.

It "migh" just work if you run `GIT_CHECKOUT/daemon/razer_bcd -fv` `GIT_CHECKOUT` being the path to the folder you cloned. If it runs and doesn't exit straight away, start typing into a text editor and see if the daemon effect is working.

Flennic, 2015-11-29:
Hey there, hexadecimal works.

But therefore I have another strange thing: When I start my computer and booting into arch, the keyboard mode is set so spectrum. So somehow the driver is started instantly, even before decrypting my harddrive.
After booting I have to active the driver, so that the 0003:... file appears and after that I can sent instructions to the keyboard again. But I can't "disable" colors. What worked is setting static mode, but I was unable to stop the spectrum mode.

How can the keybord be illuminated even before the OS booted? I thought there is no daemon running? :-)

dubbleohnexus, 2015-11-29:
The keyboard illumination is at a hardware level, it'll turn on the
backlights as soon as its given power. Evidence of this is if you were to
use the keyboard in your bios, it would light up.

Flennic, 2015-11-29:
Okay, seems like I do not fully understand how this works. After my PC starts, which part sends the data to the keyboard? You say it's on hardware level, so which hardware is doing that? As far as I know the keyboard itself is stupid (unlike Logitech's keyboards) and is not able to save any inforamtion on itself.

terrycain, 2015-11-29:
No the keyboard is not stupid. There are a few effects that the keyboard can perform itself, the driver only allows us to tell the keyboard "do this effect".

The keyboard knows how to do spectrum,wave,reactive,static,none and breathing effects

dubbleohnexus, 2015-11-29:
I'm not real sure if the keyboard can save what backlighting mode to run,
but the LEDs will turn on as soon as they're given power. The
microcontroller in the keyboard must have a small amount of onboard memory
(either flash or RAM) to store backlighting profiles, otherwise you
wouldn't be able to use them at all.

Flennic, 2015-11-29:
Thanks for all the information. I configured the keyboard on Mac OS X before and when I plugged it into my Linux machine it didn't show any light effects. Therefore I thought the wrong way.

If you plan to develop an opensource driver, how do you know how to communicate with the keyboard? It's not like Razer is telling us how. But anyways, thats not part of this issue.

terrycain, 2015-11-29:
Back before razer went all cloud sync, there used to be ways to store stuff on the keyboard. The keyboard can probably still do that but to find that out I'd need to find an ancient version of the razer tool.

As for how we communicate with the keyboard, we find out what makes it tick by reverse engineering the USB traffic. Then using that knowledge we send stuff to the keyboard.

BenoitMartin2, 2015-11-29:
Hi,

I'm still stucked. Have all libraries mentionned by @Flennic. BTW, as told earlier, already tried the "Arch" package from pollonamid.
http://pastebin.archlinux.fr/1752329 => ln: impossible de créer le lien symbolique « /etc/rc2.d/S24razer_bcd »: Aucun fichier ou dossier de ce type

terrycain, 2015-11-29:
@BenoitMartin2  run `ls -lR` inside the repository and put it on pastebin

BenoitMartin2, 2015-11-29:
Sorry but which one do you want I ls?

terrycain, 2015-11-29:
which ever folder you ran install-driver.sh in

BenoitMartin2, 2015-11-29:
Here they are:
 - Pez2001: http://pastebin.archlinux.fr/1752340
 - Pollonamid: http://pastebin.archlinux.fr/1752367

terrycain, 2015-11-29:
We dont support anything in the Pollonamid repo, thats the job for whoever forked that (but from looking at it, it compiled sucessfully).

As for the Pez2001 repo, run `sudo make clean` in that and then give me the output of `make`

BenoitMartin2, 2015-11-29:
Okay. Here's the make output: http://pastebin.archlinux.fr/1752368

terrycain, 2015-11-29:
That looked like its working. Output of `ls -lR`

BenoitMartin2, 2015-11-29:
http://pastebin.archlinux.fr/1752389

terrycain, 2015-11-29:
Yeah its compiled fine. The `make install` most likely wont work as its not designed for arch. Look at the wiki for this repository, and google, you need to move the razerkbd.ko to the appropriate place to "install" the driver.

Flennic, 2015-11-29:
Just to clarify things: The makefile didn't work for me. What I did was:
From the downloaded git repo:
* cd install_files/init.d/
* sudo ./activate_driver.sh
* ls /sys/bus/hid/drivers/razerkbd/
* Look for a device name like 0003:1532:0203.0003
* cd into that folder
* there you will find various files. Look inside the wiki what you have to echo in which file to use the driver
* keep in mind, sudo does not work here, you have to switch to root with sudo su

BenoitMartin2, 2015-11-29:
Thank you @terrycain, it seems it's okay : http://pastebin.archlinux.fr/1752410
I'm going on with hid-generic

terrycain, 2015-11-29:
@Flennic to clarify, the makefile didn't install things but it did compile the code as otherwise nothing would of worked ;)

@BenoitMartin2 Cool

Flennic, 2015-11-29:
Oh yes I thought thats for sure. Sorry I didn't mention :-)

BenoitMartin2, 2015-11-29:
It's working!  Thanks a lot :)  Now... I just need to understand how to deal with it and bind it a better way. The wave's gonna drive me mad :/

@terrycain Do you have a BTC wallet or else?

terrycain, 2015-11-29:
@BenoitMartin2 Agreed, the wave looks cool until you actually want to use the keyboard ;)
I have paypal (terry@terrys-home.co.uk) but never really got into bitcoins.

Once you have it binded there is another page on how to change colours (the static works but some things are waiting on some of my changed to get merged)

Flennic, 2015-11-29:
Another issue is that the numblock icon looks diffuse. Any idea why is that or how to fix that?

terrycain, 2015-11-29:
@Flennic you mean its a little blurry??

Flennic, 2015-11-29:
Yes :-)

BenoitMartin2, 2015-11-29:
Have a beer (or more) then @terrycain ;)  I read your Wiki, I'll now have to configure my keyboard. Thanks again gyuz!

terrycain, 2015-11-29:
@Flennic That's just the way the case is made. Its the same on my Ultimate 2013 too, their new Ultimate 2016 [Link](http://www.razerzone.com/gb-en/store/razer-blackwidow-ultimate-2016) has changed that I think. My mates ordered that so it shall be supported soon ;)

@BenoitMartin2 Cheers that will go towards beer and towards my razer mamba which shall be added as soon as it arrives ;)

BenoitMartin2, 2015-11-29:
Do you know if there's anyway to get their Windows config tool work with Linux? Cause despite your work/help made me able to use my keyboard a bit better, it's still really not easy to change colors as we code or play X/Y game.

terrycain, 2015-11-29:
@BenoitMartin2 the windows tool wont work, you could run it in wine but it still wouldn't know how to use the driver. If you can get the daemon working there is a simple python app which lets you do a few things.

There are a few scripts to use the driver etc... Its a work in progress but its getting there

Flennic, 2015-11-29:
If there is a way to save things on teh keyboard itself, that would be great. Then you could configure it using a Windows or Mac OS X Machine and use it with windows. That's what I was doing with my previous keyboards from Logitech. But I don't have that much hope with this Razer keyboard. Still letting in glow in one color of my choice is already okay :-)

BenoitMartin2, 2015-11-29:
Wall I remember on my AlienWare laptop I could configure the lights using Windows before having it to Linux. And the keyboard remains colored. This might be possible with this Razer too. I'll git it a try.

terrycain, 2015-11-29:
I might look into getting the razer cloud sync to work. The keyboards used to have profiles but not sure if that'll work anymore.

The keyboard "should" remember the last setting that the razer synapse tool set but im not sure if it realises this once its powered off and on

dubbleohnexus, 2015-11-29:
After unplugging and replugging my keyboard it remembers the last setting
it was on. I dunno about rebooting the computer though, I'll have to try
that later.

terrycain, 2015-11-29:
@Flennic Cheers, more beer for me ;)

silentbirdie, 2015-12-01:
can someone explain to me how to install it as this is all going over my head and I do not understand it.

dubbleohnexus, 2015-12-01:
I thought the wiki did a pretty decent job of explaining it. Pretty much
just c/p the commands from there and do what it says and it'll work.

[37 more replies]