pez2001/razer_chroma_drivers #39: Some modes not working on BlachWidow Chroma Tournament Edition

dubbleohnexus, 2015-11-29:
I was very excited to get this working, but to my (non)surprise several modes did not work while others worked perfectly.

# Modes that worked
* Mode_game
* Mode_none
* Mode_spectrum
* Mode_wave

# Modes that did not work
* Mode_breath
* Mode_custom (just went blank, not sure if theres something else i had to do)
* Mode_reactive
* Mode_static

The modes that did not work seem to all be ones that might require some kind of extra configuration, is it possible that I just missed something in my reading? How could I test if these really do not work or if I'm just not getting something?

OS: Xubuntu 3.19.0-33-generic
Keyboard: Razer Blackwidow Chroma Tournament Edition

dubbleohnexus, 2015-11-29:
After messing around with the wiki more, I found that reactive mode does work, while breathing mode still does not. I havent touched custom mode or static mode again, but i will report back shortly

terrycain, 2015-11-29:
@dubbleohnexus Breathing mode needs to be patched, am waiting on a pull request.

lah7, 2015-12-29:
Can confirm **Breath Mode** isn't working on the Razer BlackWidow Chroma (`RZ03-0122`)

terrycain, 2016-01-13:
Breath mode now works. Closed

closed 2016-01-13 by terrycain