pez2001/razer_chroma_drivers #41: I'm getting married soon!

pez2001, 2015-12-17:
just to let you know why im not working on this right now. I will continue work in january, atm i just dont have time for it. sorry. 
i will try to merge new pull requests though.
best wishes and and a happy holiday + new year

Sclarki, 2016-01-09:
Congratulations! I wish you both the best in your time together. It's a very special time in your life, you should take the time to celebrate it and enjoy it :3

Also, I would like to add: I bought this keyboard earlier this evening and the only motivating factor to actually bother getting it (for use on a *nix system) was knowing your repo was here. For that, I (and I'm sure everyone else) am truly grateful. I don't know how many other projects exist that support keyboards of this nature for UNIX-like systems, but I'm glad they do exist. Thank you!

pez2001, 2016-01-12:
Thank you very much , it was the happiest moment of my life ;-) an awesome experience.

closed 2016-01-12 by pez2001