pez2001/razer_chroma_drivers #42: Can't install on Fedora 23

Pant, 2015-12-18:
I get this error when I am trying to run command as described in the readme file for non-debian distros: 

    make -s all install
The [output](http://pastebin.com/Wc0JNa8b) seems like [this](https://github.com/pez2001/razer_blackwidow_chroma_driver/issues/37) issue but i didn't find any answers.
Is it possible to install this driver on Fedora?
Thanks in advance.

terrycain, 2015-12-18:
Does [this](https://github.com/pez2001/razer_blackwidow_chroma_driver/issues/37#issuecomment-160288381) help

Pant, 2015-12-18:
No package dbus-headers exist. I have dbus-devel, dbus-glib-devel, kernel-headers, and kernel-devel installed.

Pant, 2015-12-18:
Could you give me a list with ALL the required packages needed?

terrycain, 2016-01-13:
Run `make clean`
Then pastebin the output of `make` (not `make install`)

pablomenino, 2016-01-15:
Hi, I downloaded the source code today and compile under Fedora 23 whit no problems.
I go to try to manually install.
The installation script is not intended for this distribution?

There is any info for installing on Fedora?

terrycain, 2016-01-15:
@pablomenino can you give me a list of packages you had to install?

The install script isn't meant for fedora but as I have some free time, I'll make a fedora VM and create a script to create a .rpm for you. Having a list of packages would speed this up as the names are different to Debian/Ubuntu.

pablomenino, 2016-01-15:
Hi Terry, I see that the Makefile only copy some files but not all, I go to try manually but if you can test to create a RPM Package that go to be perfect, let me now if you need to test something.

**Command to install packages:**
sudo dnf install jq jq-devel dbus-devel SDL2 SDL2-devel SDL2_image SDL2_image-devel fftw-libs-double fftw-libs-single

**Packages information:**

Package | Current version
------------ | -------------
jq.x86_64 | 1.5-1.fc23
jq-devel.x86_64 | 1.5-1.fc23
dbus-devel.x86_64 | 1:1.10.6-1.fc23
SDL2.x86_64 | 2.0.3-7.fc23
SDL2-devel.x86_64 | 2.0.3-7.fc23
SDL2_image.x86_64 | 2.0.0-8.fc23
SDL2_image-devel.x86_64 | 2.0.0-8.fc23
fftw-libs-double.x86_64 | 3.3.4-6.fc23
fftw-libs-single.x86_64 | 3.3.4-6.fc23

terrycain, 2016-01-15:
@pablomenino am attempting to get fc13 in a VM, once I've done that I'll get on creating a an RPM and then a script to make the RPM. Ideally I want to get it working with either akmod or dkms as then the system will recompile the driver when you update the kernel.

Pant, 2016-01-16:
I still cannot install under fedora 23 64 bit and kernel-4.2.8-300.fc23.x86_64. I tried to install the packages that @pablomenino said he installed but nothing changed.
Here is the output of make (after make clean).
http://pastebin.com/p7xdBKSD
Thanks for your help. I am here if you need me to test something else.

pablomenino, 2016-01-16:
@terrycain , yes that go to be great.

Hi @Pant, ensure that you have the gcc packages and the kernel-devel package for your kernel version (In my case the kernel is 4.3.3-300.fc23). I test this whit a new fresh install of Fedora 23 (installed from Fedora-Live-Workstation) and works fine.

**GCC Packages:**
sudo dnf install gcc gcc-c++ gcc-objc make

**Kernel Devel**
sudo dnf install kernel-devel

I don't see the version 4.2.8 for the package kernel-devel on the fedora repository, may be you can find it on the package Datadase https://admin.fedoraproject.org/pkgdb/

[pablo.menino@NORC ~]$ sudo dnf --showduplicates list kernel-devel | expand
Last metadata expiration check performed 1:32:42 ago on Fri Jan 15 21:23:10 2016.
Installed Packages
kernel-devel.x86_64                   4.3.3-300.fc23                    @updates
Available Packages
kernel-devel.x86_64                   4.2.3-300.fc23                    fedora  
kernel-devel.x86_64                   4.3.3-300.fc23                    @updates
kernel-devel.x86_64                   4.3.3-300.fc23                    updates

terrycain, 2016-01-16:
There seems to be issues with Virtualbox and fc23 so I'm using fc21 but everything should work the same.

terrycain, 2016-01-16:
Ok @Pant I had the same error as you. Try
`dnf install jq dbus-devel SDL2 SDL2-devel SDL2_image SDL2_image-devel fftw-libs-double fftw-libs-single gcc gcc-c++ make kernel-devel` then do `make clean` then `make` and as always pastebin the output ;)

terrycain, 2016-01-16:
Ok making an RPM was far too much hassle. You can try `make fedora_install` to install it (there's no uninstall yet)

Pant, 2016-01-16:
Still not working. I have install all the packages you mentioned (some were already installed).

Installed Packages

kernel-devel.x86_64                   4.2.7-300.fc23                    @updates
kernel-devel.x86_64                   4.2.8-300.fc23                    @updates
kernel-devel.x86_64                   4.3.3-300.fc23                    @updates

Available Packages

kernel-devel.x86_64                   4.2.3-300.fc23                    fedora  
kernel-devel.x86_64                   4.2.7-300.fc23                    @updates
kernel-devel.x86_64                   4.2.8-300.fc23                    @updates
kernel-devel.x86_64                   4.3.3-300.fc23                    @updates
kernel-devel.x86_64                   4.3.3-300.fc23                    updates 

make fedora_install
http://pastebin.com/DMA7Cu5X
make -s all install
http://pastebin.com/x4VmRtQ1
make
http://pastebin.com/bAhb4Zm6
make clean
http://pastebin.com/xrY7ESEe

terrycain, 2016-01-16:
What's the contents of `daemon/fx/Makefile`

Pant, 2016-01-16:
daemon/fx/Makefile:
http://pastebin.com/Qjzany2i

terrycain, 2016-01-16:
Output of `pkg-config --libs fftw3`

Pant, 2016-01-16:
pkg-config --libs fftw3 :
http://pastebin.com/z9zCKrtq
Perhaps this is the problem

terrycain, 2016-01-16:
Ok well the issue is that libfft is not installed or LD can't find it. On my fc21 VM `locate libfftw` returns some paths on `/usr/lib64`, what does it return for you?

Pant, 2016-01-16:
I runned ``updatedb`` first
locate libfftw:
http://pastebin.com/FsWqdT3g

terrycain, 2016-01-16:
Most strange, you have the same libraries I have. Try `sudo ldconfig -v` <- it prints a ton of output. Also `sudo ldconfig -v | grep fft`

First one will tell me what paths its searching. 2nd one will say if its found `libfft`

Pant, 2016-01-16:
``ldconfig -v``:
http://pastebin.com/QNPV5yp9
``ldconfig -v | grep fft``:
http://pastebin.com/y2SuB1L6

terrycain, 2016-01-16:
Ok so ld is finding the library. Did you try `sudo ldconfig; make clean; make` ?

Pant, 2016-01-16:
Just tried here is the output of ``make``:
http://pastebin.com/T9w2bH8d
``ldconfig`` did not gave any output.
I think make does not have any errors. Make fedora_install does.

pablomenino, 2016-01-16:
Hi @terrycain , for the  **Install Python3 library**, the directory is different on Fedora.
/usr/lib/python3.4/site-packages/

The driver work's fine (Manual Install. and I can change the effect using the bash scripts), I go to try the daemon and the GUI and let you know.

terrycain, 2016-01-16:
Ahh yes, some use dist-packages and some use site-packages. Will change that.

terrycain, 2016-01-16:
@Pant Try `sudo make fedora_install` with the latest changes as of now (fixed mistake @pablomenino found)

As in theory sudo make fedora_install *should* work at least it should compile all the files sucessfully and then might just fail at installation.

pablomenino, 2016-01-16:
@terrycain , for the first test, I install manually all the files, and I can confirm that the driver and the razer_chroma_controller is working fine.
I can change the effects and brightness whit this tool, no errors, (later i go to try installing whit Makefile).

I have some problems whit python scripts, but I have to resolve some dependencies for this scripts, I let you know what I found, thanks.

terrycain, 2016-01-16:
Cheers @pablomenino, I updated the makefile to put the python scripts in the correct place now ;)

Pant, 2016-01-16:
I still have errors. I run:
 ``git clone https://github.com/pez2001/razer_chroma_drivers.git``
``cd razer_chroma_drivers/``
``make fedora_install``
Here is the output: http://pastebin.com/79ea4R45

pablomenino, 2016-01-16:
@terrycain , there is one more package to install, whit this package the razer_tray_applet.py and the chroma_controller.py work's perfect.

**Install**
sudo dnf install libappindicator-gtk3

**Packagee info:**
libappindicator-gtk3.x86_64

terrycain, 2016-01-16:
ahhhhhhhh it all goes back to this
```
Package fftw3 was not found in the pkg-config search path.
Perhaps you should add the directory containing `fftw3.pc'
to the PKG_CONFIG_PATH environment variable
No package 'fftw3' found
```
Basically, make works as it just skips the fft library. The install expects it to be there, but it wasn't compiled as pkg_config isn't working propperly. The install is working technically, it just cant find the file we expect to exist.

More importantly do you have `fftw-devel` installed???????????

Pant, 2016-01-16:
``fftw-devel`` was the problem. Now there is an other error but i have no clue why.
``make fedora_install``: http://pastebin.com/rf53hs5s

terrycain, 2016-01-16:
Ahh yeah, that wouldn't be an issue if we had an uninstall script. Manually remove `/etc/systemd/system/razer_bcd.service`

terrycain, 2016-01-16:
@pablomenino 
So do you reckon this would be the package list? Ill add it to the readme soon.

`jq dbus-devel SDL2 SDL2-devel SDL2_image SDL2_image-devel fftw-libs-double fftw-libs-single gcc gcc-c++ make kernel-devel dbus-devel fftw-devel libappindicator-gtk3`

Pant, 2016-01-16:
So now if the driver is installed how should i start using it?

terrycain, 2016-01-16:
Has the whole install worked? Some rubbish about dkms should have appeared?

Pant, 2016-01-16:
I run ``make fedora_install`` and the output is the one i told you before. Is there something else I should do and I am missing?

pablomenino, 2016-01-16:
Yes @terrycain , I double check, whit that packages all can be compiled and works.

pablomenino, 2016-01-16:
@Pant , you manually removed the file /etc/systemd/system/razer_bcd.service ?

Pant, 2016-01-16:
Yes I did!

terrycain, 2016-01-16:
@pablomenino that file should just be a symlink to the one in /lib

lm92, 2016-02-14:
Hi, I'm not sure if I should open a new issue or just post this here, seems related...

I'm running Fedora 23, installed the following dependencies:

` dbus-devel SDL2 SDL2-devel SDL2_image SDL2_image-devel fftw-libs-double fftw-libs-single gcc gcc-c++ make kernel-devel dbus-devel fftw-devel libappindicator-gtk3 `

` $ sudo make fedora_install ` gives me: http://pastebin.com/Y9x3LS4F

The DKMS stuff is normal, yes?

Then, when I try to start the service it fails, ` systemctl statur razer_bcd.service ` outputs: http://pastebin.com/FmCyWAWq

Am I doing something wrong?

Thanks in advance!

terrycain, 2016-02-15:
Have you tried rebooting. And no the DKMS stuff shouldn't error, you need to install a DKMS package though I cant remember the exact one but it might just be called `dkms`, its not required but it becomes very helpful when you update kernels

lm92, 2016-02-15:
Rebooting didn't help, but installing ` dkms ` and re-compiling did it!
I'm not running the nvidia proprietary driver on this machine, which requires DKMS, so I forgot I didn't have it...

New problem however. When I try ` python razer_tray_applet.py `, I get: http://pastebin.com/s65nswZg

`python chroma_controller.py` gives: http://pastebin.com/XifZcF9j

Also, is there any way to enable macro keys just by using `razer_bcd_controller`? 

Thanks

lm92, 2016-02-15:
Ok, I forgot that Python 3 won't be default in Fedora until 24, so `sudo python3 chroma_controller.py` works!

However, `sudo python3 razer_tray_applet.py` now gives: http://pastebin.com/Us1JmNpk

pablomenino, 2016-02-15:
Hi @lm92 , for the Python script you need to install libappindicator-gtk3.

sudo dnf install libappindicator-gtk3

lm92, 2016-02-15:
Hi @pablomenino .
libappindicator-gtk3 was installed.
I restarted a gnome extension (TopIcons, places tray applets on a much saner place than gnome's default) and it now works! The critical error in the terminal output remains, but the app works normally.

Thanks, everyone!

PS: Does anyone know of an easy way to make the 'Enable macro keys' option persistent across reboots?

terrycain, 2016-02-15:
Yeah the critical error can be ignored.

@lm92 Thats probably a feature request ;). I do want to add that but am busy at the moment. Ask @lah7.
One solution can be to put `echo -n "1" > `/sys/bus/hid/drivers/razerkbd/KEYBOARD_DIR/macro_keys` in a startup file somewhere but that command requires root.

lm92, 2016-02-15:
@terrycain I might do something like that, I'm not too concerned about security on this machine.

PS: Thanks for making this, it is the definitive razer driver for Linux. Even under Windows it would be a nice alternative to their cloud based thing. I'd love to see on-the-fly macro recording supported, but I imagine it to be the hardest part to reverse engineer...

Regards

lah7, 2016-02-15:
@lm92 You're in luck. On-the-fly macros have already been (mostly) done. See @igorbb 's MarcoW tool, it's just not merged into this repository.
https://github.com/igorbb/MacroW (https://github.com/pez2001/razer_chroma_drivers/issues/76)

Will add it crashes for me if I haven't recorded a new marco for a while.

I'll add the "Enable macro keys" option to the GUI preferences as a TODO.

igorbb, 2016-02-15:
I will put a pull request soon.. But priority is finishing a paper for my
PhD.. And I am not the best writer.. But if you are in a hurry the macro on
my git is working.

lm92, 2016-02-15:
Awesome, this is exactly why I love open source stuff.

Thanks, everyone!

terrycain, 2016-02-15:
> it is the definitive razer driver for Linux

Cheers its always nice to hear that.

Once I've finished Uni I do intend on putting in a fair few hours a week to this. What's point of having all the flashy hardware if you can't show it off better than on windoze ;)

Pant, 2016-02-15:
Compilation works! But when i run ``sudo python3 razer_tray_applet.py`` it gives [this](http://pastebin.com/AwBnLYeG). When i run ``systemctl status razer_bcd.service`` it gives [this](http://pastebin.com/8w1mxUSV)

terrycain, 2016-02-15:
@Pant the GTK warnings can be ignored for now, cant replicate them but they shouldn't affect program operation. The `Failed to connect to the dbus service. Is the razer_bcd service running?` means the daemon isn't running.

As for the latter error, do you have the latest driver installed and have you tried rebooting?. I believe its because the start script expects a `razerfirefly` folder to exist but that might only get loaded if you actually have a firefly.

lah7, 2016-02-16:
I've "fixed" those PyGIWarnings in a future commit. They appear in newer versions of Python 3.x, harmless as you say. :+1:

terrycain, 2016-04-05:
Is anything outstanding or can the issue be closed?

Pant, 2016-04-06:
Yes it can be closed, but would like to mention that DKMS isn't installed by default on Fedora 23 and I had to install it.

closed 2016-04-06 by Pant

terrycain, 2016-04-06:
Ahh thanks for telling me. I've got some changes to how the packaging works
which might allow us to make rpms which would solve this