pez2001/razer_chroma_drivers #48: [....] Starting razer_bcd (via systemctl): razer_bcd.serviceJob for razer_bcd.service failed because the control process exited with error code. See "systemctl status razer_bcd.service" and "journalctl -xe" for details.

snok87, 2016-01-12:
It cant start in Ubuntu 15.10

terrycain, 2016-01-12:
what does running `journalctl -xe` output?

snok87, 2016-01-13:
-- Logs begin at dc 2016-01-13 20:00:35 CET, end at dc 2016-01-13 20:19:04 CET. --
gen 13 20:00:35 UbuntuPC kernel: zswap: using zbud pool
gen 13 20:00:35 UbuntuPC kernel: zswap: using lzo compressor
gen 13 20:00:35 UbuntuPC kernel: Key type trusted registered
gen 13 20:00:35 UbuntuPC kernel: Key type encrypted registered
gen 13 20:00:35 UbuntuPC kernel: AppArmor: AppArmor sha1 policy hashing enabled
gen 13 20:00:35 UbuntuPC kernel: ima: No TPM chip found, activating TPM-bypass!
gen 13 20:00:35 UbuntuPC kernel: evm: HMAC attrs: 0x1
gen 13 20:00:35 UbuntuPC kernel:   Magic number: 0:921:40
gen 13 20:00:35 UbuntuPC kernel: bdi 7:5: hash matches
gen 13 20:00:35 UbuntuPC kernel: memory memory76: hash matches
gen 13 20:00:35 UbuntuPC kernel: rtc_cmos 00:03: setting system clock to 2016-01-13 19:00:32 UTC (1452711632)
gen 13 20:00:35 UbuntuPC kernel: acpi-cpufreq: overriding BIOS provided _PSD data
gen 13 20:00:35 UbuntuPC kernel: BIOS EDD facility v0.16 2004-Jun-25, 0 devices found
gen 13 20:00:35 UbuntuPC kernel: EDD information not available.
gen 13 20:00:35 UbuntuPC kernel: PM: Hibernation image not present or could not be loaded.
gen 13 20:00:35 UbuntuPC kernel: Freeing unused kernel memory: 1460K (ffffffff81d37000 - ffffffff81ea4000)
gen 13 20:00:35 UbuntuPC kernel: Write protecting the kernel read-only data: 12288k
gen 13 20:00:35 UbuntuPC kernel: Freeing unused kernel memory: 32K (ffff8800017f8000 - ffff880001800000)
gen 13 20:00:35 UbuntuPC kernel: Freeing unused kernel memory: 296K (ffff880001bb6000 - ffff880001c00000)
gen 13 20:00:35 UbuntuPC kernel: random: systemd-udevd urandom read with 8 bits of entropy available
gen 13 20:00:35 UbuntuPC kernel: wmi: Mapper loaded
gen 13 20:00:35 UbuntuPC kernel: r8169 Gigabit Ethernet driver 2.3LK-NAPI loaded
gen 13 20:00:35 UbuntuPC kernel: r8169 0000:02:00.0: can't disable ASPM; OS doesn't have ASPM control
gen 13 20:00:35 UbuntuPC kernel: ahci 0000:00:11.0: version 3.0
gen 13 20:00:35 UbuntuPC kernel: ahci 0000:00:11.0: AHCI 0001.0200 32 slots 3 ports 6 Gbps 0x7 impl SATA mode
gen 13 20:00:35 UbuntuPC kernel: ahci 0000:00:11.0: flags: 64bit ncq sntf ilck pm led clo pmp pio slum part 
gen 13 20:00:35 UbuntuPC kernel: r8169 0000:02:00.0 eth0: RTL8168f/8111f at 0xffffc900018a6000, ac:9e:17:4e:9c:6e, XID 08000800 IRQ 36
gen 13 20:00:35 UbuntuPC kernel: r8169 0000:02:00.0 eth0: jumbo features [frames: 9200 bytes, tx checksumming: ko]
gen 13 20:00:35 UbuntuPC kernel: scsi host0: ahci
gen 13 20:00:35 UbuntuPC kernel: scsi host1: ahci
gen 13 20:00:35 UbuntuPC kernel: scsi host2: ahci
gen 13 20:00:35 UbuntuPC kernel: ata1: SATA max UDMA/133 abar m1024@0xfeb0b000 port 0xfeb0b100 irq 19
gen 13 20:00:35 UbuntuPC kernel: ata2: SATA max UDMA/133 abar m1024@0xfeb0b000 port 0xfeb0b180 irq 19
gen 13 20:00:35 UbuntuPC kernel: ata3: SATA max UDMA/133 abar m1024@0xfeb0b000 port 0xfeb0b200 irq 19
gen 13 20:00:35 UbuntuPC kernel: [drm] Initialized drm 1.1.0 20060810
gen 13 20:00:35 UbuntuPC kernel: usb 5-1: new high-speed USB device number 2 using ehci-pci
gen 13 20:00:35 UbuntuPC kernel: [drm] radeon kernel modesetting enabled.
gen 13 20:00:35 UbuntuPC kernel: r8169 0000:02:00.0 enp2s0: renamed from eth0
gen 13 20:00:35 UbuntuPC kernel: AMD IOMMUv2 driver by Joerg Roedel <jroedel@suse.de>
gen 13 20:00:35 UbuntuPC kernel: AMD IOMMUv2 functionality not available on this system
gen 13 20:00:35 UbuntuPC kernel: CRAT table not found
gen 13 20:00:35 UbuntuPC kernel: Finished initializing topology ret=0
gen 13 20:00:35 UbuntuPC kernel: kfd kfd: Initialized module
gen 13 20:00:35 UbuntuPC kernel: checking generic (c0000000 7f0000) vs hw (c0000000 10000000)
gen 13 20:00:35 UbuntuPC kernel: fb: switching to radeondrmfb from VESA VGA
gen 13 20:00:35 UbuntuPC kernel: Console: switching to colour dummy device 80x25
gen 13 20:00:35 UbuntuPC kernel: [drm] initializing kernel modesetting (PITCAIRN 0x1002:0x6810 0x174B:0xE271).
gen 13 20:00:35 UbuntuPC kernel: [drm] register mmio base: 0xFEA00000
gen 13 20:00:35 UbuntuPC kernel: [drm] register mmio size: 262144
gen 13 20:00:35 UbuntuPC kernel: ATOM BIOS: C63101
gen 13 20:00:35 UbuntuPC kernel: radeon 0000:01:00.0: VRAM: 2048M 0x0000000000000000 - 0x000000007FFFFFFF (2048M used)
gen 13 20:00:35 UbuntuPC kernel: radeon 0000:01:00.0: GTT: 2048M 0x0000000080000000 - 0x00000000FFFFFFFF
gen 13 20:00:35 UbuntuPC kernel: [drm] Detected VRAM RAM=2048M, BAR=256M
gen 13 20:00:35 UbuntuPC kernel: [drm] RAM width 256bits DDR
gen 13 20:00:35 UbuntuPC kernel: [TTM] Zone  kernel: Available graphics memory: 8165950 kiB
gen 13 20:00:35 UbuntuPC kernel: [TTM] Zone   dma32: Available graphics memory: 2097152 kiB
gen 13 20:00:35 UbuntuPC kernel: [TTM] Initializing pool allocator
gen 13 20:00:35 UbuntuPC kernel: [TTM] Initializing DMA pool allocator
gen 13 20:00:35 UbuntuPC kernel: [drm] radeon: 2048M of VRAM memory ready
gen 13 20:00:35 UbuntuPC kernel: [drm] radeon: 2048M of GTT memory ready.
gen 13 20:00:35 UbuntuPC kernel: [drm] Loading pitcairn Microcode
gen 13 20:00:35 UbuntuPC kernel: [drm] Internal thermal controller with fan control
gen 13 20:00:35 UbuntuPC kernel: [drm] probing gen 2 caps for device 1002:5a16 = 33ed02/0
gen 13 20:00:35 UbuntuPC kernel: usb 1-1: new full-speed USB device number 2 using xhci_hcd
gen 13 20:00:35 UbuntuPC kernel: [drm] radeon: dpm initialized
gen 13 20:00:35 UbuntuPC kernel: [drm] Found VCE firmware/feedback version 50.0.1 / 17!
gen 13 20:00:35 UbuntuPC kernel: [drm] GART: num cpu pages 524288, num gpu pages 524288
[... cut ...]
gen 13 20:17:01 UbuntuPC CRON[2106]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
gen 13 20:17:01 UbuntuPC CRON[2105]: pam_unix(cron:session): session closed for user root
gen 13 20:18:43 UbuntuPC gnome-keyring-daemon[1439]: couldn't allocate secure memory to keep passwords and or keys from being written to the disk
gen 13 20:19:04 UbuntuPC com.canonical.Unity.Scope.LocalFiles[1535]: (process:2999): unity-files-daemon-WARNING **: folder.vala:65: Failed to read favorites: Falló al abrir el archivo «/home/david/.gtk-bookmarks»: No existe el archivo o el directorio

snok87, 2016-01-13:
I just started Ubuntu with UpStart and then it work like a charm. :S

lah7, 2016-01-13:
It should work under systemd on Ubuntu 15.10 (it's perfectly fine here), all I can suggest is that everything has compiled properly (without errors) and try packaging it again for installation.

terrycain, 2016-01-13:
Uninstall it. restart, reinstall it and see.

snok87, 2016-01-14:
The same.
http://pastebin.com/Wi43KQE0

Everything compiled without errors.
http://pastebin.com/DgWzpuxC

terrycain, 2016-01-14:
Run `ls -l /usr/share/razer_bcd`

snok87, 2016-01-16:
david@UbuntuPC:~$ ls -l /usr/share/razer_bcd -R
/usr/share/razer_bcd:
total 8
-rw-rw-r-- 1 david david 2321 gen 14 19:19 bash_keyboard_functions.sh
drwxrwxr-x 2 david david 4096 gen 14 19:19 fx

/usr/share/razer_bcd/fx:
total 68
-rwxrwxr-x 1 david david 22320 gen 14 19:19 pez2001_collection.so
-rwxr-xr-x 1 root  root   5584 gen 12 21:34 pez2001_compute_fft.so
-rwxrwxr-x 1 david david  9968 gen 14 19:19 pez2001_light_blast.so
-rwxrwxr-x 1 david david 13920 gen 14 19:19 pez2001_mixer.so
-rwxrwxr-x 1 david david  5584 gen 14 19:19 pez2001_progress_bar.so

terrycain, 2016-01-17:
your missing `/usr/share/razer_bcd/systemd_helpers.sh`, Did you install the .deb? and if so it should of been made from `./package_for_debian.sh` as that one deals with systemd

aboettger, 2016-01-27:
Same problem here. DEB created and installed:

```
./package_for_ubuntu.sh
sudo dpkg -i razer-chroma-driver_1.0.0_amd64.deb
```

```
$ systemctl status razer_bcd.service
● razer_bcd.service - LSB: start/stops razer_bcd
   Loaded: loaded (/etc/init.d/razer_bcd)
   Active: failed (Result: exit-code) since Mi 2016-01-27 17:52:51 CET; 10s ago
     Docs: man:systemd-sysv-generator(8)

Jan 27 17:52:51 sturm systemd[1]: Stopped LSB: start/stops razer_bcd.
Jan 27 17:52:51 sturm systemd[1]: Starting LSB: start/stops razer_bcd...
Jan 27 17:52:51 sturm razer_bcd[26286]: * Starting Razer Keyboard daemon razer_bcd
Jan 27 17:52:51 sturm razer_bcd[26286]: Binding device (0003:1532:011A.0006) to razerkbd
Jan 27 17:52:51 sturm razer_bcd[26286]: /etc/init.d/razer_bcd: 20: /etc/init.d/razer_bcd: cannot create /sys/bus/hid/drivers/razerkbd/bind: Directory nonexistent
Jan 27 17:52:51 sturm systemd[1]: razer_bcd.service: Control process exited, code=exited status=2
Jan 27 17:52:51 sturm systemd[1]: Failed to start LSB: start/stops razer_bcd.
Jan 27 17:52:51 sturm systemd[1]: razer_bcd.service: Unit entered failed state.
Jan 27 17:52:51 sturm systemd[1]: razer_bcd.service: Failed with result 'exit-code'.
```

List of files in the package

```
$ sudo dpkg -L razer-chroma-driver                   
/.
/usr
/usr/lib
/usr/lib/librazer_chroma_controller.so
/usr/lib/librazer_chroma.so
/usr/lib/python3
/usr/lib/python3/dist-packages
/usr/lib/python3/dist-packages/razer
/usr/lib/python3/dist-packages/razer/keyboard.py
/usr/lib/python3/dist-packages/razer/daemon_dbus.py
/usr/lib/python3/dist-packages/razer/__init__.py
/usr/src
/usr/src/razer_chroma_driver-1.0.0
/usr/src/razer_chroma_driver-1.0.0/Makefile
/usr/src/razer_chroma_driver-1.0.0/driver
/usr/src/razer_chroma_driver-1.0.0/driver/Makefile
/usr/src/razer_chroma_driver-1.0.0/driver/razercommon.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerkbd_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razercommon.h
/usr/src/razer_chroma_driver-1.0.0/driver/razerkbd_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.h
/usr/src/razer_chroma_driver-1.0.0/dkms.conf
/usr/sbin
/usr/sbin/razer_bcd
/usr/share
/usr/share/razer_tray_applet
/usr/share/razer_tray_applet/razer_tray_applet.py
/usr/share/razer_tray_applet/tray_icon.png
/usr/share/razer_bcd
/usr/share/razer_bcd/bash_keyboard_functions.sh
/usr/share/razer_bcd/fx
/usr/share/razer_bcd/fx/pez2001_light_blast.so
/usr/share/razer_bcd/fx/pez2001_mixer.so
/usr/share/razer_bcd/fx/pez2001_collection.so
/usr/share/razer_bcd/fx/pez2001_progress_bar.so
/usr/share/applications
/usr/share/applications/razer_chroma_controller.desktop
/usr/share/applications/razer_tray_applet.desktop
/usr/share/razer_chroma_controller
/usr/share/razer_chroma_controller/chroma_controller.py
/usr/share/razer_chroma_controller/dynamic
/usr/share/razer_chroma_controller/data
/usr/share/razer_chroma_controller/data/fonts
/usr/share/razer_chroma_controller/data/fonts/Play-Regular.ttf
/usr/share/razer_chroma_controller/data/lib
/usr/share/razer_chroma_controller/data/lib/fontawesome-webfont.woff
/usr/share/razer_chroma_controller/data/lib/jquery-2.1.4.min.js
/usr/share/razer_chroma_controller/data/lib/font-awesome.min.css
/usr/share/razer_chroma_controller/data/lib/snap.svg-min.js
/usr/share/razer_chroma_controller/data/chroma_profiles.html
/usr/share/razer_chroma_controller/data/img
/usr/share/razer_chroma_controller/data/img/header.jpg
/usr/share/razer_chroma_controller/data/img/blackwidow-chroma-keyboard-layout.svg
/usr/share/razer_chroma_controller/data/img/marco-keys.svg
/usr/share/razer_chroma_controller/data/img/warning.svg
/usr/share/razer_chroma_controller/data/img/cursors
/usr/share/razer_chroma_controller/data/img/cursors/mode-clear.gif
/usr/share/razer_chroma_controller/data/img/cursors/mode-picker.gif
/usr/share/razer_chroma_controller/data/img/cursors/mode-set.gif
/usr/share/razer_chroma_controller/data/img/cursors/LICENSE
/usr/share/razer_chroma_controller/data/img/devices
/usr/share/razer_chroma_controller/data/img/devices/blackwidow_chroma.png
/usr/share/razer_chroma_controller/data/img/game-mode.svg
/usr/share/razer_chroma_controller/data/img/app-icon.svg
/usr/share/razer_chroma_controller/data/controller_devices.html
/usr/share/razer_chroma_controller/data/controller.css
/usr/share/razer_chroma_controller/data/chroma_menu.html
/usr/share/razer_chroma_controller/data/chroma_preferences.html
/usr/share/razer_chroma_controller/data/js
/usr/share/razer_chroma_controller/data/js/global.js
/usr/share/razer_chroma_controller/data/js/chroma_menu.js
/usr/share/razer_chroma_controller/data/js/profile_editor.js
/usr/share/razer_chroma_controller/data/js/devices.js
/usr/share/razer_chroma_controller/data/controller_service_error.html
/usr/bin
/usr/bin/razer_bcd_controller
/etc
/etc/init.d
/etc/init.d/razer_bcd
/etc/udev
/etc/udev/rules.d
/etc/udev/rules.d/95-razerkbd.rules
/etc/init
/etc/init/razer_bcd.conf
/etc/xdg
/etc/xdg/autostart
/etc/xdg/autostart/razer_tray_applet.desktop
/etc/dbus-1
/etc/dbus-1/system.d
/etc/dbus-1/system.d/org.voyagerproject.razer.daemon.conf
```

terrycain, 2016-01-27:
Try pulling all changes from the repo and recreate the deb.

snok87, 2016-01-28:
No way with ubuntu deb, and with debian deb i get a dependencies error

http://pastebin.com/SDyfNRQW

terrycain, 2016-01-29:
run `apt update` and then get output of `apt search libsdl2-image`

snok87, 2016-01-29:
http://pastebin.com/sa5G15Mc

terrycain, 2016-01-29:
I think the issue is the version numbers in the `control` file. Does `apt-get install -f` output anything interesting?

aboettger, 2016-01-30:
You have masked the service? Why that?

```
$ sudo systemctl restart razer_bcd.service
Failed to restart razer_bcd.service: Unit razer_bcd.service is masked.
```

snok87, 2016-02-03:
It works after unmasking! If someone dont know how:

sudo systemctl unmask razer_bcd.service

closed 2016-04-05 by terrycain