pez2001/razer_chroma_drivers #49: [gentoo] few bugs

Mat9999, 2016-01-12:
Hello, 
Thank you for working on this driver.

Here are some bugs I had during tests:

----------------------------------------------------------------------------------------------
### Compilation:
I had to modify razer_blackwidow_chroma_driver/daemon/fx/Makefile
I had to comment: 
ifeq ($(LIBFFTW3), 0) and endif
I don't know why this ifeq doesnt work.

----------------------------------------------------------------------------------------------
### UI and tray applet:
``` 
goldorak razer_blackwidow_chroma_driver # ./gui/chroma_controller/chroma_controller.py 
Traceback (most recent call last):
  File "./gui/chroma_controller/chroma_controller.py", line 22, in <module>
    import razer.daemon_dbus
**ImportError: No module named 'razer'**
``` 
``` 
goldorak razer_blackwidow_chroma_driver # ./gui/tray_applet/razer_tray_applet.py 
Traceback (most recent call last):
  File "./gui/tray_applet/razer_tray_applet.py", line 7, in <module>
    import appindicator
**ImportError: No module named appindicator**
``` 
----------------------------------------------------------------------------------------------

### dynamic:
some rgb arguments are not well parsed ?
##### example:
cat  fps.template
``` 
    2
    255 255 0
    1234
    [...]
```
goldorak dynamic-examples # cat fps.template |../dynamic
```
Enter r g b: Enter str: Setting keys of [1234] to 0 255 0...
Setting 1...
Setting 2...
Setting 3...
Setting 4...
Done!
[...]
Exiting...
```
----------------------------------------------------------------------------------------------
### Key bindings:
* using a french AZERTY keyboard, some keys mismatch since they are setup for qwerty. I had to indicate Q when the key was A on my keyboard (for example in examples/dota_keys.c)
* And I think some keys binding are incorrect:
I had to indicate Z key to setup the Y key, or Y to setup Z

##### example:
```    
        razer_convert_ascii_to_pos('Z',&pos);
        razer_set_key_pos(chroma->keys,&pos,&light_blue);
```
 will setup the Y key in light blue.

----------------------------------------------------------------------------------------------

terrycain, 2016-01-12:
> I had to modify razer_blackwidow_chroma_driver/daemon/fx/Makefile
> I had to comment: 
> ifeq ($(LIBFFTW3), 0) and endif
> I don't know why this ifeq doesnt work.

You probably need to install libfftw3 library for Fast Fourier transform. Or if its installed then the issue is that the line below doesnt do its job propperly which is detecting the presence of the library.
```
LIBFFTW3=$(shell echo \\\#include \<fftw3.h\>\\n int main\(int c,char**v\){} | gcc `pkg-config --libs fftw3` `pkg-config --cflags fftw3`  -xc -o a.out - 2>/dev/null ;echo $$? ;rm a.out 2>/dev/null )
```
---
> **ImportError: No module named 'razer'**

You need to move gui/lib/razer into the systems python path (for python3), usually `/usr/lib/python3/site-packages` or `/usr/lib/python3/dist-packages` or set by the enviromental variable PYTHONPATH. Look up the correct procedure for your distribution.

> **ImportError: No module named appindicator**

The new app is fully Python3 which might work on your system. Otherwise you'll have to look to installing it.

---
> dynamic:
> 
> some rgb arguments are not well parsed ?
Not a clue, I would recommend using the new python dbus combo to set the colours.

---
> Key bindings
Look at `scripts/driver/keyboard_keys.py` You'll need to edit the 2 lines near the top so it points to the driver files. When its runnning you need to press enter to cycle through the keys. Each key should light up red (sometimes they might not light up as there are gaps).

If you get it working, submit a document with a list of Keys and what row x:x the key matches up to.
Eg
```
ESC 0:1
...
M1 1:0
` KEY 1:1
...
```
If you can get that and manage to get the razer chroma gui to work then I can work on creating a keymap for other layouts.

Mat9999, 2016-01-12:
Thank you for your fast answer :)

###### libfftw3
```
goldorak chroma_controller # ls /usr/lib64/libfftw3*
/usr/lib64/libfftw3f_mpi.so        /usr/lib64/libfftw3f.so.3.4.4          /usr/lib64/libfftw3l_omp.so.3          /usr/lib64/libfftw3_mpi.so        /usr/lib64/libfftw3.so.3.4.4
/usr/lib64/libfftw3f_mpi.so.3      /usr/lib64/libfftw3f_threads.so        /usr/lib64/libfftw3l_omp.so.3.4.4      /usr/lib64/libfftw3_mpi.so.3      /usr/lib64/libfftw3_threads.so
/usr/lib64/libfftw3f_mpi.so.3.4.4  /usr/lib64/libfftw3f_threads.so.3      /usr/lib64/libfftw3l.so                /usr/lib64/libfftw3_mpi.so.3.4.4  /usr/lib64/libfftw3_threads.so.3
/usr/lib64/libfftw3f_omp.so        /usr/lib64/libfftw3f_threads.so.3.4.4  /usr/lib64/libfftw3l.so.3              /usr/lib64/libfftw3_omp.so        /usr/lib64/libfftw3_threads.so.3.4.4
/usr/lib64/libfftw3f_omp.so.3      /usr/lib64/libfftw3l_mpi.so            /usr/lib64/libfftw3l.so.3.4.4          /usr/lib64/libfftw3_omp.so.3
/usr/lib64/libfftw3f_omp.so.3.4.4  /usr/lib64/libfftw3l_mpi.so.3          /usr/lib64/libfftw3l_threads.so        /usr/lib64/libfftw3_omp.so.3.4.4
/usr/lib64/libfftw3f.so            /usr/lib64/libfftw3l_mpi.so.3.4.4      /usr/lib64/libfftw3l_threads.so.3      /usr/lib64/libfftw3.so
/usr/lib64/libfftw3f.so.3          /usr/lib64/libfftw3l_omp.so            /usr/lib64/libfftw3l_threads.so.3.4.4  /usr/lib64/libfftw3.so.3
goldorak chroma_controller # 
```
```
goldorak chroma_controller # echo \\\#include \<fftw3.h\>\\n int main\(int c,char**v\){} | gcc `pkg-config --libs fftw3` `pkg-config --cflags fftw3`  -xc -o a.out - 2>/dev/null ;echo $$? ;rm a.out 2>/dev/null
30943?
```
###### ImportError: No module named 'razer'
Ok I copied cp -prf gui/lib/razer /usr/lib/python3.4/site-packages/
But now:
```
goldorak chroma_controller # LANG=en_US.UTF-8 ./chroma_controller.py
Successfully connected to dbus service.
dirname: missing operand
Try 'dirname --help' for more information.

(acroread:12346): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(acroread:12346): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

```
And a blank window "Razer BlackWidow Chrome Configuration" is displaying
###### Key bindings
Ok, i will check  keyboard_keys.py and do a feedback

terrycain, 2016-01-12:
> libfftw3

OK yeah I suspected that was the issue. Try this line to see if it compiles with libfftw.
```
LIBFFTW3=$(shell pkg-config --libs fftw3 > /dev/null 2>&1 && echo 0)
```
---
> ImportError: No module named 'razer

Looks like the library is in the path. I would re-clone the repo I make some big changes recently. It should be installed into `/usr/share/razer_chroma_controller/` but should work fine being ran from the repo. I have no clue where the `dirname: mising operand` is coming from.

Mat9999, 2016-01-12:
###### llibfftw3
looks like
```
LIBFFTW3=$(shell pkg-config --libs fftw3 > /dev/null 2>&1 && echo 0)
```
do the job

###### ImportError: No module named 'razer
the dirname error is a acroread error

terrycain, 2016-01-13:
Good good. I used that command to fix some previous bugs but just missed
some evidently.

Acroread?

Mat9999, 2016-01-13:
Yes acroread is launched. The first time I succed to launch chroma_controller.py, an acroread window has been displayed. And then, I dont see it anymore, but the chroma_controller window is still opening but it is blank.

terrycain, 2016-01-13:
Hmm acroread shoulnt open I don't think. But the blank issue sounds like
something that happened to @lah7 a while ago. Have you re-cloned the repo
after my pull request got accepted?

Mat9999, 2016-01-13:
I just re-clone after your comment, and recompile, an cp gui/lib/razer/ to python path.
now:
```
oldorak razer_blackwidow_chroma_driver # LANG=en_US.UTF-8 gui/chroma_controller/chroma_controller.py 
Successfully connected to dbus service.
dirname: missing operand
Try 'dirname --help' for more information.

(acroread:11821): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(acroread:11821): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
```

terrycain, 2016-01-13:
Hmm, I'll look into the non absolute path stuff. Acroread is *probably* has something to do with either the GTK GUI parts or GTKWebkit. Next week sometime ill create a gentoo VM and investigate. How did oyu go about installing acroread (so I can reproduce this). After some googling it could be the fact that acroread has registered itself as a webkit plugin so therefore wants to load.

I'll patch that LIBFFT line soonish.

So all that's left is looking into is possibly packaging this for gentoo and fixing for the azerty keymap.

terrycain, 2016-01-13:
Have reproduced the white screen. I shall look into this.

terrycain, 2016-01-13:
Fixed `Makefile` and white screen issue in latest pull request #50

Mat9999, 2016-01-13:
/etc/rc?.d  directories are missing in gentoo, I had to temporary create them to avoid error:
```
goldorak razer_blackwidow_chroma_driver # make install
cp install_files/init.d/razer_bcd /etc/init.d
chown root:root /etc/init.d/razer_bcd
cp install_files/share/bash_keyboard_functions.sh /usr/share/razer_bcd/bash_keyboard_functions.sh
chown root:root /usr/share/razer_bcd/bash_keyboard_functions.sh
ln -fs ../init.d/razer_bcd /etc/rc2.d/S24razer_bcd
ln: failed to create symbolic link ‘/etc/rc2.d/S24razer_bcd’: No such file or directory
Makefile:44: recipe for target 'install' failed
make: *** [install] Error 1
```
there is /etc/runlevels/ insteed:
```
goldorak razer_blackwidow_chroma_driver # ls /etc/runlevels/
boot  default  nonetwork  shutdown  single  sysinit
goldorak razer_blackwidow_chroma_driver # ls /etc/runlevels/default/ -la
total 8
drwxr-xr-x 2 root root 4096 19 mars   2013 .
drwxr-xr-x 8 root root 4096  9 mai    2011 ..
lrwxrwxrwx 1 root root   17 30 avril  2010 acpid -> /etc/init.d/acpid
lrwxrwxrwx 1 root root   19  2 mai    2010 amavisd -> /etc/init.d/amavisd
[...]
lrwxrwxrwx 1 root root   15 28 avril  2010 xdm -> /etc/init.d/xdm

```
I am curently using systemd. systemd scripts installation looks OK
For gentoo packaging,I have never done an ebuild package until now, but I maybe I can help.

Mat9999, 2016-01-13:
azerty layout:

[image]

terrycain, 2016-01-13:
Cheers for the picture. As for `rc.d` that'll only work for systems that use sysvinit. For systemd use the appropriate scripts.

You fancy making a script to package it for gentoo?

terrycain, 2016-01-13:
Cheers for the picture (ill make layouts at some point). As for `rc.d` that'll only work for systems that use sysvinit. For systemd use the appropriate scripts.

You fancy making a script to package it for gentoo?

Mat9999, 2016-01-14:
Hello Terrycain,
I am sorry, I will not have the time for working on and maintaining a gentoo package.
I can however regularly manually pull update and inform you if I find some other bug.

terrycain, 2016-01-14:
Ok no worries. Eventually I might get to making a gentoo package. I am however working on the application having multiple keyboard layouts.

terrycain, 2016-01-15:
@Mat9999 Does the chroma_controller app work for you? and if so does the uk layout on the app light up the keys on the keyboard in the same place?

So like Q would light up your A, Z would light up W etc...

Mat9999, 2016-01-15:
Hi,
Yes it works very well :) I have try each Effects, and add two templates Dota and FPS.
The physical keys are lighting up exactly as the  diagram show:
Activating Q on the diagram will light up A on the physical azerty keyboard, 
W will light up Z and Z will light W
A will light up W, Q will light up A etc...
So an effect template or a profile for QWERTY will fit perfectly on an AZERTY one :)

terrycain, 2016-01-15:
Ok, cool thats nice thats what I expected to happen. I am figuring out how to best detect the layout, then I'll get the app to change accordingly. What's your keyboard code called? Like the UK one is en-gb, US is en-us etc...

Mat9999, 2016-01-15:
I live in France. I'm not sure AZERTY is only used in France, but in France the standard is AZERTY.
If it can help:
My linux locale is set up to fr_FR.utf8
My X11 settings are "xkb_model" "pc105",  "xkb_layout" "fr",  "xkb_variant" "latin9"

terrycain, 2016-01-15:
Whats the output of `setxkbmap -print`? and the contents of `/etc/default/keyboard`

Mat9999, 2016-01-15:
```
goldorak ~ # setxkbmap -print
xkb_keymap {
        xkb_keycodes  { include "evdev+aliases(azerty)" };
        xkb_types     { include "complete"      };
        xkb_compat    { include "complete"      };
        xkb_symbols   { include "pc+fr(latin9)+inet(evdev)+terminate(ctrl_alt_bksp)"  };
        xkb_geometry  { include "pc(pc104)"     };
};
```
```
goldorak ~ # cat /etc/vconsole.conf 
KEYMAP=fr-latin9

```
```
goldorak ~ # cat /etc/conf.d/keymaps 
# Use keymap to specify the default console keymap.  There is a complete tree
# of keymaps in /usr/share/keymaps to choose from.
keymap="fr-latin9"

# Should we first load the 'windowkeys' console keymap?  Most x86 users will
# say "yes" here.  Note that non-x86 users should leave it as "no".
# Loading this keymap will enable VT switching (like ALT+Left/Right)
# using the special windows keys on the linux console.
windowkeys="YES"

# The maps to load for extended keyboards.  Most users will leave this as is.
extended_keymaps=""
#extended_keymaps="backspace keypad euro2"

# Tell dumpkeys(1) to interpret character action codes to be
# from the specified character set.
# This only matters if you set unicode="yes" in /etc/rc.conf.
# For a list of valid sets, run `dumpkeys --help`
dumpkeys_charset=""

# Some fonts map AltGr-E to the currency symbol instead of the Euro.
# To fix this, set to "yes"
fix_euro="NO"
```

terrycain, 2016-01-15:
`setxkbmap -query`?

Mat9999, 2016-01-15:
It gives me information about model: logitech_base....It should be my mouse ?
```
goldorak razer_blackwidow_chroma_driver # setxkbmap -query
rules:      evdev
model:      logitech_base
layout:     fr
variant:    latin9
options:    terminate:ctrl_alt_bksp
```

terrycain, 2016-01-15:
Cool. Need to make the keyboard layout detection a bit more robust but in theory it should work. Ill make a layout in the SVG and upload it later tonight.

Mat9999, 2016-01-15:
Ok,
i just pull last commit, compilation is fine, module is loaded, but new systemd unit failed to launch:
```
goldorak razer_blackwidow_chroma_driver # systemctl daemon-reload
goldorak razer_blackwidow_chroma_driver # systemctl status razer_bcd
● razer_bcd.service - Razer Keyboard Daemon
   Loaded: loaded (/usr/lib64/systemd/system/razer_bcd.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since ven. 2016-01-15 18:21:54 CET; 3s ago
  Process: 6298 ExecStopPost=/usr/share/razer_bcd/systemd_helpers.sh unbind (code=exited, status=0/SUCCESS)
  Process: 6293 ExecStart=/usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid (code=exited, status=0/SUCCESS)
  Process: 6260 ExecStartPre=/usr/share/razer_bcd/systemd_helpers.sh bind (code=exited, status=0/SUCCESS)
 Main PID: 6296 (code=killed, signal=SEGV)

janv. 15 18:21:54 goldorak.actarus systemd_helpers.sh[6260]: Unbinding device (0003:1532:0203.0003) from hid-generic
janv. 15 18:21:54 goldorak.actarus systemd_helpers.sh[6260]: Binding device (0003:1532:0203.0003) to razerkbd
janv. 15 18:21:54 goldorak.actarus systemd_helpers.sh[6260]: Bind Successful
janv. 15 18:21:54 goldorak.actarus razer_bcd[6293]: Starting razer blackwidow chroma daemon as a daemon
janv. 15 18:21:54 goldorak.actarus systemd[1]: razer_bcd.service: main process exited, code=killed, status=11/SEGV
janv. 15 18:21:54 goldorak.actarus systemd_helpers.sh[6298]: Unbinding device (0003:1532:0203.0003) from razerkbd
janv. 15 18:21:54 goldorak.actarus systemd_helpers.sh[6298]: Binding device (0003:1532:0203.0003) to hid-generic
janv. 15 18:21:54 goldorak.actarus systemd[1]: Unit razer_bcd.service entered failed state.
janv. 15 18:21:54 goldorak.actarus systemd[1]: razer_bcd.service failed.
janv. 15 18:21:54 goldorak.actarus systemd_helpers.sh[6298]: Bind Successful
```
I think the culprit is ```/usr/sbin/razer_bcd```
when I replace new version of /usr/sbin/razer_bcd with old (from 13january), old is working again

terrycain, 2016-01-15:
That would be the settings daemon.

If you go to the [Wiki](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver) and bind the driver manually. Then run `strace /usr/sbin/razer_bcd`

Mat9999, 2016-01-15:
```
goldorak ~ # strace /usr/sbin/razer_bcd
execve("/usr/sbin/razer_bcd", ["/usr/sbin/razer_bcd"], [/* 80 vars */]) = 0
brk(0)                                  = 0x10ea000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f1e56750000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=329078, ...}) = 0
mmap(NULL, 329078, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f1e566ff000
close(3)                                = 0
open("/lib64/librazer_chroma.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0' \232?\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=44928, ...}) = 0
mmap(0x3f9a200000, 2138696, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x3f9a200000
mprotect(0x3f9a20a000, 2093056, PROT_NONE) = 0
mmap(0x3f9a409000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0x3f9a409000
close(3)                                = 0
open("/lib64/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0 U`\231?\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1053168, ...}) = 0
mmap(0x3f99600000, 3146072, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x3f99600000
mprotect(0x3f99700000, 2093056, PROT_NONE) = 0
mmap(0x3f998ff000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xff000) = 0x3f998ff000
close(3)                                = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f1e566fe000
open("/lib64/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p\16\340\231?\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=16768, ...}) = 0
mmap(0x3f99e00000, 2109712, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x3f99e00000
mprotect(0x3f99e02000, 2097152, PROT_NONE) = 0
mmap(0x3f9a002000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x3f9a002000
close(3)                                = 0
open("/usr/lib64/libdbus-1.so.3", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260\347`\244?\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=282768, ...}) = 0
mmap(0x3fa4600000, 2375928, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x3fa4600000
mprotect(0x3fa4643000, 2093056, PROT_NONE) = 0
mmap(0x3fa4842000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x42000) = 0x3fa4842000
mmap(0x3fa4844000, 248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x3fa4844000
close(3)                                = 0
open("/lib64/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p\t\"\231?\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1661288, ...}) = 0
mmap(0x3f99200000, 3762584, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x3f99200000
mprotect(0x3f9938e000, 2093056, PROT_NONE) = 0
mmap(0x3f9958d000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x18d000) = 0x3f9958d000
mmap(0x3f99593000, 14744, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x3f99593000
close(3)                                = 0
open("/lib64/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0 q\240\231?\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=133160, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f1e566fd000
mmap(0x3f99a00000, 2208976, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x3f99a00000
mprotect(0x3f99a16000, 2097152, PROT_NONE) = 0
mmap(0x3f99c16000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x16000) = 0x3f99c16000
mmap(0x3f99c18000, 13520, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x3f99c18000
close(3)                                = 0
open("/usr/lib64/libsystemd.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\340\237?\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=189416, ...}) = 0
mmap(0x3f9fe00000, 190672, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x3f9fe00000
mmap(0x3f9fe2d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2c000) = 0x3f9fe2d000
close(3)                                = 0
open("/lib64/libcap.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p\32 \237?\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=25248, ...}) = 0
mmap(0x3f9f200000, 2118256, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x3f9f200000
mprotect(0x3f9f204000, 2097152, PROT_NONE) = 0
mmap(0x3f9f404000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0x3f9f404000
close(3)                                = 0
[... 48 lines cut ...]
mprotect(0x3f9a805000, 4096, PROT_READ) = 0
mprotect(0x3f9f404000, 4096, PROT_READ) = 0
mprotect(0x3f9fe2d000, 4096, PROT_READ) = 0
mprotect(0x3fa4842000, 4096, PROT_READ) = 0
mprotect(0x3f9a002000, 4096, PROT_READ) = 0
mprotect(0x3f998ff000, 4096, PROT_READ) = 0
mprotect(0x3f9a409000, 4096, PROT_READ) = 0
mprotect(0x60e000, 4096, PROT_READ)     = 0
mprotect(0x3f99020000, 4096, PROT_READ) = 0
munmap(0x7f1e566ff000, 329078)          = 0
[... cut ...]
getrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
fstat(1, {st_mode=S_IFCHR|0600, st_rdev=makedev(136, 0), ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f1e5674f000
write(1, "Starting razer blackwidow chroma"..., 52Starting razer blackwidow chroma daemon as a daemon
) = 52
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f1e566f7a10) = 12060
exit_group(0)                           = ?
+++ exited with 0 +++
goldorak ~ # 
```

terrycain, 2016-01-15:
`ldd /usr/sbin/razer_bcd`

Mat9999, 2016-01-15:
```
goldorak ~ # ldd /usr/sbin/razer_bcd
        linux-vdso.so.1 (0x00007ffc907e1000)
        librazer_chroma.so => /lib64/librazer_chroma.so (0x0000003f9a200000)
        libm.so.6 => /lib64/libm.so.6 (0x0000003f99600000)
        libdl.so.2 => /lib64/libdl.so.2 (0x0000003f99e00000)
        libdbus-1.so.3 => /usr/lib64/libdbus-1.so.3 (0x0000003fa4600000)
        libc.so.6 => /lib64/libc.so.6 (0x0000003f99200000)
        /lib64/ld-linux-x86-64.so.2 (0x0000003f98e00000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x0000003f99a00000)
        libsystemd.so.0 => /usr/lib64/libsystemd.so.0 (0x0000003f9fe00000)
        libcap.so.2 => /lib64/libcap.so.2 (0x0000003f9f200000)
        librt.so.1 => /lib64/librt.so.1 (0x0000003f9a600000)
        liblzma.so.5 => /lib64/liblzma.so.5 (0x0000003f9d200000)
        liblz4.so.1 => /usr/lib64/liblz4.so.1 (0x0000003fa0000000)
        libresolv.so.2 => /lib64/libresolv.so.2 (0x0000003f9c200000)
        libattr.so.1 => /lib64/libattr.so.1 (0x0000003f9ea00000)
goldorak ~ # 
```

terrycain, 2016-01-15:
Not sure about the bugs above (ill look into that another time), but on the other hand there is this ;)
[image]

Mat9999, 2016-01-15:
nice :)

Mat9999, 2016-01-15:
some cosmetic:
[image]

pez2001, 2016-01-15:
gdb output would really help here. if possible try running sudo gdb --args
daemon/razer_bcd_debug_static -f
after the crash enter the 'bt' command hit enter and copy the output.
bg

Mat9999, 2016-01-15:
:) A looooong time ago since the last time I used gdb :)
here's output:
```
goldorak daemon # pwd
/root/razer_blackwidow_chroma_driver/daemon
goldorak daemon # gdb --args ./razer_bcd_debug_static -f
GNU gdb (Gentoo 7.10.1 vanilla) 7.10.1
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./razer_bcd_debug_static...done.
(gdb) run
Starting program: /root/razer_blackwidow_chroma_driver/daemon/razer_bcd_debug_static -f
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Starting razer blackwidow chroma daemon in the foreground
opening chroma lib
No chroma devices found![Inferior 1 (process 21243) exited with code 01]
(gdb) bt
No stack.
(gdb) 

```

pez2001, 2016-01-16:
sorry , i was sure i saw a crash in your output, i was wrong.
is the driver bound to the devices you are using ?

Mat9999, 2016-01-16:
There is a segmentation fault, but I dont understand how to reproduce it with debug exec
```
oldorak razer_blackwidow_chroma_driver # make uninstall
make -C lib uninstall
[...]
rm /etc/rc0.d/K02razer_bcd
rm /etc/rc1.d/K02razer_bcd
rm /etc/rc6.d/K02razer_bcd
rm /etc/dbus-1/system.d/org.voyagerproject.razer.daemon.conf
rm /usr/sbin/razer_blackwidow_chroma_activate_driver.sh
rm: impossible de supprimer « /usr/sbin/razer_blackwidow_chroma_activate_driver.sh »: Aucun fichier ou dossier de ce type
Makefile:76 : la recette pour la cible « uninstall » a échouée
make: *** [uninstall] Erreur 1
```
```
goldorak razer_blackwidow_chroma_driver # make clean
make -C lib clean 
[...]
make[1] : on quitte le répertoire « /root/razer_blackwidow_chroma_driver/examples »
make -C /lib/modules/4.1.12-gentoo/build SUBDIRS=/root/razer_blackwidow_chroma_driver/driver clean > /dev/null 2>&1
#added redirect to remove output of a useless makefile warning
```
```
goldorak razer_blackwidow_chroma_driver # make all
make -C lib all
make[1] : on entre dans le répertoire « /root/razer_blackwidow_chroma_driver/lib »
[...]
========================================
make -C /lib/modules/4.1.12-gentoo/build SUBDIRS=/root/razer_blackwidow_chroma_driver/driver modules > /dev/null 2>&1
goldorak razer_blackwidow_chroma_driver # 
```
```
goldorak razer_blackwidow_chroma_driver # modprobe razerkbd
goldorak razer_blackwidow_chroma_driver # systemctl start razer_bcd
Job for razer_bcd.service failed. See "systemctl status razer_bcd.service" and "journalctl -xe" for details.

```
```
goldorak razer_blackwidow_chroma_driver # make install
make -C lib all
make[1] : on entre dans le répertoire « /root/razer_blackwidow_chroma_driver/lib »
ar -c -rs librazer_chroma.a razer_chroma.o list.o razer_string.o
[...]
ln -fs ../init.d/razer_bcd /etc/rc1.d/K02razer_bcd
ln -fs ../init.d/razer_bcd /etc/rc6.d/K02razer_bcd
```
```
goldorak razer_blackwidow_chroma_driver # modprobe razerkbd
goldorak razer_blackwidow_chroma_driver # lsmod | grep razer
razerkbd                9372  0
```
```
goldorak razer_blackwidow_chroma_driver # systemctl start razer_bcd
goldorak razer_blackwidow_chroma_driver # systemctl status razer_bcd
● razer_bcd.service - Razer Keyboard Daemon
   Loaded: loaded (/usr/lib64/systemd/system/razer_bcd.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since sam. 2016-01-16 08:45:22 CET; 17s ago
  Process: 30494 ExecStopPost=/usr/share/razer_bcd/systemd_helpers.sh unbind (code=exited, status=0/SUCCESS)
  Process: 30488 ExecStart=/usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid (code=exited, status=0/SUCCESS)
  Process: 30459 ExecStartPre=/usr/share/razer_bcd/systemd_helpers.sh bind (code=exited, status=0/SUCCESS)
 Main PID: 30489 (code=killed, signal=SEGV)

janv. 16 08:45:22 goldorak.actarus systemd_helpers.sh[30459]: Device already binded
janv. 16 08:45:22 goldorak.actarus razer_bcd[30488]: Starting razer blackwidow chroma daemon as a daemon
janv. 16 08:45:22 goldorak.actarus systemd[1]: razer_bcd.service: main process exited, code=killed, status=11/SEGV
janv. 16 08:45:22 goldorak.actarus systemd_helpers.sh[30494]: Unbinding device (0003:1532:0203.0003) from razerkbd
janv. 16 08:45:22 goldorak.actarus systemd_helpers.sh[30494]: Binding device (0003:1532:0203.0003) to hid-generic
janv. 16 08:45:22 goldorak.actarus systemd_helpers.sh[30494]: Bind Successful
janv. 16 08:45:22 goldorak.actarus systemd[1]: Unit razer_bcd.service entered failed state.
janv. 16 08:45:22 goldorak.actarus systemd[1]: razer_bcd.service failed.
[... 27 lines cut ...]
gdb) bt
No stack.
(gdb) quit
```
```
goldorak razer_blackwidow_chroma_driver # cp ../razer_blackwidow_chroma_driver.ok/daemon/razer_bcd /usr/sbin/razer_bcd
cp : voulez-vous écraser « /usr/sbin/razer_bcd » ? y
goldorak razer_blackwidow_chroma_driver # systemctl start razer_bcd
goldorak razer_blackwidow_chroma_driver # status^C
goldorak razer_blackwidow_chroma_driver # systemctl status razer_bcd
● razer_bcd.service - Razer Keyboard Daemon
   Loaded: loaded (/usr/lib64/systemd/system/razer_bcd.service; enabled; vendor preset: enabled)
   Active: active (running) since sam. 2016-01-16 08:54:41 CET; 17s ago
  Process: 30494 ExecStopPost=/usr/share/razer_bcd/systemd_helpers.sh unbind (code=exited, status=0/SUCCESS)
  Process: 2066 ExecStart=/usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid (code=exited, status=0/SUCCESS)
  Process: 2029 ExecStartPre=/usr/share/razer_bcd/systemd_helpers.sh bind (code=exited, status=0/SUCCESS)
 Main PID: 2068 (razer_bcd)
   CGroup: /system.slice/razer_bcd.service
           └─2068 /usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid

janv. 16 08:54:41 goldorak.actarus systemd_helpers.sh[2029]: Unbinding device (0003:1532:0203.0003) from hid-generic
janv. 16 08:54:41 goldorak.actarus systemd_helpers.sh[2029]: Binding device (0003:1532:0203.0003) to razerkbd
janv. 16 08:54:41 goldorak.actarus systemd_helpers.sh[2029]: Bind Successful
janv. 16 08:54:41 goldorak.actarus razer_bcd[2066]: Starting razer blackwidow chroma daemon as a daemon
```

terrycain, 2016-01-16:
Could be my udev script acting up, looks like some systems might not have interface num provided by udev, but they should. And the pre-start scripts in systemd should unsure the driver is binded.

What does `ls /sys/bus/hid/drivers/razerkbd/` give you?
Also to write in /sys you need to be root so running the daemon as a standard user would always result in errors.

Rerun this `./daemon/razer_bcd_debug_static -f` as root

Mat9999, 2016-01-16:
OK, some new tests:
```
goldorak razer_blackwidow_chroma_driver # ls /sys/bus/hid/drivers/razerkbd/ -la
total 0
drwxr-xr-x  2 root root    0 16 janv. 12:13 .
drwxr-xr-x 20 root root    0 16 janv. 08:44 ..
--w-------  1 root root 4096 16 janv. 12:13 bind
lrwxrwxrwx  1 root root    0 16 janv. 09:39 module -> ../../../../module/razerkbd
--w-------  1 root root 4096 16 janv. 12:10 new_id
--w-------  1 root root 4096 16 janv. 12:10 uevent
--w-------  1 root root 4096 16 janv. 12:13 unbind
```
```
goldorak razer_blackwidow_chroma_driver # ./install_files/share/systemd_helpers.sh bind
Unbinding device (0003:1532:0203.0003) from hid-generic
Binding device (0003:1532:0203.0003) to razerkbd
Bind Successful
goldorak razer_blackwidow_chroma_driver # ls /sys/bus/hid/drivers/razerkbd/ -la
total 0
drwxr-xr-x  2 root root    0 16 janv. 12:18 .
drwxr-xr-x 20 root root    0 16 janv. 08:44 ..
lrwxrwxrwx  1 root root    0 16 janv. 12:18 0003:1532:0203.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.3/1-10.3:1.2/0003:1532:0203.0003
--w-------  1 root root 4096 16 janv. 12:18 bind
lrwxrwxrwx  1 root root    0 16 janv. 09:39 module -> ../../../../module/razerkbd
--w-------  1 root root 4096 16 janv. 12:10 new_id
--w-------  1 root root 4096 16 janv. 12:10 uevent
--w-------  1 root root 4096 16 janv. 12:18 unbind
goldorak razer_blackwidow_chroma_driver # 
goldorak razer_blackwidow_chroma_driver # ./daemon/razer_bcd --pid-file /var/run/razer_bcd.pid
Starting razer blackwidow chroma daemon as a daemon
```
kernel.log:
```
2016/01/16 12:19:53 kernel          Jan 16 12:19:53 goldorak kernel: traps: razer_bcd[2700] general protection ip:3f9a205514 sp:7ffd4e8c7d80 error:0 in librazer_chroma.so[3f9a200000+a000]
```
```
goldorak razer_blackwidow_chroma_driver # ps auxww|grep razer
root      3668  0.0  0.0 112608  2204 pts/0    S+   12:22   0:00 grep --color=auto razer
goldorak razer_blackwidow_chroma_driver # 
```
```
goldorak razer_blackwidow_chroma_driver # ./daemon/razer_bcd_debug_shared --pid-file /var/run/razer_bcd.pid
./daemon/razer_bcd_debug_shared: error while loading shared libraries: librazer_chroma_debug.so: cannot open shared object file: No such file or directory
goldorak razer_blackwidow_chroma_driver # cp lib/librazer_chroma_debug.so /lib64/
goldorak razer_blackwidow_chroma_driver # ./daemon/razer_bcd_debug_shared --pid-file /var/run/razer_bcd.pid
Starting razer blackwidow chroma daemon as a daemon
killing razer_bcd parent process
goldorak razer_blackwidow_chroma_driver # ps auxww|grep razer
root      6027  0.6  0.0  36548  2528 ?        Ss   12:27   0:00 ./daemon/razer_bcd_debug_shared --pid-file /var/run/razer_bcd.pid
```
this last previous command works
```
goldorak razer_blackwidow_chroma_driver # kill -9 6027
oldorak razer_blackwidow_chroma_driver # ./daemon/razer_bcd_debug_static --pid-file /var/run/razer_bcd.pid
Starting razer blackwidow chroma daemon as a daemon
killing razer_bcd parent process
goldorak razer_blackwidow_chroma_driver # ps auxww| grep razer
root      6796  0.5  0.0  34480  2496 ?        Ss   12:29   0:00 ./daemon/razer_bcd_debug_static --pid-file /var/run/razer_bcd.pid
root      6817  0.0  0.0 112608  2252 pts/0    S+   12:29   0:00 grep --color=auto razer
```
this too. And now:
```
goldorak razer_blackwidow_chroma_driver # ./daemon/razer_bcd --pid-file /var/run/razer_bcd.pid
Starting razer blackwidow chroma daemon as a daemon
```
----kernel.log:
```
2016/01/16 12:30:38 messages        Jan 16 12:30:38 goldorak kernel: traps: razer_bcd[7247] general protection ip:3f9a205514 sp:7ffcdf171490 error:0 in librazer_chroma.so[3f9a200000+a000]
2016/01/16 12:30:38 kernel          Jan 16 12:30:38 goldorak kernel: traps: razer_bcd[7247] general protection ip:3f9a205514 sp:7ffcdf171490 error:0 in librazer_chroma.so[3f9a200000+a000]
```
[... 130 lines cut ...]
mprotect(0x60e000, 4096, PROT_READ)     = 0
mprotect(0x3f99020000, 4096, PROT_READ) = 0
munmap(0x7fa6b02d3000, 329568)          = 0
set_tid_address(0x7fa6b02cba10)         = 11690
set_robust_list(0x7fa6b02cba20, 24)     = 0
rt_sigaction(SIGRTMIN, {0x3f99a06bf0, [], SA_RESTORER|SA_SIGINFO, 0x3f99a10410}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {0x3f99a06c80, [], SA_RESTORER|SA_RESTART|SA_SIGINFO, 0x3f99a10410}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
getrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
brk(0)                                  = 0xe48000
brk(0xe69000)                           = 0xe69000
fstat(1, {st_mode=S_IFCHR|0600, st_rdev=makedev(136, 0), ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa6b0323000
write(1, "Starting razer blackwidow chroma"..., 52Starting razer blackwidow chroma daemon as a daemon
) = 52
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fa6b02cba10) = 11691
exit_group(0)                           = ?
+++ exited with 0 +++
goldorak razer_blackwidow_chroma_driver # 
```
---kernel.log
```
2016/01/16 12:43:51 messages        Jan 16 12:43:51 goldorak kernel: traps: razer_bcd[13276] general protection ip:3f9a205514 sp:7ffc1f97ccc0 error:0 in librazer_chroma.so[3f9a200000+a000]
2016/01/16 12:43:51 kernel          Jan 16 12:43:51 goldorak kernel: traps: razer_bcd[13276] general protection ip:3f9a205514 sp:7ffc1f97ccc0 error:0 in librazer_chroma.so[3f9a200000+a000]
```

terrycain, 2016-01-16:
run `ldd daemon/razer_bcd`

Mat9999, 2016-01-16:
```
goldorak razer_blackwidow_chroma_driver # ldd daemon/razer_bcd
        linux-vdso.so.1 (0x00007ffe395c7000)
        librazer_chroma.so => /lib64/librazer_chroma.so (0x0000003f9a200000)
        libm.so.6 => /lib64/libm.so.6 (0x0000003f99600000)
        libdl.so.2 => /lib64/libdl.so.2 (0x0000003f99e00000)
        libdbus-1.so.3 => /usr/lib64/libdbus-1.so.3 (0x0000003fa4600000)
        libc.so.6 => /lib64/libc.so.6 (0x0000003f99200000)
        /lib64/ld-linux-x86-64.so.2 (0x0000003f98e00000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x0000003f99a00000)
        libsystemd.so.0 => /usr/lib64/libsystemd.so.0 (0x0000003f9fe00000)
        libcap.so.2 => /lib64/libcap.so.2 (0x0000003f9f200000)
        librt.so.1 => /lib64/librt.so.1 (0x0000003f9a600000)
        liblzma.so.5 => /lib64/liblzma.so.5 (0x0000003f9d200000)
        liblz4.so.1 => /usr/lib64/liblz4.so.1 (0x0000003fa0000000)
        libresolv.so.2 => /lib64/libresolv.so.2 (0x0000003f9c200000)
        libattr.so.1 => /lib64/libattr.so.1 (0x0000003f9ea00000)
goldorak razer_blackwidow_chroma_driver # 
```

terrycain, 2016-01-16:
have you tried replacing /lib64/librazer_chroma.so with the one in git_checkout/lib/librazer_chroma.so

Mat9999, 2016-01-16:
I just did it, and: its working :)
```
goldorak razer_blackwidow_chroma_driver # cp lib/librazer_chroma.so /lib64/librazer_chroma.so 
cp : voulez-vous écraser « /lib64/librazer_chroma.so » ? y
goldorak razer_blackwidow_chroma_driver # ldconfig
goldorak razer_blackwidow_chroma_driver # ./daemon/razer_bcd --pid-file /var/run/razer_bcd.pid
Starting razer blackwidow chroma daemon as a daemon
goldorak razer_blackwidow_chroma_driver # ps auxww | grep razer
root     26430  0.6  0.0  36512  2488 ?        Ss   13:14   0:00 ./daemon/razer_bcd --pid-file /var/run/razer_bcd.pid
root     27219  0.0  0.0 112608  2252 pts/0    S+   13:16   0:00 grep --color=auto razer
goldorak razer_blackwidow_chroma_driver # 
```

terrycain, 2016-01-16:
The new version has some debug logging to syslog with `--verbose=7` (getopt requires the = and not a space :( ) 

Does the chroma_controller show you the correct keyboard? (Might need to download new changes for that though)

Mat9999, 2016-01-16:
Yes, I can see the azerty layout :)

terrycain, 2016-01-16:
Would you mind testing each key? In theory they should work.

Mat9999, 2016-01-16:
Ok, i will in a moment.

Mat9999, 2016-01-16:
I have done this test:
[image]
It's working fine

terrycain, 2016-01-16:
Awesome cheers. Mind if I mark this as closed now?

Mat9999, 2016-01-16:
Its Ok, thank you for this good work.

closed 2016-01-16 by Mat9999

Mat9999, 2016-01-16:
If you are interrested in, there profiles that I copied from official razer 
templates here;

http://www.razerzone.com/gaming-keyboards-keypads/razer-blackwidow-chroma

starcraft 2 and League of legends are playable with wine so it can be usefull 
to have those profiles under linux.

-- 
Mathieu OKUYAMA

terrycain, 2016-01-16:
Nice thanks, I look into that

G33N, 2016-03-02:
Hello , I have the following problem when compiling in archlinux

## sh install_driver.sh

`=====================================================
::::::::::::::::\033[32m COMPILING librazer_chroma.so\033[0m ::::::::::::::
=====================================================
razer_chroma.c: In function ‘convert_pos_to_keycode’:
razer_chroma.c:575:47: warning: unused parameter ‘pos’ [-Wunused-parameter]
 void convert_pos_to_keycode(struct razer_pos *pos,int *keycode)
                                               ^
razer_chroma.c:575:56: warning: unused parameter ‘keycode’ [-Wunused-parameter]
 void convert_pos_to_keycode(struct razer_pos *pos,int *keycode)
                                                        ^
razer_chroma.c: In function ‘convert_ascii_to_pos’:
razer_chroma.c:751:3: warning: case label value exceeds maximum value for type
   case 280:
   ^
razer_chroma.c: In function ‘razer_update’:
razer_chroma.c:1236:17: warning: unused variable ‘tv’ [-Wunused-variable]
  struct timeval tv,select_tv;
                 ^
=====================================================
::::::::::::::::\033[32m COMPILING librazer_chroma.so [DEBUG]\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m COMPILING razer_bcd [SHARED]\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m COMPILING razer_bcd daemon\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m COMPILING razer_bcd daemon [DEBUG]\033[0m ::::::::::::::
=====================================================
In file included from razer_daemon.c:1:0:
razer_daemon.h:21:41: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_open)(struct razer_fx_render_node *render);
                                         ^
razer_daemon.h:21:41: warning: its scope is only this definition or declaration, which is probably not what you want
razer_daemon.h:22:42: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_close)(struct razer_fx_render_node *render);
                                          ^
razer_daemon.h:23:43: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_update)(struct razer_fx_render_node *render);
                                           ^
razer_daemon.h:24:46: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_key_event)(struct razer_fx_render_node *render);
                                              ^
razer_daemon.h:25:47: warning: ‘struct razer_fx_render_node’ declared inside parameter list
 typedef int (*razer_effect_dbus_event)(struct razer_fx_render_node *render);
                                               ^
razer_daemon.h:27:58: warning: ‘struct razer_daemon’ declared inside parameter list
 typedef struct razer_effect *(*razer_effect_init)(struct razer_daemon *daemon);
                                                          ^
razer_daemon.c: In function ‘dbus_handle_messages’:
razer_daemon.c:85:11: warning: passing argument 1 of ‘printf’ from incompatible pointer type [-Wincompatible-pointer-types]
    printf(stderr, "Out Of Memory!\n");
           ^
In file included from razer_daemon.h:4:0,
                 from razer_daemon.c:1:
/usr/include/stdio.h:362:12: note: expected ‘const char * restrict’ but argument is of type ‘struct _IO_FILE *’
 extern int printf (const char *__restrict __format, ...);
            ^
razer_daemon.c:117:11: warning: passing argument 1 of ‘printf’ from incompatible pointer type [-Wincompatible-pointer-types]
    printf(stderr, "Out Of Memory!\n");
           ^
In file included from razer_daemon.h:4:0,
                 from razer_daemon.c:1:
/usr/include/stdio.h:362:12: note: expected ‘const char * restrict’ but argument is of type ‘struct _IO_FILE *’
 extern int printf (const char *__restrict __format, ...);
[... 367 lines cut ...]
=====================================================
=====================================================
::::::::::::::::\033[32m COMPILING razer chroma kernel module\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m INSTALLING librazer_chroma.so\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m INSTALLING razer_bcd daemon\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m INSTALLING razer chroma kernel module\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m INSTALLING razer chroma udev rules\033[0m ::::::::::::::
=====================================================
=====================================================
::::::::::::::::\033[32m INSTALLING razer daemon init.d file\033[0m ::::::::::::::
=====================================================
cp: cannot stat 'init.d/razer_bcd': No such file or directory
Makefile:31: recipe for target 'install' failed
make: *** [install] Error 1
`

Could you help me?

terrycain, 2016-03-02:
Does arch use systemd for its boot up?