pez2001/razer_chroma_drivers #51: Cant close issues

terrycain, 2016-01-13:
@pez2001 Can I get permissions to close issues? Would be rather useful.
Also having the capability to submit issues other than bugs or beign able to assign people to issues etc... would be really helpful.

pez2001, 2016-01-13:
ive added you to the collaborators..hope thats correct.

im tryiing to get the firefly custom mode to work right now.  i pushed the changes.
it uses another effect for updates (0x0C instead of 0x0B) and has 15 leds in one row.
but im having problems receiving the right amount of pixel data from scrips (scripts/tests/set_firefly4.sh)
strange buffering that kicks in with 50 bytes packets (blackwidow seems to work fine).
maybe you have some idea.
bg 
tim

terrycain, 2016-01-13:
Hmm that's weird. Got some exams etc.. to do, after that I'll start working with my firefly so can look at that too.

pez2001, 2016-01-14:
ive written an additional example (raw_example) . works fine. the echo command will still fail. but i will move on... and leave this for another life.

pez2001, 2016-01-14:
renamed the repo.
updated the readme but cant push while working on multi device support

terrycain, 2016-01-14:
Ahh cool. Nice. Yeah, after tomorrow ill have some ram free so can start looking at the firefly. Also want to finish up my mamba driver before I move onto the firefly but I have most of the Mambas protocol documented so that shouldn't take too long.

terrycain, 2016-01-16:
@pez2001 what IDE do you use for the C / Kernel driver development? Or are you one of those *hardcore* noobs who is vim all the way?

pez2001, 2016-01-17:
no vim ;-) no emacs. not that leet. im using sublime at the moment but im thinking about switching to atom.

terrycain, 2016-01-17:
+1 for no vim. I've worked with the vim elitists before and frankly they normally are annoying :-). When I can be bothered I'm going to set up eclipse for development in C/C++ as that should be nicer.

pez2001, 2016-01-19:
since i pretty much finished the multi device support ,im searching for something new to add.
bindings?
valgrind search for leaks?
device event file(automatic notifications for adding/removing ofdevices)?
automatic builds/deploys?
completing the event system?
debug/finish json outputs?
automatic adding of input devices (mice,keyboards,joysticks)?
support for other led devices(maybe even a non razer one?)?

terrycain, 2016-01-19:
I'm still thinking a redesign of the daemon would do some good. Would easily allow extensions then like other devices (especially non razer ones if we build it modular enough).

Valgrind'ing would probably be good. Valgrind'ing the kernel module would be better though ;), I know valgrind can't be used but I'm sure there must be a way.

terrycain, 2016-01-20:
Don't suppose you have any idea how to get that damned macro led to light up?, It's been bothering me for a while now. Wireshark is not being too helpful this time.

closed 2016-05-20 by terrycain