pez2001/razer_chroma_drivers #52: rm: cannot remove ‘/razer-chroma*.deb’: No such file or directory

kamhagh, 2016-01-13:
I was trying to install this and i got this on the ./package_for_ubuntu.sh step

rm: cannot remove ‘/razer-chroma*.deb’: No such file or directory
dpkg-deb: building package `razer-chroma-driver' in `/tmp/tmp.k6ycKYsRpC.deb'.
dpkg-name: warning: cannot move '/tmp/tmp.k6ycKYsRpC.deb' to existing file

thanks for your work btw :)

kamhagh, 2016-01-13:
Also, in the README.md in the part :Installation for Debian/Ubuntu based distros (creating a .deb package)

you forgot the R after driver :D

pez2001, 2016-01-13:
i pushed an update. fixed ?

kamhagh, 2016-01-13:
I actually tried the other method, that was a fast answer :D do you want me to reinstall it so i can tell you :?

and yours better then synapse :D I like it! just the customs is grey for now... let me see documentations :D

pez2001, 2016-01-13:
i use the german layout and the author of the profile app (dynamic.c) maybe too i guess.

terrycain, 2016-01-13:
@kamhagh Try again please. If its fixed can then mark this as closed.

kamhagh, 2016-01-14:
I did install without .deb, There's nothing on the README.md about uninstalling it 
I'm not sure how i should do that :|

terrycain, 2016-01-14:
Try `sudo make uninstall`

closed 2016-04-08 by terrycain