pez2001/razer_chroma_drivers #53: "razer_drv" compile fails when directory path has spaces.

lah7, 2016-01-13:
Running Ubuntu MATE 15.10 (amd64) here, I've noticed that `make -s all` will fail to compile when the path to my working directory contains spaces.

This is the last few lines leading up to the error:
<pre>:: COMPILING snake game
======================
 
 :: COMPILING dynamic
======================

:: COMPILING razer kernel modules
========================================
Makefile:9: recipe for target 'razer_drv' failed
make: *** [razer_drv] Error 2</pre>

The directory is `/media/Data/Programming/Open Source/BlackWidow Chroma Driver/razer_blackwidow_chroma_driver` - Note spaces in the path.

If I compile from `/tmp/razer_blackwidow_chroma_driver`, no errors occur.

terrycain, 2016-01-13:
Make doesn't handle spaces well at all. It would require some major hack to fix that which would just fill the Makefile with rubbish. Just don't have spaces in the path.

lah7, 2016-01-13:
I see, we ought to note that in the README to avoid anyone else getting the same error.

terrycain, 2016-01-13:
You can create a Wiki page ;-). I feel we should have one for troubleshooting common issues. And possibly one for guidelines for submitting bugs, so we don't end up with people saying "It doesn't work"

closed 2016-01-13 by terrycain

Yasgur99, 2016-04-06:
I am in a path that does not have spaces, yet I get the same error

terrycain, 2016-04-07:
Do you have the correct kernel headers installed?

Yasgur99, 2016-04-07:
how do I tell?? @terrycain

terrycain, 2016-04-07:
What distribution are you using?

Yasgur99, 2016-11-15:
ubuntu 16.04.1

terrycain, 2016-11-15:
You using code from my fork or from this repo?