pez2001/razer_chroma_drivers #56: Still searching for a solution for issue #26

tnais, 2016-01-14:
Issued #26 has been closed without any solution or suggestion to identify the cause of the problem.

The problem is still there:

ls: cannot access /sys/bus/hid/drivers/razerkbd: No such file or directory

That means that it is not possible to control the lights or activate the M keys.

rajaksc, 2016-01-17:
Saw the same error logged here on Ubuntu 14.04 when following the "Installation for Debian/Ubuntu based distros (creating a .deb package)" install steps. 
```
# tail /var/log/upstart/razer_bcd.log
/proc/self/fd/9: 20: /proc/self/fd/9: cannot create /sys/bus/hid/drivers/razerkbd/bind: Directory nonexistent
Binding device (*:1532:020[39]*) to hid-generic
```

Fix:  A reboot was required after installing the .deb package. After a reboot, everything worked well. Might be worth adding this to the install steps just in case people forget (it's mentioned in the instructions for the all the install methods besides this package creation one).

terrycain, 2016-01-17:
@tnais `lsmod`

@rajaksc running `modprobe razerkbd` should of fixed that. Also the dkms step should of loaded the driver. I'll investigate.

tnais, 2016-04-05:
New machine, new install, works like a charm.

Thank you vewry much for the time you spent with my issue and for the driver!

closed 2016-04-05 by tnais