pez2001/razer_chroma_drivers #58: Profiles + effects

ghetolay, 2016-01-15:
Is it possible to have effects to work with profiles ? 

1) Is it possible through configuration ?
2) Could it be added to the library ? (I could help)
3) Impossible because of the firmware of some other reason ?

Thks a lot for those drivers (razer should thks you too because this is the reason I bought this keyboard)

terrycain, 2016-01-15:
Currently the only way to do this is by creating a library and utilising the effects part of the daemon. Obviously once it works in the daemon, we could add support for things like that in the app.

pez2001, 2016-01-16:
ive just added a 'display profile' effect. so it is actually now possible  :-)

terrycain, 2016-01-16:
Nice any examples? I'm working on adding logging to the daemon so probably wont commit for a while.

pez2001, 2016-01-16:
using the daemon_controller:

-C 11 "test" test"
-S 2 0 -1 "~/.config/razer_chroma/profiles/test"
-A 2
-b 2

pez2001, 2016-01-16:
im adding a parameter_changed event, so the -A 2 step is not needed in the near future

closed 2016-05-22 by terrycain