pez2001/razer_chroma_drivers #69: Razer Blackwidow Ultimate Stealth 2016 support?

minneyar, 2016-01-20:
I noticed that on the main page it says support for the Blackwidow Ultimate 2016 is in progress, so I'm not sure how much to expect; I just got an Ultimate Stealth 2016 keyboard and installed the driver, and when I try to start it, /var/log/upstart/razer_bcd.log contains:

```
Unknown device, --name=, --path=, or absolute path in /dev/ or /sys expected.
Unknown device, --name=, --path=, or absolute path in /dev/ or /sys expected.
Binding device (*:1532:020[39]*) to hid-generic
```

At a glance, I can see that in 95-razerkbd.rules, it does not have a line that matches my keyboard; the product ID according to lsusb is 0214.

terrycain, 2016-01-20:
Yeah I have access to my mates Ultimate 2016, Your saying the stealth's product ID is `0214`?

It'll have to be added to the driver, plus I need to double check the commands used in the driver, as the Ultimate 2016 has individually accessible LED's but they are not RGB they are only green.

Give me a few days to make a proper push with it.

minneyar, 2016-01-20:
Right.  I actually decided to look into it a bit myself, and I got the driver and keyboard bindings to recognize the Ultimate Stealth 2016, although the razer_bcd daemon still exits immediately.  What I've done so far is in https://github.com/minneyar/razer_chroma_drivers/tree/ultimate_stealth_2016_support .

terrycain, 2016-01-20:
Yeah getting it to bind is trivial, what really needs to be done is to ensure that the numbers in the commands the driver sends are correct

melkor217, 2016-01-20:
Wow, i got the same error with my 1532:011a (it's probably blackwidow 2013).

melkor217, 2016-01-20:
`for device in /sys/bus/hid/devices/*:1532:020[39]*` in bash_keyboard_functions.sh

Well, readme told me 0x011a is supported.

minneyar, 2016-01-20:
Try it with the code in my branch, I think it will manage to bind your keyboard.

melkor217, 2016-01-20:
I've already figured out with binding my keyboard. Your branch works as well. But deamon refuses to work with by keyboard still :(

    Jan 20 22:55:17 dan-desktop razer_daemon[27665]: started in foreground
    Jan 20 22:55:17 dan-desktop razer_daemon[27665]: opening razer chroma
    Jan 20 22:55:17 dan-desktop razer_daemon[27665]: loading FX libraries
    Jan 20 22:55:17 dan-desktop razer_daemon[27665]: loaded 2 FX libraries
    Jan 20 22:55:17 dan-desktop razer_daemon[27665]: looking for chroma devices
    Jan 20 22:55:17 dan-desktop razer_daemon[27665]: could not find devices

terrycain, 2016-01-20:
Have fixed all of the binding issues, we sort of left the boot scripts alone for ages. Added support for firefly mamba etc..., In my pull request, I also added the starlight effect to the BW Ultimate 2016, though custom colours didn't want to work for me despite the fact that the protocol seemed identical.

The daemon isn't really designed for the non chroma devices as by default it tries to run its heatmap colour like effect which obviously would not work. If we ever rebuild the daemon it should work better.

melkor217, 2016-01-21:
Wow, bw2013 driver actually works.

minneyar, 2016-01-21:
When I build from @terrycain's fork, the driver loads and I can bind my keyboard.  When I try to launch the razer_bcd daemon, it immediately exits with "No chroma devices found!", and without that running the GUI apps refuse to launch.  It would be kind of nice if, even if there aren't any devices, it would still run in the background and watch for them.

I can successfully echo values to the mode_ nodes under /sys/bus/hid/drivers/razerkbd/ to change the lighting mode, though!

minneyar, 2016-01-21:
After looking into it a little, getting the daemon to work was as easy as adding the product ID to the right place in the razer_chroma lib.  With that in place, the daemon runs and I can use the GUI and tray applet to control my keyboard.

terrycain, 2016-01-21:
Created a pull request. *Should* be good with the BW Ultimates then.

terrycain, 2016-04-05:
Anything left outstanding or should I close this issue?

minneyar, 2016-04-05:
I've been using it for a while now and it's working fine for me.

terrycain, 2016-04-06:
Cool, shall mark this as closed. Glad its working, will take another look at individual key lighting once I get hold of my mates ultimate2016 again

closed 2016-04-06 by terrycain

UriaMorP, 2017-08-15:
Sorry for the late question: I didn't understand - Do you have support for the **Razer Blackwidow Ultimate Stealth 2016**?

lah7, 2017-08-15:
It is supported, please see the highly successful "fork" of this project:

https://terrycain.github.io/razer-drivers/