pez2001/razer_chroma_drivers #73: Help with troubleshooting (Linux Novice)

UltraHex, 2016-01-21:
So I followed the directions for debian and when I execute the command `sudo systemctl start razer_bcd` it says
```
Job for razer_bcd.service failed. See 'systemctl status razer_bcd.service' and 'journalctl -xn' for details.
```

I did that and this is what I got:
[Pastebin](http://pastebin.com/LpAjPfJp)

I also looked in `/sys/bus/hid/devices/`
```
0003:046D:C283.0002  0003:1532:003E.0004  0003:1532:011A.0009  0003:1532:011A.000B
0003:1532:003E.0003  0003:1532:003E.0005  0003:1532:011A.000A  0003:1532:0504.0001
```
nothing that matches the pattern the script is looking for.

Any assistance you can give will be greatly appreciated.

melkor217, 2016-01-21:
Seems blackwidow2013 (1532:011A) is not supported atm. See #69.

Same story with ur naga (1532:003E).

UltraHex, 2016-01-21:
shame

closed 2016-01-21 by UltraHex

terrycain, 2016-01-21:
Whilst I'd like to add support for the Naga, it's going to be mighty difficult without having

terrycain, 2016-01-21:
Whilst I'd like to add support for the Naga, it's going to be mighty difficult without having one. If you had a mamba you'd be good ;). I have a Bw2013, it does work with the driver just not the service, as well the 2013 cant exactly do anything apart from the breathing effect and static mode.