pez2001/razer_chroma_drivers #75: ISO Keyboard not fully supported.

igorbb, 2016-01-24:
Profile Edition does not have ISO profiles. 
Thus It is not possible do edit RGB profiles for some keys.

terrycain, 2016-01-24:
Output of `setxkbmap -query` and a picture of the keyboard (preferably a stock one from the internet if possible with the correct keys)

igorbb, 2016-01-24:
setxkbmap -query
rules:      evdev
model:      pc105
layout:     gb,us
variant:    ,

----

The correct keys are this ( not raze kyb)
http://s23.postimg.org/pmdjqczl7/filco_iso.png

---
The closest I could ind for The Chroma was this

https://www02.cp-static.com/objects/high_pic/c/c02/1349373811_game-assecoires-razer-blackwidow-chroma-gaming-keyboard-azerty-rz03-01220400-r3f1.jpg

terrycain, 2016-01-24:
That keyboard map does exist as it appears for me. What layout is it giving you?

Also what distro + version you running

igorbb, 2016-01-24:
Ubuntu 14.04 LTS

My layout is actually Norwegian. But I use the UK or US layout.
Any other way I can help debug ? Do you want the key codes of the specific keys ?

It is funny because they do turn on on the spectrum or wave modes.

terrycain, 2016-01-24:
Screenshot of the profile app thingy

igorbb, 2016-01-24:
I think you are right. After reboot it detects as ISO kybd.. Maybe we should recommend a reboot after installation.  

I will close this issue then.

closed 2016-01-24 by igorbb

terrycain, 2016-01-24:
Yes, will add that to the instructions