pez2001/razer_chroma_drivers #77: razer_bcd service is not running in my system

Grefusa, 2016-01-28:
Hello, I have linux mint 17.3 and I try to install the driver in my system. I have a few problems.

1- I don't have the .desktop app in my system. I have to go manualy to /usr/share/razer_chroma_controller to initiate the app.

2- The second thing when I use the app it says that.
[image]

I have razer mamba 2012.

lah7, 2016-01-28:
1) I presume this is the Cinnamon desktop edition? The `.desktop` probably wants editing, I had to do the same for the MATE desktop.

(I think we ought to remove the "`OnlyShowIn=Unity;MATE;`" entry)

------------
2) What do you see when you type:
`sudo service razer_bcd start`

Grefusa, 2016-01-28:
Hello, yes I'm using the Cinnamon desktop edition.

sudo service razer_bcd start
razer_bcd start/post-stop, process 12904

terrycain, 2016-01-28:
Also the mamba isn't fully integrated into the daemon. You'll have to use the raw driver files for that

incryptx, 2016-02-24:
Does someone have a fix for this?

I get the graphic shown above, with the D-Dbus with the razer_bcd error?

I also get the application icons and clicking on the razer taskbar icon does nothing.

I blew up my system following some instructions about root level permissions for D-Bus and how easy it is to muck it up, which I apparently did.

Also,  when "making" this, I get some "system violation" about poorly written code?

What's that about?

A fix or solution, would be nice.

New to all of this.

Thanks

terrycain, 2016-02-24:
@incryptx 
> Also, when "making" this, I get some "system violation" about poorly written code?

Examples of this would help.

incryptx, 2016-02-24:
It's some dialog presented by Ubuntu 14.04 in a GUI Window referencing 
the package would not work, and giving me two buttons, "Continue Anyway" 
or "Cancel". I think it may be from the repository?

terrycain, 2016-02-24:
I don't think anything in this repo would bring up such dialog. If you could get a screenshot that would help more.

incryptx, 2016-02-25:
I'll try!

incryptx, 2016-02-25:
Ok,

In following your instructions from the readme, there is a "typo" in the 
line which changes directories, and ends in the word "drive".

This word should be "driver" and those who cut & paste the line will get 
an error, path not found.

 1.

    |git clone --depth=1
    https://github.com/pez2001/razer_blackwidow_chroma_driver.git cd
    razer_blackwidow_chroma_*drive"r"* |

 2.

    Install the packages needed to build the software:

    |sudo apt-get install -y dpkg-dev libdbus-1-dev jq libsdl2-dev
    libsdl2-image-dev libfftw3-dev |

 3.

    Build the software and driver:

    |make |

 4.

    Build the package:

    |./package_for_ubuntu.sh OR ./package_for_debian.sh|

After I built the .deb file and installed it with ./ubuntu.sh, the 
warning comes from the "Ubuntu System OS" warning of "bad code" which 
can break the system, and warning not to install this. I ignored the 
warning and installed anyway, and now on backups, I get warnings that 
DBus could not be backed up.

I think the error might be associated or repo-ed by building the .deb 
and installing with Ubuntu.sh on Ubuntu (which is based on debian)/

I wonder if I installed the .deb with the ./debian.sh if I would still 
get the warning, or if it would allow it to work?

I will attempt to take a screenshot (but as I just came from Windows and 
this stuff is new), I will need to figure out how first, and then remove 
the package and attempt to reinstall it.

Thanks for the assist here.

terrycain, 2016-02-25:
How did you install the deb?

incryptx, 2016-02-25:
Through USC, by right clicking on the .deb

I was going to try both processes today, installing from the .deb and 
installing pure debian and see if I could capture some screens?

Right now, I have a problem with the build tools on the same machine, 
which this install (via DBus) blew the OS and had to be rebuilt. Now 
rebuilt, I am reloading build tools and one is balking.

Depends: libsdl2-2.0-0 (= 2.0.2+dfsg1-3ubuntu1.1)

Something about wayland and mesa i386 binaries. Attempting to resolve 
the issues now.

Although I run a AMD64, most of the tools and binaries for this and 
steam, seem to require the i386 set to get anything done.

I was a SDET for MS on NT for ten years. Although coding is second 
nature, it's been twenty years since I looked at Linux.

It's matured now. Back then it was childish and confusing.

I will take some screenshots and get back to you.

Thanks for the assist. I just don't understand why HW makers like 
Logitech, Saitek, Razor, and others don't release Linux versions of 
drivers just to keep there options open in the Linux community, should 
the market place open, they would be set. It's also just a neighborly 
thing to do.

terrycain, 2016-02-25:
Not a clue, would be so much easier if razer provided a driver, and in future don't use software centre, its a pile of rubbish. Just `sudo dpkg -i file.deb` and then `sudo apt-get install -f`

incryptx, 2016-02-25:
Attached is a screenshot recieved after attempting to install the .deb

It is a warning thrown by the Software Center.

incryptx, 2016-02-25:
Here are the details.

I did build and use the .deb on a user called /susan and saved the .deb

Then used it again on the same machine, but renamed the user /bill when 
I installed the .deb

I see the permissions problem, and can rebuild the OS with that name to 
test it again by rebuilding the .deb

BUT, this same error happened on that machine, with the proper user 
name, and still gave the "bad code" pop-up.

Details:
Lintian check results for 
/media/bill/C8CD-ECD4/razer-chroma-driver_1.0.0_amd64.deb:
E: razer-chroma-driver: control-file-has-bad-owner postinst susan/susan 
!= root/root
E: razer-chroma-driver: control-file-has-bad-owner postrm susan/susan != 
root/root
E: razer-chroma-driver: control-file-has-bad-owner prerm susan/susan != 
root/root
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/dbus-1/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/dbus-1/system.d/ 
1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
etc/dbus-1/system.d/org.voyagerproject.razer.daemon.conf 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/init.d/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/init.d/razer_bcd 
1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/init/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
etc/init/razer_bcd.conf 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/udev/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/udev/rules.d/ 
1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
etc/udev/rules.d/95-razerkbd.rules 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/xdg/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid etc/xdg/autostart/ 
1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
etc/xdg/autostart/razer_tray_applet.desktop 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid usr/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid usr/bin/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/bin/razer_bcd_controller 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid usr/lib/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/lib/librazer_chroma.so 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/lib/librazer_chroma_controller.so 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid usr/lib/python3/ 
1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/lib/python3/dist-packages/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/lib/python3/dist-packages/razer/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/lib/python3/dist-packages/razer/__init__.py 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/lib/python3/dist-packages/razer/daemon_dbus.py 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/lib/python3/dist-packages/razer/keyboard.py 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid usr/sbin/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid usr/sbin/razer_bcd 
1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid usr/share/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
[... 104 lines cut ...]
usr/src/razer_chroma_driver-1.0.0/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/Makefile 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/dkms.conf 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/ 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/Makefile 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/razercommon.c 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/razercommon.h 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.c 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.h 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/razerkbd_driver.c 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/razerkbd_driver.h 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.c 1000/1000
E: razer-chroma-driver: wrong-file-owner-uid-or-gid 
usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.h 1000/1000

incryptx, 2016-02-25:
It's not my place to say, being the lesser knowledgeable one here, but 
some "dialog" in the .deb file which warns the user, "Hey, Your not 
Susan" or "Your not Root, either" or "Dummy, you can't copy the .deb to 
another machine" would be useful in warning us less fortunate of the 
errors of our ways.

Just a thought.

incryptx, 2016-02-25:
I'm sure I'm making mistakes you never thought a user would do, so I will try your suggestions, rebuild the .deb on the right machine, log in as root and use your non-SC commands and see where I am.

I will post those results when I have them.

Has anyone other than you been able to use this?

I ask, because sometimes devs have tools and dependancies for build processes they don't consider when the final product works in their build environment. As well, you only have us dumb end users for beta test, and the first OP has lost interest and moved on. Let's work it out so this works for everyone. My next project, after the mouse runs, is getting my Saitek X52Pro builds to work. They are just GUI front ends and don't mess with the HID layer.

lah7, 2016-02-25:
> Attached is a screenshot recieved after attempting to install the .deb
Please could you re-add the screenshot to your previous page? GitHub doesn't accept e-mail attachments, so we haven't seen the screenshot.

From the sounds of things, that's the Ubuntu Software Centre producing the message, possibly because of how its packaged - it's not a sophisticated Debian package after all.

> E: razer-chroma-driver: wrong-file-owner-uid-or-gid
Just to clarify, is your project directory on a Unix filesystem? (Not one like NTFS, as it appears you have media mounted?). Only root (`sudo` for Debian/Ubuntu) is needed to install the package.

> Has anyone other than you been able to use this?
Works perfectly fine in Ubuntu 15.10.

incryptx, 2016-02-25:
If I may suggest:

These should be in the "readme" if this is the preferred method of 
install for uneducated users.

You know the SC is crap, but the rest of us are stuck with it.

|sudo dpkg -i file.deb|
|sudo apt-get install -f

|
On 02/25/2016 08:45 AM, Terry Cain wrote:
> |sudo dpkg -i file.deb| and then |sudo apt-get install -f|

incryptx, 2016-02-25:
Bad Code Message

incryptx, 2016-02-25:
The .deb file was on a USB stick, probably formatted FAT32 but not NTFS 
and run from there.

I guess I was dumb again and should have moved it to a local drive 
first. Opps. But that did not cause th bad code error, as when I first 
built the .deb file, I was /susan and I was /root.

The GUID error is most likely the Razor Mamba not being connected at the 
time the .deb was run and install took place. Just a guess on my part, 
but again, when first attempted, the mouse was connected and nothing has 
changed. I still get the GUI for the Razer Configuration with the DBus 
error first reported by the OP.

Maybe some error checking in the scripts would help. i.e., is the mouse 
present? Is the user root? Is the user actually the user?

(This has always, throughout the existence of Linux, been it's 
deficiency. Granted, the scripts are very simple, but unintelligent and 
uninformative as well)

terrycain, 2016-02-25:
Yes lintian moans about the package, that is irrelevant (just a lot of spam, but should be fixed eventually).

Give me a few months, am not touching the code base until my year at uni is finished but last time I checked the deb installed on both Ubuntu 14.04 and 15.04, using the `dpkg` and `apt` commands, software centre is too dumb to handle this.

incryptx, 2016-02-25:
Ok. Ah, what is was to be a student.

incryptx, 2016-02-26:
Terry

I have a clean install, no errors.

After looking at the graphic the OP posted, I think now the only ting 
lacking or in error, is the "razer_bcd" service is not starting/started.

Because this is v14.04, as explained in the readme, I have an "upstart" 
system as opposed to a "systemd" service system. The command "sudo 
status razer_bcd" always says "razer_bcd stop/waiting" and does not 
change even after rebooting.

I think the service is just stuck somehow. I read one article about the 
service being "masked" somehow?

Any ideas on starting the service?

And if I get it to start, how can I insure it starts at bootup?

Thanks for the assist.

incryptx, 2016-02-26:
Hello Terry

If I do "status razer_bcd" it returns: unknown job.

Upstart states this means the job or service has not been converted to 
upstart. A converted service would typically output the current status, 
i.e. Starting, Stopping, and a process ID. A non-converted service would 
give an error about an unknown job.

Your readme states you included an "upstart" equivilent script or 
converted service, but I don't think it is loading or active.

I think this is the reason the razer_bcd service will not start, which 
leads to the dialog error when starting the Razer Chroma Application.

The init script is not running or starting. From here, I don't know what 
to do or how to fix it.

Thanks for the assist.

terrycain, 2016-02-26:
@incryptx run `status razer_bcd` as root. Ubuntu's upstart gets very funny about who runs the `status` and `service` commands.

Could you put the `/var/log/upstart/razer_bcd.log` on pastebin. If its converted it to a .some_number.gz file then uncompress (or `lesspipe`) and put on pastebin.

nekkz, 2016-03-01:
Same problem with Ubuntu 12.04 (elementary OS freya), using razer mamba tournament edition.

dmesg dump, after a `sudo service razer_bcd start`

`[  585.228859] init: razer_bcd main process (4396) terminated with status 1
[  585.228873] init: razer_bcd main process ended, respawning
[  585.258179] init: razer_bcd main process (4425) terminated with status 1
[  585.258192] init: razer_bcd main process ended, respawning
[  585.278786] init: razer_bcd main process (4448) terminated with status 1
[  585.278798] init: razer_bcd main process ended, respawning
[  585.290727] init: razer_bcd main process (4471) terminated with status 1
[  585.290745] init: razer_bcd main process ended, respawning
[  585.301776] init: razer_bcd main process (4494) terminated with status 1
[  585.301784] init: razer_bcd main process ended, respawning
[  585.312930] init: razer_bcd main process (4517) terminated with status 1
[  585.312943] init: razer_bcd main process ended, respawning
[  585.325580] init: razer_bcd main process (4540) terminated with status 1
[  585.325589] init: razer_bcd main process ended, respawning
[  585.345174] init: razer_bcd main process (4563) terminated with status 1
[  585.345186] init: razer_bcd main process ended, respawning
[  585.358303] init: razer_bcd main process (4586) terminated with status 1
[  585.358317] init: razer_bcd main process ended, respawning
[  585.370537] init: razer_bcd main process (4609) terminated with status 1
[  585.370544] init: razer_bcd main process ended, respawning
[  585.381221] init: razer_bcd main process (4632) terminated with status 1
[  585.381228] init: razer_bcd respawning too fast, stopped
`

terrycain, 2016-03-02:
If I remember correctly the mamba may or may not be added to the daemon.

terrycain, 2016-04-08:
The mamba's been added to the daemon now. Anyone still having problems

apol1, 2016-06-12:
I am having trouble as well, Ive installed the Chroma Linux Drivers from the website here

https://pez2001.github.io/razer_chroma_drivers/

Everything went smoothly as far as I can tell (new to linux)

Then I attempted to install Polychromatic, from both the software center and command line, here

https://github.com/lah7/polychromatic/releases/tag/v0.1.0

I get the exact same error message as the screen shot and when installing in command line I get this message

The razer-chroma-drivers Python modules are missing!
Please install them from http://pez2001.github.io/razer_chroma_drivers/
It could be possible that they are in a different path to '/usr/lib/python3/dist-packages/razer/

apol1, 2016-06-12:
I'm having trouble as well, I first installed the Chroma Linux Drivers from the website here:

https://pez2001.github.io/razer_chroma_drivers/

I followed the instructions for Ubuntu and Debian

Everything seemed to work as far as I can tell (new to Linux)

I then installed PolyChromatic from here:

https://github.com/lah7/polychromatic/releases/tag/v0.1.0

However in command line after entering ./install/install.sh I received this message: 

The razer-chroma-drivers Python modules are missing!
Please install them from http://pez2001.github.io/razer_chroma_drivers/
It could be possible that they are in a different path to '/usr/lib/python3/dist-packages/razer/'
*************************
Continue anyway? [y]

I tried running the script as root by adding sudo, same message appeared 

I still see the same error message as the screen shot above

Im running Linux Ubuntu 16.04 LTS

lah7, 2016-06-12:
It sounds like the service isn't running or didn't auto-start. What happens when you run:

    sudo service razer_bcd restart

---------
> The razer-chroma-drivers Python modules are missing!
> It could be possible that they are in a different path to '/usr/lib/python3/dist-packages/razer/'

I just realised `v0.1.0` has a typo in the install script, so it will keep saying that. :-1:  [I've released a new version of Polychromatic.](https://github.com/lah7/polychromatic#installation), so everyone will be on the same page again. :+1:

apol1, 2016-06-13:
~$ sudo service razer_bcd restart
[sudo] password for apollo: 
Job for razer_bcd.service failed because a configured resource limit was exceeded. See "systemctl status razer_bcd.service" and "journalctl -xe" for details.
~$

apol1, 2016-06-15:
Ive installed the newest version of Polychromatic, the screen looked glorious until...
[image]

terrycain, 2016-06-16:
What device? What's the output of journelctl -xe

apol1, 2016-06-16:
Im unsure what you mean by device, for hardware Ive not bought any razer ware, Im hoping to make the application functional first if possible. Heres a screenshot :)
[image]

terrycain, 2016-06-16:
Yeah the current daemon won't work without a device. My new version will work without but it's not finished yet

apol1, 2016-06-17:
Wonderful, I really like the sound of that, I'll keep an eye out on this thread for when it comes out.

terrycain, 2016-06-17:
No problem, am currently moving house so development is well not happening at the moment but I aim to be back finishng it by the end of the month

tarpediem, 2016-06-19:
I have a razer mamba tournament edition, it works great on my Ubuntu 16.04 until I install the chroma drivers ... as soon as I do a sudo modprobe razermouse it stops working and the only way to get it working again is to completely purge the package.
My razer keyboard works fine though.

terrycain, 2016-06-19:
@tarpediem modprobe razermouse, then `ls -l /sys/bus/hid/drivers/razermouse/`

tarpediem, 2016-06-24:
```
total 0
--w------- 1 root root 4096 Jun 24 15:21 bind
lrwxrwxrwx 1 root root    0 Jun 24 15:21 module -> ../../../../module/razermouse
--w------- 1 root root 4096 Jun 24 15:21 new_id
--w------- 1 root root 4096 Jun 24 15:21 uevent
--w------- 1 root root 4096 Jun 24 15:21 unbind
```
and dmesg returns me this : 

`[  613.320024] usb 3-9.3: new full-speed USB device number 12 using xhci_hcd
[  613.425466] usb 3-9.3: New USB device found, idVendor=1532, idProduct=0046
[  613.425471] usb 3-9.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  613.425474] usb 3-9.3: Product: Razer Mamba Tournament Edition
[  613.425476] usb 3-9.3: Manufacturer: Razer
[  613.426994] input: Razer Razer Mamba Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.3/3-9.3:1.0/0003:1532:0046.0015/input/input30
[  613.427224] hid-generic 0003:1532:0046.0015: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Mamba Tournament Edition] on usb-0000:00:14.0-9.3/input0
[  613.428940] input: Razer Razer Mamba Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.3/3-9.3:1.1/0003:1532:0046.0016/input/input31
[  613.536207] hid-generic 0003:1532:0046.0016: input,hidraw6: USB HID v1.11 Keyboard [Razer Razer Mamba Tournament Edition] on usb-0000:00:14.0-9.3/input1
[  613.536992] input: Razer Razer Mamba Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.3/3-9.3:1.2/0003:1532:0046.0017/input/input32
[  613.592426] hid-generic 0003:1532:0046.0017: input,hidraw9: USB HID v1.11 Keyboard [Razer Razer Mamba Tournament Edition] on usb-0000:00:14.0-9.3/input2
`

terrycain, 2016-06-24:
Tournament edition not supported yet. Should be an easy fix. Am currently moving house atm but will look into this when I'm all settled.

RetroEggy, 2016-08-13:
Hey guys,
I have the same issue with razer_bcd not starting. I'm using a Razer Blackwidow X Chroma and compiled the corresponding branch, I also unbinded the keyboard from hid generic and binded it to razer kbd. Starting Polychromatic results in the same message as seen in the first post. Starting razer_bcd always outputs  start/post-stop and a stop/waiting status. dmesg gives the same output as the previous users had : razer_bcd respawning too fast, stopped. Help would be greatly appreciated:)

GabMus, 2016-08-14:
This repo is legacy, please install the driver from terrycain's fork:
https://github.com/terrycain/razer_drivers

GabMus, 2016-08-14:
As for the frontend I don't know the state of polychromatic, but my
frontend razerCommander works fine https://github.com/gabmus/razercommander

lah7, 2016-08-14:
Polychromatic still uses the original driver and daemon provided by this repository, not **daemonv2** from the fork yet. So, once the service works, Polychromatic will work too.

To me, it looks like [Razer BlackWidow X Chroma isn't supported yet](https://github.com/pez2001/razer_chroma_drivers/issues/141).

RetroEggy, 2016-08-14:
I tried using the razer_drivers you suggested, but I wasn't able to bind my Blackwidow X to razerkbd, that's why I switched to the Blackwidow X branch (https://github.com/terrycain/razer_drivers/tree/RazerBlackWidowXChroma_0x0216). Guess I'll have to wait until X support is integrated into the main branch then:/

terrycain, 2016-08-14:
Thats because people don't actually tell me when things work. I'll merge bwX into master in a bit

RetroEggy, 2016-08-14:
That would be awesome! Let's hope it will work then - after all, only binding the keyboard worked, but not razer bcd.

terrycain, 2016-08-14:
@RetroEggy yeah the daemon would have to edited and I cba for that (hence I made a new one). Have you by any chance used any features of the driver so I know what works and what doesnt

RetroEggy, 2016-08-14:
I binded the keyboard to razerkbd - that worked according to ls -l. I haven't tried anything else (well, except for trying to start razer_bcd) since my primary goal was simply to change the colours using Polychromatic.

terrycain, 2016-08-14:
If you can do it, go through this list and see what works "https://github.com/pez2001/razer_chroma_drivers/wiki/Using-the-keyboard-driver"

Skiping the mode_custom sections (as set_key_row will do that)

Also use the feature_pythonlib branch of my fork as that has some extra driver fixes

RetroEggy, 2016-08-14:
Okay, so those are the results using the Blackwidow X branch (I've compiled the driver 3 times already this weekend, I'll get to trying the feature_pythonlib branch some time later)
cat device_type
Razer BlackWidow X Chroma
macro_keys: I'm getting no error messages, so I assume it's alright.
mode_breath: All 3 examples working!
No effect: Working!
Reactive mode: Working!
Spectrum: Working!
Pulsate: As expected, not working since it's not a Blackwidow Ultimate 2013.
Wave: Works in both directions:D
Reset: Not sure about it. The lighting effect stays the same, but I'm not getting an error message either.
Brightness: No change, but also no error.
Set logo: Not found (as expected).
set key row: Working.
Clear row: Working.
Get serial: Shows me an empty row.

I should have tried that before, that's pretty impressive to me. I can write my custom layout myself this way without the need of a frontend:D

terrycain, 2016-08-14:
Ok repeat that with the pythonlib driver at some point, as it might fix some things but looks good. Also look in dmesg for any lines containing razerkbd

:-)

terrycain, 2016-08-14:
Also what about mode_static

RetroEggy, 2016-08-14:
Static doesn't work. dmesg outputs the following messages regarding razerkbd:
[21546.122641] Wrong Amount of RGB data provided: 1 of 67
[22387.211846] razerkbd: Unknown product ID '534'
[22402.888592] razerkbd: Unknown product ID '534'razerkbd: Unknown product ID '534'
[22447.499102] razerkbd: Unknown product ID '534'<1>[22724.597375] Wrong Amount of RGB data provided: 1 of 67
[22992.037788] razerkbd: Invalid Report Type. Start Marker: 01 id: ff Num Params: 16 Reserved: 00 Command: 82 Params: 000000000000 .
[24123.914257] razerkbd: Cannot set static mode for this device
Hope this helps:)

terrycain, 2016-08-14:
Yeah am afk atm. But should be able to fix that by end of Monday. In that case hold off trying the different driver

terrycain, 2016-08-14:
Try running get_serial a couple of times, see if it works. The unknown product id should go away with the driver in the pythonlib driver.

RetroEggy, 2016-08-14:
Getting the serial worked the second time I tried. Interesting. I'll try the other driver after your update.

terrycain, 2016-08-14:
Is the keyboard wireless by any chance?

[11 more replies]