pez2001/razer_chroma_drivers #81: Doesn't detect Razer Mamba TE

Vilsol, 2016-02-04:
It seems like it doesn't detect my Razer Mamba TE, if I use the python library and tell it to enumerate the devices, it only lists the firefly.

```
>>> import razer.daemon_dbus
>>> import razer.keyboard
>>> daemon = razer.daemon_dbus.DaemonInterface()
Successfully connected to dbus service.
>>> daemon.enumerate_devices()
Device ID 0
Device Type "Razer Firefly", Device Serial "ST1531135701269"
[{'index': 0, 'type': dbus.String('Razer Firefly'), 'serial': dbus.String('ST1531135701269')}]
```

Are there any log files I could provide? I checked ```/var/log/upstart/razer_bcd.log``` but the file doesn't seem to exist.

terrycain, 2016-02-04:
I'll look into this.

laxdragon, 2016-04-04:
Any updates on this. I can also help provide any needed information.

terrycain, 2016-04-04:
Sorry I've been busy with uni. Give me a couple of days and ill update the daemon and then if your up for testing it, it should be good.

terrycain, 2016-04-04:
@laxdragon Do you have a firefly, chroma and mamba? also what system you using?

laxdragon, 2016-04-04:
Yes, I have all 3 (Blackwidow Chroma, Firefly, and Mamba TE [wired]). I'm using Debian Jessie.

terrycain, 2016-04-04:
Does the udev script autobind the devices for you?

laxdragon, 2016-04-04:
The systemd startup fails on boot, but I'm able to issue a "service razer_bcd restart" and it stars successfully. But, is not longer able to be controlled via the config or tray applet since the patch.

terrycain, 2016-04-04:
Yeah the GUI and tray applet has no clue about multidevice yet. I take it, it doesn't bind the mamba te to the razermouse driver?

laxdragon, 2016-04-04:
doing the service restart also causes the keyboard and firefly to go green for about 2 seconds, then static rainbow. Mouse is stuck in static rainbow, never changes.

laxdragon, 2016-04-04:
does not look that way, there is no symlink to the usb id in /sys/bus/hid/drivers/razermouse

terrycain, 2016-04-04:
OK, that means the boot scripts are not aware of the TE as its probably only looking for `0x0045`. Gimme a minute and ill add that in that pull request.

terrycain, 2016-04-04:
Added some changes to the makefile, mind trying them out and seeing if that binds the razermouse, (bear in mind that if the service stops, systemd will run the unbind stuff which is annoying). If it binds successfully then it should make the mouse flash.

Also mind running the following, changing the path to one of the 3 mamba ones. It'll end in something like "0046.0001"
`udevadm info -q all -a /sys/bus/hid/devices/0003:1532:0045.0009`

laxdragon, 2016-04-04:
OK, that patch works, All three now bind.

jnewman@jnewman:~$ ls -l /sys/bus/hid/drivers/razer*/0003:1532*
lrwxrwxrwx 1 root root 0 Apr  4 13:53 /sys/bus/hid/drivers/razerfirefly/0003:1532:0C00.000C -> ../../../../devices/pci0000:00/0000:00:1c.3/0000:04:00.0/usb5/5-2/5-2.4/5-2.4:1.0/0003:1532:0C00.000C
lrwxrwxrwx 1 root root 0 Apr  4 13:53 /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:1532:0203.0003
lrwxrwxrwx 1 root root 0 Apr  4 13:53 /sys/bus/hid/drivers/razermouse/0003:1532:0046.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:1532:0046.0004

I put the log for udevadm here: http://pastebin.com/nEQdsAQ9

While all bind, I'm no longer able to control them either via the gui, which you explained, or via the echo commands to the device. For example
`echo -n -e "\xFF\x00\xFF" > /sys/bus/hid/drivers/razerkbd/0003\:1532\:0203.0003/mode_static`
does nothing.

terrycain, 2016-04-04:
Ahh is the daemon running?

terrycain, 2016-04-04:
Also udev looks good

laxdragon, 2016-04-04:
Yep, running.

`root@jnewman:/sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003# service razer_bcd status
● razer_bcd.service - Razer Keyboard Daemon
   Loaded: loaded (/lib/systemd/system/razer_bcd.service; enabled)
   Active: active (running) since Mon 2016-04-04 13:53:09 CDT; 12min ago
  Process: 10378 ExecStopPost=/usr/share/razer_bcd/systemd_helpers.sh unbind (code=exited, status=0/SUCCESS)
  Process: 10516 ExecStart=/usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid (code=exited, status=0/SUCCESS)
  Process: 10423 ExecStartPre=/usr/share/razer_bcd/systemd_helpers.sh bind (code=exited, status=0/SUCCESS)
 Main PID: 10517 (razer_bcd)
   CGroup: /system.slice/razer_bcd.service
           └─10517 /usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid

Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Unbinding device (0003:1532:0203.0003) from hid-generic
Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Binding device (0003:1532:0203.0003) to razerkbd
Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Bind Successful
Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Unbinding device (0003:1532:0046.0004) from hid-generic
Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Binding device (0003:1532:0046.0004) to razermouse
Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Bind Successful
Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Unbinding device (0003:1532:0C00.000C) from hid-generic
Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Binding device (0003:1532:0C00.000C) to razerfirefly
Apr 04 13:53:09 jnewman systemd_helpers.sh[10423]: Bind Successful
Apr 04 13:53:09 jnewman razer_bcd[10516]: Starting razer blackwidow chroma daemon as a daemon
root@jnewman:/sys/bus/hid/drivers/razerkbd/0003:1532:0203.0003# ps aux | grep razer
root     10517  1.8  0.0  24276  2024 ?        Ds   13:53   0:14 /usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid`

terrycain, 2016-04-04:
Ok I believe (dont quote me on this) that whilst the daemon is running it will be spamming custom colour mode on the keyboard. run `razer_bcd_controller -p` then try the echo

terrycain, 2016-04-04:
I think -p is pause

laxdragon, 2016-04-04:
OK, that did it. I can now control all three via the command line at least.

terrycain, 2016-04-04:
Yeah technically the echo was working but the daemon doesn't know you are doing this so it then still does custom colour so quickly it looks like echo isn't working. Using the DBus interface when doing the effects like static etc... will automatically pause the daemon.

I believe this issue is now solved for now. Now its just a matter of getting the gui and tray to do multi device.

closed 2016-04-04 by terrycain