pez2001/razer_chroma_drivers #82: Does it work for Razer Blade Stealth's chroma keyboard?

felipeolifre, 2016-02-09:


terrycain, 2016-02-10:
I don't know, it would of been helpful if you tried it first ;-)

What's the USB VID and PID for the keyboard?

felipeolifre, 2016-02-10:
Yes, but I can´t yet. 
Planning to buy the Stealth in the next months. Then, I would try and leave a comment (may be someone try it before).

About the USB VID/PID.. It's an ultrabook, and has chroma lights in the default keyboard. So, I think the inner connection is an FFC connector (or something like that).

terrycain, 2016-02-10:
ugh, hmm, I'm not sure. You might be lucky and it might still present itself as a USB device, if that's the case then its more than likely something can be done.

smevawala, 2016-02-23:
I think it still presents itself in lsusb.  Is this what you're looking for?: 
idVendor 0x1532 Razer USA, Ltd
idProduct 0x0205

The entire lsusb output for the keyboard is here ( in the spoiler): https://insider.razerzone.com/index.php?threads/razer-blade-stealth-linux.11334/page-2#post-207690

I unfortunately do not have a stealth yet, its on backorder, but I'm sure people on that forum would be willing to try it.

felipeolifre, 2016-02-23:
Thanks for the info @smevawala ! 

Look at the first line of the lsusb output: 
Bus 001 Device 004: ID 1532:0205 Razer USA, Ltd

The VID/PID for the device is 1532:0205.

I'm still waiting for my Stealth too.. but it looks like it gonna work.

terrycain, 2016-02-23:
It might take some modifications but its looking more promising.

smevawala, 2016-03-06:
I have a razer blade stealth running ubuntu if that helps.

terrycain, 2016-03-06:
Run `lsusb`, get me the VID/PID, think its 1532:0205. Can you get Windoze running in virtualbox?, we could then start attempting to collectively reverse engineer the protocol

smevawala, 2016-03-07:
yeah, it is 1532:0205.  I currently have my stealth dual booting with Ubuntu and windows 10.  Would that work?  Or do you need to have both OSes to be running at the same time?

terrycain, 2016-03-07:
We would need both running at the same time as we want to wireshark the usb
bus whilst the Windows vm is doing its thing

smevawala, 2016-03-07:
Ok, no problem, I'll set that up tonight.  Windows 7 and virtualbox.  What should I do once thats set up?

terrycain, 2016-03-07:
Install wireshark, go through a few tutorials in using it and then wait for me to eventually make a wiki page on what to do.

Also install the razer synapse thing and get virtualbox to pass through the usb device as then windows should be able to use it with the razer app.

This is all assuming the razer synapse app will acknowledge the stealth and work like the normal keyboards do, as its a laptop after all so things have a chance of not working at all.

terrycain, 2016-03-07:
Basically if you cant get the razer synapse app to control the keyboard there's nothing we can do. Also if you pass through a usb device to virtualbox it will have sole ownership over it so if you give it a keyboard I normally plug in an additional one to use on the host os

smevawala, 2016-03-08:
I'm not sure how to make virtualbox passthrough a keyboard completely.  Even ones I plug in, it detects a keyboard and uses it for both OSes.

lah7, 2016-03-08:
There's a USB icon at the bottom of the virtual machine's window like this:
[image]

If it says "No USB devices connected", you may need to install the [Oracle Extension Pack](https://www.virtualbox.org/wiki/Downloads) and add the group `vboxusers` to your user first, then log out and back in.

terrycain, 2016-03-08:
Have started to create a page with instructions on reverse engineering the USB protocol. [Wiki page](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol) its missing images which I shall add but feel free to read it.

smevawala, 2016-03-11:
Ok, sorry for the delay.  I got it working in a VM.  The stealth doesn't support windows 7, but windows 10 works.  I'll start reverse engineering asap.

smevawala, 2016-03-11:
So I was reverse engineering for a little bit and it seems the stealth's commands match the blackwidow chroma. (At least for the two commands shown on the wiki).  Switching from none to static green to wave right gave this output in wireshark:

Leftover Capture Data: 00ff00000002020600000000000000000000000000000000...
Leftover Capture Data: 00ff00000004030a0600ff00000000000000000000000000...
Leftover Capture Data: 00ff000000020e0401ff0000000000000000000000000000...
Leftover Capture Data: 00ff00000003030001050100000000000000000000000000...
Leftover Capture Data: 00ff00000002020600000000000000000000000000000000...
Leftover Capture Data: 00ff00000002030a01010000000000000000000000000000...
Leftover Capture Data: 00ff000000020e0401ff0000000000000000000000000000...

From that output, the relevant output seems to be:
00ff00000004030a0600ff00000000000000000000000000... (static green)
00ff00000002030a01010000000000000000000000000000...(wave right)

From these promising results, I think it may just be better to set up the stealth with the same functions as the blackwidow chroma and see what breaks.

terrycain, 2016-03-11:
ooooohhh its looking good. Ideally I would try and get captures of all of it, I have some stuff to go through the dumps if you send them to me. It makes sense that most would be similar to the blackwidow chroma but I'm sure there will be some that are different, also I think the stealth has extra features, I think I read somewhere you can set the fan speed.

terrycain, 2016-03-11:
Cheers for the help though, Our compatibility for razer devices is growing :smiley:

smevawala, 2016-03-11:
ok cool. What should I do in the dumps I send you?  Like how would you know what actions lead to what capture data?  Is there an order that would help? Trying to make this as easy as possible for you.

smevawala, 2016-03-11:
also, this is just an idea, have any of you approached Razer for support or just letting them know this exists?  With their new focus on openness and linux gaming expanding greatly, they may be willing to send you devices to support or help with the integration with chroma apps.  Razer has a this page on their site for developers to contact them about developer stuff: http://developer.razerzone.com/chroma/contact/
Just pointing it out

terrycain, 2016-03-11:
> have any of you approached Razer for support

Yeah, I emailed them, they were a new level of unhelpful. I told them we were making a Linux driver and could I get any information on the USB protocol etc... They responded with this, as you can see I was not impressed by this generic crap. I also believe that they wont willingly give up their USB technical docs.
> We appreciate your support and dedication to Razer. We often get requests to send out free products and swag from our customers. In order to be fair to all of our customers and due to the overwhelming requests we do get, we are unable to honor these requests

As for what you do. Could I get screenshots of all the options in the razer synapse app as I don't know what it looks like on the stealth. And the a dump per feature please. Oh and the Wireshark filter string you were using too, should contain the bus id and the device id.

E.g. If your going after the static colour effect
A dump containing going from None effect to static effect, and repeating that a few times, perhaps with a few colours. Then a text file saying this dump is static effect preferably containing what static colours was used.
E.g. If your after brightness
A dump containing  brightness going up and down, perhaps a few times. Text file saying that its brightness going up and down.

Then bung all the dumps, screenshots and text files into a zip and put it somewhere I can get.

Make sense?

smevawala, 2016-03-11:
sounds good

smevawala, 2016-03-14:
Also, about the support from razer, I tried contacting Min-Liang Tan and he responded enthusiastically.  Maybe its normal CEO talk, but he sent me an email asking me what information we needed in creating linux drivers.  I can add you guys in the emails if you're interested

terrycain, 2016-03-14:
CC in me at `terry@terrys-home.co.uk`. All we need is technical docs that detail the USB protocol each razer device uses, we can do the rest ourselves.

smevawala, 2016-03-29:
Ill send you some dumps and screenshots later this week.  Sorry, I've been busy.  Also, hopefully, this will also work with the new Razer blade announced.  It seems as though every new laptop razer releases will have a chroma keyboard.

centaur2048, 2016-03-29:
My Razer Blade Stealth is coming this Friday. Would be more than happy to help with anything I can, as well.

terrycain, 2016-03-29:
Me and @smevawala are currently attempting to negotiate with Razer some technical docs so I can whip out drivers for all but until this happens, if it happens your welcome to drop me some packet dumps if you wish. We might have some docs by then but I have a feeling these negotiations might drag on for a while but at least progress is being made so far.

terrycain, 2016-03-29:
@smevawala Wicked cheers. Hopefully we get docs that will help but even if we do get them they might not be that useful to us, but we'll see :wink:

storborg, 2016-03-31:
I tossed up a start to this in storborg/razer_chroma_drivers@cc294328bef6a2d5d8a228ecd94a716028af0c2b

Some pcaps are attached. On my device (128GB QHD variant purchased March 2016) the keyboard is on bus 1, device address 4. VID/PID are the same as above, 1532:0205.

[blade-stealth-chroma-pcaps.zip](https://github.com/pez2001/razer_chroma_drivers/files/198177/blade-stealth-chroma-pcaps.zip)

Some stuff works more or less as-is with a similar protocol to the Blackwidow Chroma or Ultimate 2016, including ``set_none.sh``, ``set_wave.sh``, ``set_spectrum.sh``, and ``set_breath.sh`` with no colors passed. Other stuff, including e.g. ``set_static.sh`` doesn't work: I haven't spent much time looking at why yet.

This is a pretty fun thing, thanks for building it!

terrycain, 2016-03-31:
Nice, from the packet dumps you gave me static mode should work and looks identical to the chroma. 

As for the logo try replacing the razer_test function with this

```
int razer_test(struct usb_device *usb_dev)
{
    int retval;
    struct razer_report report;
    razer_prepare_report(&report);
    report.parameter_bytes_num = 0x03;
    report.command = 0x00; /*init command id ?*/
    report.sub_command = 0x01;/*unknown*/
    report.command_parameters[0] = 0x04;
    report.command_parameters[1] = 0x00;
    report.crc = razer_calculate_crc(&report);
    retval = razer_send_report(usb_dev, &report);
    return retval;
}
```
Then `echo -n "1" > DRIVERFOLDER/test` this "should" turn off the logo, but make sure the logo is on first ;)

storborg, 2016-03-31:
Ah yeah, my mistake: all of the modes that have scripts for them work just fine. I was just using ``dash`` instead of ``bash`` before.

That command/subcommand works, and I've added the plumbing for that in storborg/razer_chroma_drivers@1338ed36fc80d388c5bf8c52eba75612be7371a3

Let me know if you see anything so far that needs to change... I'll clean up some of the supported/unsupported notes and do a bit more testing on all of this stuff before opening a PR, since I think there might be a few things that still don't work.

storborg, 2016-03-31:
By the way, I couldn't get the Lua dissector working right in Wireshark. Not sure if I'm doing something wrong there or it's actually broken. The decoded fields seem to be offset by a couple bytes and it throws this error:

```
Expert Info (Error/Undecoded): Lua Error: [string "/home/me/.config/wireshark/plugins/razer.l..."]:105: Range is out of bounds
```

terrycain, 2016-03-31:
Dont worry about the dissector for now, it was a quick example to see if I could actually get things to work, I actually find it easier not to use it for now :wink:

smevawala, 2016-04-01:
I can help test this.  I tried to yesterday, but I cant get razer_bcd to start.  Im on Linux Mint 17.3, installed through the packaged for ubuntu script.  I can send the logs later today, but the they mostly say the system cant find  the  razer_mouse and razer_firefly directories.  Have you guys seen that issue before?

terrycain, 2016-04-01:
@smevawala I don't think the daemon has been modified for this yet so it wouldn't work regardless. But attempt to use the driver manually. And regarding the can't find directories, yeah that was my fault, I have all 3 you see so it worked fine for me, after I finish Uni I'm going to go through and tidy up everything including fixing this and possibly restructuring the daemon.

terrycain, 2016-04-01:
@storborg Can you confirm that the Udev script automatically binds the stealth to the correct driver?

smevawala, 2016-04-01:
Awesome, I'll confirm the driver tonight.  I already have a zip of synapses screenshots.  I'll upload them when I get home

storborg, 2016-04-01:
@terrycain Yeah, the udev rules after my patches bind the stealth to the correct driver. Related, though, I'd love to figure out a way to allow specific group access to the razer sysfs files.

terrycain, 2016-04-01:
I think udev can do that but I'm not sure. I looked into this a while ago and couldn't see any way to do it in the module.

Worst case you could hack udev to run a command to `chown` it

smevawala, 2016-04-01:
I can confirm modes static, wave, breath, spectrum, reactive, and none all work!  Set_logo also works.  Really don't know how to test mode_game or mode_custom.

storborg, 2016-04-01:
Just to link it back to this ticket, some of this support is merged as of 30c15363f00c3f8a8f8b7c295465f289f309e46d from PR #103 

Still working on individual-key control...

terrycain, 2016-04-01:
If you `echo -n "1" > mode_game` it should disable the win key.

Will look into the individual control.

storborg, 2016-04-01:
That doesn't work at the moment. Here's some pcap dumps of enabling and disabling game mode:
[disable-gaming-mode.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200456/disable-gaming-mode.pcapng.gz)
[enable-gaming-mode.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200457/enable-gaming-mode.pcapng.gz)

And individual dumps for selectively enabling and disabling alt+tab and alt+f4 (in the Synapse app, the windows key is always disabled in game mode):
[gaming-mode-disable-alt-tab.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200461/gaming-mode-disable-alt-tab.pcapng.gz)
[gaming-mode-enable-alt-tab.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200462/gaming-mode-enable-alt-tab.pcapng.gz)
[gaming-mode-disable-alt-f4.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200463/gaming-mode-disable-alt-f4.pcapng.gz)
[gaming-mode-enable-alt-f4.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200465/gaming-mode-enable-alt-f4.pcapng.gz)

storborg, 2016-04-01:
I also noticed that Synapse lets you selectively remap individual keys (I don't know if this feature exists on other Razer devices?). That setting doesn't persist when reattaching the keyboard from Virtualbox back to the host, but maybe it just gets cleared on a USB reset. In any case, here are some dumps of the USB traffic to rebind the "S" key to a few different things.
[keyboard-set-s-to-a.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200470/keyboard-set-s-to-a.pcapng.gz)
[keyboard-set-s-disabled.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200471/keyboard-set-s-disabled.pcapng.gz)
[keyboard-set-s-to-default.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200472/keyboard-set-s-to-default.pcapng.gz)
[keyboard-set-s-to-left-click.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200473/keyboard-set-s-to-left-click.pcapng.gz)
[keyboard-set-s-to-volume-down.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200474/keyboard-set-s-to-volume-down.pcapng.gz)
[image]

terrycain, 2016-04-01:
Ok think i've got the fkeys done and brightness on AC.

The brightness might only have one function and the windows handles changing it when on battery and on AC.  Enable/Disable lighting on battery didn't have anything useful so try that one again but could also be controlled by windows.

Game mode wasn't useful either so try again. All of the keyboard ones also had nothing useful so try those again.

Think I've got the individual key lighting figured out but it requires some work as the USB commands depend on the row length of the keyboard which is less on the stealth. Could you get me a dump of you setting the keys alternating colours, like red,green,blue,red,green,blue... and a picture of the layout that should confirm that.

terrycain, 2016-04-01:
Ok guys would you mind testing out my pull request #105.

Test brightness `echo -n "0" > set_brigthness` `echo -n "100" > set_brigthness` `echo -n "255" > set_brigthness`

Test Fkeys so `echo -n "0" > set_fn_toggle` `echo -n "1" > set_fn_toggle`

Would you also re-try that python script keyboard_keys (once modified) and when its running just mash the enter key and it _should_ light up each key row by row, it might mess up a bit though as that function still needs a fair bit of work.

storborg, 2016-04-01:
I went through and set an individual key layout in the synapse configuration tool. It looks like this (note that the up arrow is set to 255,255,255 and the down arrow is set to 127,127,127):
[image]

Here's the pcap dump-- it looks like it might rewrite it over and over, as it keeps generating USB traffic for a while.
[set-individual-keys.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/200553/set-individual-keys.pcapng.gz)

terrycain, 2016-04-01:
Yeah the synapse app is inefficient and will spam the static layout many
times a second.

smevawala, 2016-04-02:
I tested set_fn_toggle and set_brightness.  It works perfectly :)

How do I test the keyboard_keys?  its not in the device hid folder

storborg, 2016-04-02:
@smevawala you'll have to update the device path in ``razer_chroma_drivers/scripts/driver/keyboard_keys.py`` prior to calling it.

I've been playing around with a Python API that manipulates the sysfs files directly in this branch:
https://github.com/storborg/razer_chroma_drivers/tree/python-api

There are some additional examples there, including stuff that pushes the frame rate a bit more, a snake game, etc.

terrycain, 2016-04-02:
This sounds awesome, will look at it tomorrow. Shall also pester Razer
again tomorrow :-)

terrycain, 2016-04-03:
Am busy with family atm but from what I've seen of it, it looks like something I wanted to add to the library  :wink:, I have a lot of improvements you could make off the top of my head but its a good idea @storborg. I was thinking of rolling my own driver in python with multidevice support in mind from the start.

terrycain, 2016-04-04:
OK before you think I'm ripping it to pieces this is just my opinions from the things I see and from experiences of making python daemon v1.

\__init\__.py

@storborg This [line](https://github.com/storborg/razer_chroma_drivers/blob/python-api/python/razerdriver/__init__.py#L10) There are other devices like the mamba which uses `0045` I believe.

This [line](https://github.com/storborg/razer_chroma_drivers/blob/python-api/python/razerdriver/__init__.py#L99), see below as that might fix it, some devices like the firefly, mamba, chroma and stealth all have different row lengths (stealth has 16 I think).

This [line](https://github.com/storborg/razer_chroma_drivers/blob/python-api/python/razerdriver/__init__.py#L33), below stuff _should_ solve this but what if the file doesn't exist.

This [line](https://github.com/storborg/razer_chroma_drivers/blob/python-api/python/razerdriver/__init__.py#L10). There are folders called `razerkbd` `razermouse` `razerfirefly` etc... in this folder `/sys/bus/hid/drivers/`, every device symlink in those folders will be a usable device. It might be easier to glob in those folders instead of looking for a specific file inside each symlinked folder (but it doesn't matter toooo much).

+1 for using python3 & bytes

Docstrings, docstrings everywhere! My preference is the sphinx style as that makes the introspection of my IDE nicer :wink: 

I would factor out the set effect functions as not all devices do that (some do brightness, breath and that's it). I would subclass RazerDevice and create one like RazerStealth and add the stealth methods (or subclass to make RazerKeyboard then subclass again for Stealth).
What would be useful if you could have a class of a specific device which dynamically adds all the relevant effect methods as many devices use the same methods (that's just my opinion).

canvas.py

I believe somewhere in chromaprofiles class in the main python library there is an attempt at a keymap which might be useful, it might be useful to improve that to work for different devices. (Well it should work with the stealth as it uses the same key layout minus the numpad). Also after lots of debugging, different keyboard occupy different spaces of keys, the keyboard driver should have comments documenting this, and `keyboard_keys.py` is designed to debug this.

I like how the canvas works but what is the performance like and memory overhead like?
I do like the ideas for animations in the readme :+1:

animate.py

Docstrings!!!!

Also would it not be worth merging this with the main python library for the project? (In one of my branches I've already packaged it :wink:) as `import razer.driver` would be nicer that having 2 seperate libraries.

Run pylint over it and watch it burn with "missing docstring" "missing docstring" "missing docstring" "missing docstring" "missing docstring" "missing docstring" "missing docstring" "missing docstring" "missing docstring" "missing docstring" :wink: 

Overall I like it, once I've got time will definitely use this to aid making #PythonDaemon :)

storborg, 2016-04-04:
No complaints about lack of tests? :-)

Yeah, I definitely agree with all of those things. I don't think any of this is ready to merge yet-- mostly I am just trying to figure out a good high-level animation API, and then plan on cleaning up the underlying bits. The performance / memory overhead of the canvas style is pretty decent (it's easy when your display is only 6x16 pixels or so) but I think it may be better to go with an approach where the API manages timing for you, using a "recommended" frame rate.

Also I think you definitely want to be able to do animations that are synchronized across devices... might be time for me to get a Razer mouse, too.

terrycain, 2016-04-04:
I believe if you force people to make unit tests then they normally make utterly useless tests, and they make tests in places where tests are not relevant.

Agreed, and yes, synapse can do the sync so we need to do it too

terrycain, 2016-04-04:
@storborg Can you find the serial number on synapse? It'll be well hidden or I think if you can get to a product registration screen it should be there.

Also back to packet captures, if you wireshark before you boot up windows, and then open synapse does it spam packets?

storborg, 2016-04-07:
Finally got a chance to look at this again: needless to say, it appears that the Synapse app doesn't report the serial number for the Stealth at all. Loading up the device list in the warranty section doesn't include it, and if you get further in the warranty registration process for a system it simply tells you where to find the serial number on the outside of the device. I clicked pretty exhaustively through every other part of the interface, and couldn't find any other leads.

As far as other packets on boot / synapse launch, it sends a few packets and then stops. Here's a pcap of that startup behavior (starting 35sec into the capture):
[startup-then-synapse-launch.pcapng.gz](https://github.com/pez2001/razer_chroma_drivers/files/207901/startup-then-synapse-launch.pcapng.gz)

terrycain, 2016-04-07:
Does the serial the driver gives you lots a FFFFs?

[7 more replies]

closed 2016-05-20 by terrycain