pez2001/razer_chroma_drivers #83: Does not play well with others

Pachonk, 2016-02-09:
At my dev team, we have a Razer Blackwidow Chroma, Razer Firefly, and a Razer Mamba.
It seems as though we can't configure a keyboard if you have the pad and mouse plugged in at the same time.

terrycain, 2016-02-10:
What do you mean "can't configure"

I have all 3 and they all work (though the mamba doesn't work with the dameon much)

Pachonk, 2016-02-10:
The pad takes ownership of gui before keyboard...
The only way to configure both is config pad, unplug, restart service, config keyboard, turn off service.

terrycain, 2016-02-10:
If you use the chroma controller app and set wave mode or spectrum it applies to the firefly not the chroma?

Pachonk, 2016-02-10:
Yes.

terrycain, 2016-02-10:
Ok cool. I am aware of that. Basically the daemon needs to be told to switch to a different device before the change effect commands are ran. I believe I added some code in the razer python library to enumerate devices,  that just needs to be fed into all the app.

I'm quite busy now until May unfortunately, exams/CW etc... but after I was planning on making a big push with the app

terrycain, 2016-02-10:
@lah7 would also be one to prod as its his app ;)

lah7, 2016-02-16:
If I find out the exact Python DBUS instructions to use to 'select' and 'enumerate', I can implement the multi-device screen to address this bug.

(I only have the Chroma, so I wouldn't be able to properly test it works)

terrycain, 2016-02-16:
@lah7 look in the daemon library file for `enumerate_devices(self)` that contains all you should need. `set_active(id)` sets the active device in the daemon (which we should really do before every call as other programs are also able to change the active device).

laxdragon, 2016-04-05:
I can help test this when there is a pull available. I have 3 devices as well (Blackwidow, Mamba, Fierefly).

terrycain, 2016-04-06:
@laxdragon I've merged some new changes and believe i've got the daemon to recognise the mamba but it will be a while before I've got time to put some real effort into the gui/tray to do this, as 2 entirely new screens will need to be made to support the new devices, plus a screen for the razer stealth.

laxdragon, 2016-04-06:
I helped you with the Mamba the other day, and that is working just fine. It is just that with 3 devices the GUI does nothing now. Perhaps just a fix so the GUI will at least config the keyboard while ignoring the other devices will be fine. I'm in no rush for a full GUI on the Mamba and Firefly, I can do those from the command line for now.

terrycain, 2016-04-06:
I'll have a look later I might be able to fit in a quick hack so the gui always tries to set the keyboard as the active device.

lkd70, 2016-08-09:
I too am having this issue, razer blackwidow chroma and firefly, firefly takes priority of the controls..

terrycain, 2016-08-09:
Guys keep an eye on my fork. Have moved all development there now. In one of the issues there is some text about how to join our telegram chat group of you want to stay updated. Basically I've made a new daemon which is designed to work with multiple devices from the start