pez2001/razer_chroma_drivers #84: razer_bcd fails to start on Ubuntu 14.04

wolfen351, 2016-02-14:
Hi Guys

The build etc was smooth, but then the dpkg -i runs it gives an error saying it cannot start the job. 

Manually running the service start also fails. 

wolfen@wolfen-HP:/usr/share/razer_bcd$ sudo service razer_bcd start
start: Job failed to start

Where can I look for more info?

terrycain, 2016-02-14:
`cat /var/log/upstart/razer_bcd.log` I think.

wolfen351, 2016-02-14:
wolfen@wolfen-HP:~$ sudo cat /var/log/upstart/razer_bcd.log
    [sudo] password for wolfen: 
    Unbinding device (0003:1532:011A.0006) from hid-generic
    Binding device (0003:1532:011A.0006) to razerkbd
   /proc/self/fd/9: 20: /proc/self/fd/9: cannot create /sys/bus/hid/drivers/razerkbd/bind: Directory nonexistent
    ls: cannot access /sys/bus/hid/drivers/razerkbd/: No such file or directory
    ls: cannot access /sys/bus/hid/drivers/razermouse/: No such file or directory
    ls: cannot access /sys/bus/hid/drivers/razerfirefly/: No such file or directory

etc...

wolfen351, 2016-02-14:
I suspect that something is missing in the install process or dependancies?

terrycain, 2016-02-15:
Have you tried rebooting?

wolfen351, 2016-02-15:
I will try and report back

wolfen351, 2016-02-15:
Worked perfectly!

closed 2016-02-15 by wolfen351