pez2001/razer_chroma_drivers #85: [Feature Request] Ripple Effect

lah7, 2016-02-16:
As featured in Synapse. Would be a nice little addition to the default effects. :wink:

terrycain, 2016-02-16:
The ripple effect is done in software not hardware. Shoudn't be too difficult with @pez2001's effect library as all it would be is drawing circles. We could 1up the synapse and actually have the circles do collision but that would require some work

terrycain, 2016-05-22:
[Here](https://drive.google.com/open?id=0BydYgD2LzDUCdk1TR1RtU21Wd0U) #RippleEffect

GabMus, 2016-07-13:
Do you have an ETA for this effect?

lah7, 2016-07-13:
It's already implemented in the new (upcoming) daemon, I tried it a month or so ago. [See @terrycain's repository](https://github.com/terrycain/razer_drivers/tree/master/daemon2). 

[If you like, help wanted here to test the new daemon.](https://github.com/terrycain/razer_drivers/issues/5#issue-164708775)

It will be a while before it's stable **and** will work with other programs like [Polychromatic](https://github.com/lah7/polychromatic) and [razerCommander](https://github.com/gabmus/razerCommander).