pez2001/razer_chroma_drivers #86: Noob question: How can I set brightness via cli ?

wolfen351, 2016-02-18:
I'm still really a beginner at all this DBUS stuff, but I see you have provided a 

    razer_bcd_controller

to use to communicate via the cli. I have read through the help it provides and your readme page, but I still don't get it. Can someone please just give me a simple command line way to set the keyboard brightness to say 10/255 20/255 etc ?

lah7, 2016-02-18:
Providing you've compiled everything already and the daemon service is running:
`cd daemon_controller`
`sudo ./razer_bcd_controller -7 255`

255 is brightest, 0 is lowest. Commands must be ran as root. Running `./razer_bcd_controller` will display a list of all possible commands.

One way without root is to run `python3` and interface with DBUS that way by importing `razer.daemon_dbus` and running `set_brightness(255)`

terrycain, 2016-02-18:
`bcd_controller` shouldn't need root but does require the daemon running as root.

Another option is to run this as root
`echo -n '255' > /sys/bus/hid/drivers/razerkbd/KBD_ID/set_brightness`

wolfen351, 2016-02-18:
great thanks!

wolfen351, 2016-02-18:
Here is a little script i wrote to simulate the breathing effect on my ultimate:

```
#!/bin/bash

while [ : ]
do
   for i in `seq 50 5 200`
   do
      razer_bcd_controller -7 $i 
      sleep 0.15
      echo $i
   done

   for i in `seq 200 -5 50`
   do
      razer_bcd_controller -7 $i
      sleep 0.15
      echo $i
   done
   
done
```

terrycain, 2016-02-18:
Nice

closed 2016-02-18 by terrycain