pez2001/razer_chroma_drivers #87: Razer Orbweaver Chroma support

dRaiser, 2016-02-21:
Hi. I'm thinking about buying Orbweaver Chroma, is it possible that this driver would support it? Thanks!

terrycain, 2016-02-21:
Its possible the driver could support the Orbweaver, but its doubtful the
driver will work out of the box

terrycain, 2016-04-05:
@dRaiser are you willing to help us add support for the orbweaver?

dRaiser, 2016-04-07:
I'm afraid not, as I don't plan of buying it until it's feature-complete on Linux.

terrycain, 2016-04-08:
Ok, there is nothing I can do until either I get hold of an Orbweaver or unless Razer give me some technical docs for it.

closed 2016-04-08 by terrycain

dRaiser, 2016-04-08:
I understand. If I were to buy it primary for Windows usage, I'd be happy to help, but in that case we need to wait for another enthusiast.

terrycain, 2016-04-08:
Yeah. I'm currently in negotiations with Razer for information so I might
get some info to make an orbweaver driver. Who knows.