pez2001/razer_chroma_drivers #88: error installing

stefangeorge, 2016-02-22:
Thank you for a clear, concise write up on installation. This is what terminal said:

```
sudo dpkg -i /tmp/razer-chroma-driver_1.0.0_amd64.deb
Selecting previously unselected package razer-chroma-driver.
(Reading database ... 263589 files and directories currently installed.)
Preparing to unpack .../razer-chroma-driver_1.0.0_amd64.deb ...
Unpacking razer-chroma-driver (1.0.0) ...
dpkg: dependency problems prevent configuration of razer-chroma-driver:
 razer-chroma-driver depends on libfftw3-3 (>= 3.3.3-7ubuntu3); however:
  Package libfftw3-3 is not installed.

dpkg: error processing package razer-chroma-driver (--install):
 dependency problems - leaving unconfigured
Processing triggers for gnome-menus (3.13.3-6ubuntu1) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu3) ...
Processing triggers for bamfdaemon (0.5.2~bzr0+15.10.20150627.1-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for mime-support (3.58ubuntu1) ...
Processing triggers for ureadahead (0.100.0-19) ...
ureadahead will be reprofiled on next reboot
Processing triggers for systemd (225-1ubuntu9) ...
Processing triggers for dbus (1.10.0-1ubuntu1) ...
Errors were encountered while processing:
 razer-chroma-driver

```

How can I find out what errors were encountered?
I opened up Ubuntu software center afterwards and it said 'errors were encountered, do you want to fix them?'
I said yes and saw some files come and go in the tmp directory. But looking for 'razer' in the 'installed' items reveals nothing. Even when 'show technical items' is clicked at the bottom of software center.

stefangeorge, 2016-02-22:
I reran the dpg command, it installed without errors, and it works fine. Still would like to know where my ubuntu 15.10 stumbled so i could let you know.

Also, would like to be able to limit 'spectrum' to a specific gradient. Also would like to show profiles i made up in the tray dropdown.

lah7, 2016-02-22:
> Also would like to show profiles i made up in the tray dropdown.

[I've done that in my repo](https://github.com/lah7/razer_blackwidow_chroma_gui/commit/344a2693db7d59f8c3bad079a304d091ce17fb5f), but I haven't pull requested my changes to the main repository yet as I'm working on some other GUI changes too. (Been a bit busy)

terrycain, 2016-02-22:
@stefangeorge if you ran `apt-get install -f` that would of solved your issues. Using dpkg doesn't download the dependencies that the package requires which apt would do.

closed 2016-02-23 by terrycain