pez2001/razer_chroma_drivers #89: Is it possible to support the normal Blackwidow (no backlighting)?

smevawala, 2016-02-23:
I was wondering if it made sense to support the normal blackwidow and other non backlit versions of razer keyboards.  The point of this would be to have macro recordings and profiles.  I have a razer blackwidow on fedora 23 and can help test out.

terrycain, 2016-02-23:
Good idea, the main focus of this was the lighting but it wouldn't be too difficult to add support for all devices with profiles, the issue is, we haven't perfected profiles yet.

terrycain, 2016-04-05:
@smevawala what's the USB VID and PID outputted by `lsusb`

smevawala, 2016-04-05:
ID 1532:011b Razer USA, Ltd

terrycain, 2016-04-05:
Whats the output of `lsusb -d 1532:011b -vvv`

smevawala, 2016-04-05:
`Bus 003 Device 006: ID 1532:011b Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x011b 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
`

smevawala, 2016-04-05:
with better formating: [blackwidow.txt](https://github.com/pez2001/razer_chroma_drivers/files/204747/blackwidow.txt)

terrycain, 2016-04-05:
Deleted massive comment as reading the file is easier and less scrolling :wink: 

Ok cool, that means that the Blackwidow also uses interface 2 (as a mouse) to be controlled. If you were to prod the keyboard driver to allow the vid/pid it would bind. So what can RazerSynapse do with the keyboard? A list of features would be good. (obviously it cant change colour)

smevawala, 2016-04-05:
RazerSynapse can:
Change the brightness of the razer logo
Pulsate the razer logo
Create macro profiles
Turn on game mode
Change what game mode does

terrycain, 2016-04-06:
Ok profiles we think are done in software on the synapse side.
Game mode should work the same as the others.
I'm hoping brightness and pulsate will be the same as my bwultimate2013.

I'll make some changes to the driver later in a pull request and then you can try that out

laurentdinclaux, 2016-05-09:
Ant news ? I just bought the exact same keyboard and the daemon won't start, with not a single feedback about why it won't:

```
~ sudo systemctl start razer_bcd
~ sudo systemctl status razer_bcd
● razer_bcd.service - LSB: start/stops razer_bcd
   Loaded: loaded (/etc/init.d/razer_bcd)
   Active: active (exited) since lun. 2016-05-09 12:09:34 NCT; 22min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 3269 ExecStart=/etc/init.d/razer_bcd start (code=exited, status=0/SUCCESS)
   Memory: 0B
      CPU: 0

mai 09 12:09:34 nautile systemd[1]: Starting LSB: start/stops razer_bcd...
mai 09 12:09:34 nautile razer_bcd[3269]: * Starting Razer Keyboard daemon razer_bcd
mai 09 12:09:34 nautile razer_bcd[3269]: Starting razer blackwidow chroma daemon as a daemon
mai 09 12:09:34 nautile razer_bcd[3269]: ...done.
mai 09 12:09:34 nautile systemd[1]: Started LSB: start/stops razer_bcd.
mai 09 12:11:51 nautile systemd[1]: Started LSB: start/stops razer_bcd.
mai 09 12:22:41 nautile systemd[1]: Started LSB: start/stops razer_bcd.
mai 09 12:32:09 nautile systemd[1]: Started LSB: start/stops razer_bcd.
~ sudo razer_bcd --verbose=7
Starting razer blackwidow chroma daemon as a daemon
~ ps ax | grep razer
 7851 pts/11   S+     0:00 grep --color=auto --exclude-dir=.bzr --exclude-dir=CVS --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn razer
 ```

christ66, 2016-05-09:
@terrycain I was trying to work on this issue for an hour, but got pulled away. Current branch: https://github.com/christ66/razer_chroma_drivers/tree/BlackWidow. The errors I'm getting after restarting the machine:
Logs from `/var/log/upstart/razer_bcd.log:

```
Device already binded
Starting razer blackwidow chroma daemon as a daemon
Unbinding device (0003:1532:011B.0004) from razerkbd
Binding device (razerkbd) to hid-generic
Bind Successful
ls: cannot access '/sys/bus/hid/drivers/razermouse/': No such file or directory
ls: cannot access '/sys/bus/hid/drivers/razerfirefly/': No such file or directory
Unbinding device (0003:1532:011B.0004) from hid-generic
Binding device (0003:1532:011B.0004) to razerkbd
ls: cannot access '/sys/bus/hid/drivers/razermouse/': No such file or directory
ls: cannot access '/sys/bus/hid/drivers/razerfirefly/': No such file or directory
Binding device (0003:1532:011B.0004) to razerkbd
ls: cannot access '/sys/bus/hid/drivers/razermouse/': No such file or directory
ls: cannot access '/sys/bus/hid/drivers/razerfirefly/': No such file or directory
```

I'll try taking a look later tonight on this issue.

HTH

terrycain, 2016-05-09:
@laurentdinclaux The daemon needs some major work and I'm working on a replacement that should be easier to manage (coming to the end of Uni so am a little busy atm).

Ok does the normal blackwidow have macro keys/macro LED?

Is someone willing to capture some traffic so we can get the data needed to add the features? (I believe most of it I already know but I'd rather make sure). See [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol) wiki page for more details.

@christ66 whilst most of what you've done is good but I would rather make sure of the features and add the specific ones to the driver. And that issue you have with the daemon `No such file or directory` was a bug I introduced a while ago but with the new daemon I'm aiming to remove all that. Plus you've missed an entry to the daemon so it still wouldn't mount :wink:

laurentdinclaux, 2016-05-09:
It has five macro keys and macro leds

DanielVoogsgerd, 2016-05-12:
I havent tried this yet and it might already be supported, but I've a Razer Black widow Ultimate (2012 model I think, the blue one) and I'd happily capture some data if it would help. I've no clue if they have the same internals, but I thought I'd put it out there just in case.

terrycain, 2016-05-13:
In theory the 2012 should work identically to the 2013. Once I've finished Uni ill make some changes to the keyboard driver for you guys to try out

terrycain, 2016-05-22:
Ok who's willing to test what device?

DanielVoogsgerd, 2016-05-22:
I'm willing to test a `rz03-0038`

terrycain, 2016-05-22:
@DanielVoogsgerd What device is that? (I don't know the product codes ha). 

I need the following:
 * Actual product name :wink: 
 * Does it have a backlight,
 * Link to the relevant razer url would be helpful
 * A list of features it has (ideally go through the razer synapse app and list them)
 * Its USB VID and PID (can get that from running `lsusb`)

DanielVoogsgerd, 2016-05-22:
Here you go:
- Razer Blackwidow 2012 Edition
- Yes it has a backlight
- http://www.razerzone.com/store/razer-blackwidow-ultimate-classic (I think)
- I've no operating system nearby with synapse support ;)
- VID/PID: 1532:010d

terrycain, 2016-05-22:
Ok cool, do you know if it does the breathing/pulsate mode?

laurentdinclaux, 2016-05-22:
I can test the Blackwidow (1532:011b Razer USA, Ltd)

- Razer Blackwidow
- no backlight
-
http://www.amazon.com/dp/B00IG3GP84/ref=twister_B01DW7OHC4?_encoding=UTF8&psc=1
(Clicky
/ Full version)
- 5 macro keys, logo light
-

RazerSynapse can:

- Change the brightness of the razer logo
- Pulsate the razer logo
- Create macro profiles
- Turn on game mode
- Change what game mode does

DanielVoogsgerd, 2016-05-22:
@terrycain Yes it does, I can still enable it via the Fn-F12 combo

terrycain, 2016-05-22:
@laurentdinclaux thanks, no backlight, hmm, this will require some trial and error :wink: 

Hopefully the standard brightness commands will work on the logo. Pulsating the logo should work like on my 2013 keyboard. Macro profiles are done in software along with "what game mode does" (all game mode is, is technically switching to a keyboard layout without a windows key, all done in hardware)

Should be able to get you a driver to test tomorrow hopefully. Read [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver) page and become familiar with binding the device to a driver, as that will need to be done to test.

terrycain, 2016-05-22:
@DanielVoogsgerd doesnt FN+F12 just up the brightness?

DanielVoogsgerd, 2016-05-22:
@terrycain It does kinda, it cycles through 3 levels of brightness, no backlighting whatsoever and the pulsating mode.

terrycain, 2016-05-22:
Ahh ok cool. Read the comment to laurentdinclaux above, the same applies to you :smile:

DanielVoogsgerd, 2016-05-22:
@terrycain Done reading, not that difficult and a pretty decent guide. Should work just fine (if the driver is compat of course ;)) If you have any questions about possibly available features feel free to ask.

terrycain, 2016-05-22:
@DanielVoogsgerd Thanks, is good to know the guide works :smile:. And will do.

terrycain, 2016-05-23:
@laurentdinclaux On my fork of the razer repository there is a [branch](https://github.com/terrycain/razer_drivers/tree/RazerBlackWidow_0x011B) called `RazerBlackWidow_0x011B` . 

1. Checkout that branch.
1. Make driver with `make driver_verbose`.
1. Load driver with `sudo insmod ./driver/razerkbd.ko` (make sure any `razerkbd.ko` drivers have been unloaded first using `rmmod`).
1. Bind the keyboard to the driver (using info from the wiki page I linked).
1. Run `ls -l` on the driver directory and put output on pastebin and link here.

terrycain, 2016-05-23:
@DanielVoogsgerd Same as above but with the branch `RazerBlackWidow2012_0x010D`

DanielVoogsgerd, 2016-05-23:
I managed to get the driver loaded the guide was a little off for me though, but everything seems fine.
I can't find a mouse device though:
https://gist.github.com/DanielVoogsgerd/1f167e58739b76d574a55f35673ee8bc

terrycain, 2016-05-23:
@DanielVoogsgerd With the keyboard's you normally always want to bind the device with the highest number at the end :wink:. Need to update the wiki a bit.

Whats the output of `udevadm info -a /sys/bus/hid/devices/0003:1532:0203.0015 | grep bInterfaceNumber` replacing the device ID with ones from your system.

Also what is the output of the following (replacing the ID with yours)
```
ls -l /sys/bus/hid/drivers/razerkbd

ls -l /sys/bus/hid/drivers/razerkbd/0003:1532:0203.0015
```

DanielVoogsgerd, 2016-05-23:
udevadm info -a: `ATTRS{bInterfaceNumber}=="02"`

`total 0
lrwxrwxrwx 1 root root 0 May 23 23:24 0003:1532:0016.0005 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:1532:0016.0005/
lrwxrwxrwx 1 root root 0 May 23 23:24 0003:1532:010D.0001 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:1532:010D.0001/
lrwxrwxrwx 1 root root 0 May 23 23:24 0003:1532:010D.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1/0003:1532:010D.0002/
lrwxrwxrwx 1 root root 0 May 23 23:24 0003:1532:010D.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2/0003:1532:010D.0003/`

`lrwxrwxrwx 1 root root 0 May 23 23:24 /sys/bus/hid/devices/0003:1532:010D.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2/0003:1532:010D.0003/`

laurentdinclaux, 2016-05-23:
**RazerBlackWidow_0x011B**
```
# ls -l /sys/bus/hid/drivers/razerkbd/
total 0
lrwxrwxrwx 1 root root    0 mai   24 08:41 0003:1532:011B.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.2/0003:1532:011B.0005
--w------- 1 root root 4096 mai   24 08:41 bind
lrwxrwxrwx 1 root root    0 mai   24 08:40 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 mai   24 08:40 new_id
--w------- 1 root root 4096 mai   24 08:40 uevent
--w------- 1 root root 4096 mai   24 08:40 unbind
# ls -l /sys/bus/hid/drivers/razerkbd/0003\:1532\:011B.0005/
total 0
-r--r--r-- 1 root root 4096 mai   24 08:42 country
-r--r--r-- 1 root root 4096 mai   24 08:42 device_type
lrwxrwxrwx 1 root root    0 mai   24 08:42 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r--r-- 1 root root 4096 mai   24 08:42 get_firmware_version
-r--r--r-- 1 root root 4096 mai   24 08:42 get_serial
drwxr-xr-x 3 root root    0 mai   24 08:42 hidraw
drwxr-xr-x 3 root root    0 mai   24 08:42 input
--w--w---- 1 root root 4096 mai   24 08:42 macro_keys
-r--r--r-- 1 root root 4096 mai   24 08:42 modalias
-rw-rw---- 1 root root 4096 mai   24 08:42 mode_game
-rw-rw---- 1 root root 4096 mai   24 08:42 mode_macro
-rw-rw---- 1 root root 4096 mai   24 08:42 mode_macro_effect
-rw-rw---- 1 root root 4096 mai   24 08:42 mode_pulsate
--w--w---- 1 root root 4096 mai   24 08:42 mode_static
drwxr-xr-x 2 root root    0 mai   24 08:42 power
-r--r--r-- 1 root root 4096 mai   24 08:42 report_descriptor
--w--w---- 1 root root 4096 mai   24 08:42 reset
-rw-rw---- 1 root root 4096 mai   24 08:42 set_brightness
lrwxrwxrwx 1 root root    0 mai   24 08:42 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root root 4096 mai   24 08:42 test
-rw-r--r-- 1 root root 4096 mai   24 08:42 uevent
#
```

terrycain, 2016-05-23:
> udevadm info -a: ATTRS{bInterfaceNumber}=="02"

Are you saying 0005 is the only one with interface 2? (That would be correct ha).

Ok try
```
cd /sys/bus/hid/drivers/razerkbd/0003\:1532\:011B.0005/
cat ./device_type
cat ./get_firmware_version
cat ./get_serial
```

DanielVoogsgerd, 2016-05-23:
Oh sorry I only tried the last one after your comment

terrycain, 2016-05-23:
@DanielVoogsgerd check them all but only 1 should be interface 2. (and make sure you only check the razer devices not others :wink:)

terrycain, 2016-05-23:
@DanielVoogsgerd Yeah just realised that. Bind the 0003 one :wink:

laurentdinclaux, 2016-05-23:
**RazerBlackWidow_0x011B**

```
/sys/bus/hid/drivers/razerkbd/0003:1532:011B.0005# cat ./device_type
Razer BlackWidow Original
/sys/bus/hid/drivers/razerkbd/0003:1532:011B.0005# cat ./get_firmware_version
v3.0
/sys/bus/hid/drivers/razerkbd/0003:1532:011B.0005# cat ./get_serial
IO154403900027
```

DanielVoogsgerd, 2016-05-23:
The cd would work since
````
total 0
--w------- 1 root root 4.0K May 23 23:43 bind
lrwxrwxrwx 1 root root    0 May 23 23:43 module -> ../../../../module/razerkbd/
--w------- 1 root root 4.0K May 23 23:43 new_id
--w------- 1 root root 4.0K May 23 23:43 uevent
--w------- 1 root root 4.0K May 23 23:43 unbind
````

terrycain, 2016-05-23:
@DanielVoogsgerd Did you bind the device to the driver? and unbind it from hid-generic?

@laurentdinclaux :smile: Well, you've picked the right device as its its returned useful results. Now try
```
echo -n '32' > ./set_brightness
echo -n '64' > ./set_brightness
echo -n '128' > ./set_brightness
echo -n '255' > ./set_brightness
```
Does that change the brightness?

laurentdinclaux, 2016-05-23:
@terrycain yeah ! Well done! 

It changes the logo brightness and a value of 0 turns it off (as a reminder my keyboard has no back light)

DanielVoogsgerd, 2016-05-23:
@terrycain Oh sorry could have guessed that. Unbinding works but binding doesn't

terrycain, 2016-05-23:
@DanielVoogsgerd When unbinding did it disappear from the `hid-generic` directory?, what command did you use for binding? Pastebin the output of `dmesg`

@laurentdinclaux Good :smile: Try setting the brightness then reading it back (I might have been too lazy to add that)
```
echo -n '128' > ./set_brightness
cat ./set_brightness

# Should turn game mode on
echo -n '1' > ./mode_game
# Should turn game mode off
echo -n '0' > ./mode_game
# Repeat for mode_macro

laurentdinclaux, 2016-05-23:
Reading back brightness: OK
Setting/unsettings game mode from command line and reading back the value: OK
Setting/unsettings game from keyboard shortcut (FN+F10): OK

I made an Ubuntu package and installed it (using the dedicated script). I rebooted, the driver doesn't bind itself on boot. It may be an intended behavior at that stage thought ...

DanielVoogsgerd, 2016-05-23:
Yes it did disappear. I pipe the id to the bind file in the `/sys/bus/hid/drivers/razerkbd` directory
https://gist.github.com/DanielVoogsgerd/e53d664b43ec28dd0eda2f889172b521

terrycain, 2016-05-23:
@DanielVoogsgerd I've made a typo in the driver for you. Open up `driver/razerkbd_driver.h`
Find the line 
```
#define USB_DEVICE_ID_RAZER_BLACKWIDOW_ULTIMATE_2012 0x010B
```
Replace with
```
#define USB_DEVICE_ID_RAZER_BLACKWIDOW_ULTIMATE_2012 0x010D
```
Then recompile, then run `sudo rmmod razerkbd`. Then run the insmod comand etc...

@laurentdinclaux Yeah, it needs to be added to udev etc... but I wanted to get it working before I'm bothered.

There is still more to try :wink: 
```
# Turn on macro led
echo -n '1' > ./mode_macro
# Turn off macro led
echo -n '0' > ./mode_macro

# Turn on macro led for this to work
echo -n '1' > ./mode_macro
# Set macro led to blinking
echo -n '1' > ./mode_macro_effect
# Set macro led to static
echo -n '0' > ./mode_macro_effect

# Set logo to pulsate mode
echo -n '1' > ./mode_pulsate
# Set logo to static
echo -n '1' > ./mode_static
```

DanielVoogsgerd, 2016-05-23:
Still trouble with binding

laurentdinclaux, 2016-05-23:
@terrycain tests all worked

DanielVoogsgerd, 2016-05-23:
@terrycain Yes I got it working now aswell.

terrycain, 2016-05-24:
@DanielVoogsgerd The tests above also apply to you, can you try them and tell me what ones work.

@laurentdinclaux will add that to the driver tomorrow (as some other stuff needs to be done on my part).

DanielVoogsgerd, 2016-05-24:
@terrycain I did see my previous message.

terrycain, 2016-05-24:
@DanielVoogsgerd, no I didn't, page must not have refreshed/updated. Ok cool well I've got some udev stuff to mess with but should have the driver adjustments merged into the main branch by tomorrow.

I'm planning on replacing the main daemon service with a python replacement which _currently_ works rather nicely :) but that might take a while.

DanielVoogsgerd, 2016-05-24:
@terrycain Nice work :)

terrycain, 2016-05-24:
Have added the classic and 2012 to the main codebase. As the daemon replacement might take a while and technically support has now been added, I shall mark this as closed. (the less issues the better right :wink:)

Just cause I'm proud of it :smile: [here](https://drive.google.com/file/d/0BydYgD2LzDUCVm1LR04xdjRFa3c/view?usp=sharing) is a look at the ripple effect done with the _new_ daemon. It also does macro's etc... so should be good once I've got it to a point where its acceptable to be merged.

closed 2016-05-24 by terrycain

DanielVoogsgerd, 2016-05-24:
@terrycain Thanks, now I feel like I need an upgrade to a chroma... :p
Are you planning on upstreaming your changes to this repo?

terrycain, 2016-05-24:
I've merged the driver changes to this repo already :wink:. Hopefully I will merge my daemon with this repo, the only thing is that it replaces all of pez's work on the daemon. He's been inactive for ages and I don't think he'll mind as long as things will work and are progressing.

DanielVoogsgerd, 2016-05-24:
You could also reference your own repo as an updated version in the readme and consider yourself upstream.

terrycain, 2016-05-24:
Not a bad idea at all. Yeah will ponder on this.

laurentdinclaux, 2016-05-24:
I have issues with the full implementation.

```
root@nautile:/sys/bus/hid/drivers/razerkbd# cat /var/log/syslog | grep raz
May 25 07:20:30 nautile systemd[1]: Configuration file /lib/systemd/system/razer_bcd.service is marked executable. Please remove executable permission bits. Proceeding anyway.
May 25 07:20:30 nautile systemd[1]: Configuration file /lib/systemd/system/razer_bcd.service is marked executable. Please remove executable permission bits. Proceeding anyway.
May 25 07:20:38 nautile systemd[1]: Configuration file /lib/systemd/system/razer_bcd.service is marked executable. Please remove executable permission bits. Proceeding anyway.
May 25 07:20:38 nautile systemd[1]: Configuration file /lib/systemd/system/razer_bcd.service is marked executable. Please remove executable permission bits. Proceeding anyway.
May 25 07:20:38 nautile systemd[4475]: razer_bcd.service: Failed at step EXEC spawning /usr/share/razer_bcd/systemd_helpers.sh: No such file or directory
May 25 07:20:38 nautile systemd[1]: razer_bcd.service: Control process exited, code=exited status=203
May 25 07:20:38 nautile systemd[1]: razer_bcd.service: Unit entered failed state.
May 25 07:20:38 nautile systemd[1]: razer_bcd.service: Failed with result 'exit-code'.
root@nautile:/sys/bus/hid/drivers/razerkbd# ls /usr/share/razer_bcd/systemd_helpers.sh
ls: impossible d'accéder à /usr/share/razer_bcd/systemd_helpers.sh: Aucun fichier ou dossier de ce type
root@nautile:/sys/bus/hid/drivers/razerkbd# dpkg -L razer-chroma-driver 
/.
/etc
/etc/xdg
/etc/xdg/autostart
/etc/init
/etc/init/razer_bcd.conf
/etc/udev
/etc/udev/rules.d
/etc/udev/rules.d/95-razerkbd.rules
/etc/dbus-1
/etc/dbus-1/system.d
/etc/dbus-1/system.d/org.voyagerproject.razer.daemon.conf
/etc/init.d
/etc/init.d/razer_bcd
/usr
/usr/src
/usr/src/razer_chroma_driver-1.0.0
/usr/src/razer_chroma_driver-1.0.0/driver
/usr/src/razer_chroma_driver-1.0.0/driver/razercommon.h
/usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razercommon.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerkbd_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/Makefile
/usr/src/razer_chroma_driver-1.0.0/driver/razerkbd_driver.h
/usr/src/razer_chroma_driver-1.0.0/dkms.conf
/usr/src/razer_chroma_driver-1.0.0/Makefile
/usr/bin
/usr/bin/razer_bcd_controller
/usr/share
/usr/share/razer_bcd
/usr/share/razer_bcd/bash_keyboard_functions.sh
/usr/share/razer_bcd/fx
/usr/share/razer_bcd/fx/pez2001_progress_bar.so
/usr/share/razer_bcd/fx/pez2001_collection.so
/usr/share/razer_bcd/fx/pez2001_light_blast.so
/usr/share/razer_bcd/fx/pez2001_mixer.so
/usr/sbin
/usr/sbin/razer_bcd
/usr/lib
/usr/lib/librazer_chroma.so
/usr/lib/librazer_chroma_controller.so
/usr/lib/python3
/usr/lib/python3/dist-packages
/usr/lib/python3/dist-packages/razer
/usr/lib/python3/dist-packages/razer/daemon_dbus.py
/usr/lib/python3/dist-packages/razer/keyboard.py
/usr/lib/python3/dist-packages/razer/__init__.py
root@nautile:/sys/bus/hid/drivers/razerkbd# ps ax | grep razer
 6577 pts/6    S+     0:00 grep --color=auto razer
root@nautile:/sys/bus/hid/drivers/razerkbd# ls -lah
total 0
drwxr-xr-x 2 root root    0 mai   25 07:33 .
drwxr-xr-x 4 root root    0 mai   25 07:33 ..
--w------- 1 root root 4,0K mai   25 07:33 bind
lrwxrwxrwx 1 root root    0 mai   25 07:33 module -> ../../../../module/razerkbd
--w------- 1 root root 4,0K mai   25 07:33 new_id
--w------- 1 root root 4,0K mai   25 07:33 uevent
--w------- 1 root root 4,0K mai   25 07:33 unbind
```

[4 more replies]