pez2001/razer_chroma_drivers #96: Can't compile on Arch

Rain288, 2016-03-05:
Greetings,
I used to use this when I was using Ubuntu and it worked perfectly, but now that I've switched to Arch I get a problem when compiling "razer kernel modules".
Here is the "sudo make -s all install" command output:

```
\n::\033[32m COMPILING librazer_chroma.so\033[0m
================================
\n::\033[32m COMPILING librazer_chroma.so [DEBUG]\033[0m
========================================
\n::\033[32m COMPILING razer_bcd [SHARED]\033[0m
================================
\n::\033[32m COMPILING razer_bcd daemon\033[0m
==============================
\n::\033[32m COMPILING razer_bcd daemon [DEBUG]\033[0m
======================================
\n::\033[32m COMPILING pez2001 fx collection [DEBUG]\033[0m
===========================================
\n::\033[32m COMPILING pez2001 fx mixer [DEBUG]\033[0m
======================================
\n::\033[32m COMPILING pez2001 fx light blast [DEBUG]\033[0m
============================================
\n::\033[32m COMPILING pez2001 fx progress bar [DEBUG]\033[0m
============================================
\n::\033[32m COMPILING pez2001 fx compute fft [DEBUG]\033[0m
============================================
\n::\033[32m COMPILING pez2001 fx collection\033[0m
===================================
\n::\033[32m COMPILING pez2001 fx mixer\033[0m
==============================
\n::\033[32m COMPILING pez2001 fx light blast\033[0m
====================================
\n::\033[32m COMPILING pez2001 fx progress bar\033[0m
====================================
\n::\033[32m COMPILING pez2001 fx compute fft\033[0m
====================================
\n::\033[32m COMPILING razer_bcd controller [SHARED]\033[0m
===========================================
\n::\033[32m COMPILING razer_bcd controller\033[0m
=========================================
\n::\033[32m COMPILING razer_bcd controller [DEBUG]\033[0m
=================================================
\n::\033[32m COMPILING examples [DEBUG]\033[0m
==============================
\n::\033[32m COMPILING examples\033[0m
======================
\n::\033[32m COMPILING snake game\033[0m
======================
\n::\033[32m COMPILING razer kernel modules\033[0m
========================================
Makefile:10: recipe for target 'razer_drv' failed
make: *** [razer_drv] Error 2
```

Not sure what the problem is...

lah7, 2016-03-05:
I've encountered a make Error 2 before. Do you have a space in or leading up to your working directory?
For instance, `/home/user/Drivers for keyboard/` causes the error but `/home/user/Drivers_for_keyboard/` would not.

https://github.com/pez2001/razer_chroma_drivers/wiki/Troubleshooting

Rain288, 2016-03-05:
Here is my path: `/home/rain/Documents/razer_chroma_drivers/` , I don't see any spaces, I already googled it and found the previous Github issue talking about it so I made sure to check. What I'm noticing is it fails to compile "razer kernel modules" which does seem to have a space in it

terrycain, 2016-03-06:
There is a verbose make target which you can run to get more info. Will
look in the morning if you can't find it. The output from that would help us

Rain288, 2016-03-06:
Alright, I did make --debug=verbose all, here is the output

```
rain@rainarch  ~/Documents/rzr   master  make --debug=verbose all
GNU Make 4.1
Built for x86_64-unknown-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Reading makefiles...
Reading makefile 'Makefile'...
Updating goal targets....
Considering target file 'all'.
 File 'all' does not exist.
  Considering target file 'librazer_chroma'.
   File 'librazer_chroma' does not exist.
   Finished prerequisites of target file 'librazer_chroma'.
  Must remake target 'librazer_chroma'.
make -C lib all
GNU Make 4.1
Built for x86_64-unknown-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Reading makefiles...
Reading makefile 'Makefile'...
make[1]: Entering directory '/home/rain/Documents/rzr/lib'
Updating goal targets....
Considering target file 'all'.
 File 'all' does not exist.
  Considering target file 'library'.
   File 'library' does not exist.
    Considering target file 'razer_chroma_static'.
     File 'razer_chroma_static' does not exist.
      Considering target file 'razer_chroma.o'.
        Considering target file 'razer_chroma.c'.
         Finished prerequisites of target file 'razer_chroma.c'.
        No need to remake target 'razer_chroma.c'.
       Finished prerequisites of target file 'razer_chroma.o'.
       Prerequisite 'razer_chroma.c' is older than target 'razer_chroma.o'.
      No need to remake target 'razer_chroma.o'.
      Considering target file 'list.o'.
        Considering target file 'list.c'.
         Finished prerequisites of target file 'list.c'.
        No need to remake target 'list.c'.
       Finished prerequisites of target file 'list.o'.
       Prerequisite 'list.c' is older than target 'list.o'.
      No need to remake target 'list.o'.
      Considering target file 'razer_string.o'.
        Considering target file 'razer_string.c'.
         Finished prerequisites of target file 'razer_string.c'.
        No need to remake target 'razer_string.c'.
       Finished prerequisites of target file 'razer_string.o'.
       Prerequisite 'razer_string.c' is older than target 'razer_string.o'.
      No need to remake target 'razer_string.o'.
     Finished prerequisites of target file 'razer_chroma_static'.
    Must remake target 'razer_chroma_static'.
ar -c -rs librazer_chroma.a razer_chroma.o list.o razer_string.o
    Successfully remade target file 'razer_chroma_static'.
    Considering target file 'razer_chroma_shared'.
     File 'razer_chroma_shared' does not exist.
      Pruning file 'razer_chroma.o'.
      Pruning file 'list.o'.
      Pruning file 'razer_string.o'.
     Finished prerequisites of target file 'razer_chroma_shared'.
    Must remake target 'razer_chroma_shared'.
gcc -rdynamic -nostdlib -DCREATELIB -shared razer_chroma.o list.o razer_string.o -lm `pkg-config --libs dbus-1` -o librazer_chroma.so
    Successfully remade target file 'razer_chroma_shared'.
[... 429 lines cut ...]
g++ snake.cpp -Wall -Wextra -W -w -Os -std=c99 -DMAJOR_VERSION=0 -DMINOR_VERSION=3 -fPIC -DPIC -DLINUX -D_XOPEN_SOURCE=700 ../../lib/librazer_chroma.a -lm -o snake_game
  Successfully remade target file 'snake'.
 Finished prerequisites of target file 'all'.
Must remake target 'all'.
Successfully remade target file 'all'.
make[2]: Leaving directory '/home/rain/Documents/rzr/examples/snake_game'
  Successfully remade target file 'snake'.
  Considering target file 'dynamic'.
   Finished prerequisites of target file 'dynamic'.
  No need to remake target 'dynamic'.
 Finished prerequisites of target file 'all'.
Must remake target 'all'.
Successfully remade target file 'all'.
make[1]: Leaving directory '/home/rain/Documents/rzr/examples'
  Successfully remade target file 'razer_examples'.
  Considering target file 'razer_drv'.
   File 'razer_drv' does not exist.
   Finished prerequisites of target file 'razer_drv'.
  Must remake target 'razer_drv'.
\n::\033[32m COMPILING razer kernel modules\033[0m
========================================
make -C /lib/modules/4.4.3-1-ARCH/build SUBDIRS=/home/rain/Documents/rzr/driver modules > /dev/null 2>&1
Makefile:10: recipe for target 'razer_drv' failed
make: *** [razer_drv] Error 2
```

terrycain, 2016-03-06:
run `make razer_drv_verbose`

Rain288, 2016-03-06:
Oh, sorry for the misunderstanding, here is the output of that command:
```
\n::\033[32m COMPILING razer kernel modules\033[0m
========================================
make -C /lib/modules/4.4.3-1-ARCH/build SUBDIRS=/home/rain/Documents/rzr/driver modules
make[1]: Entering directory '/usr/lib/modules/4.4.3-1-ARCH/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.4.3-1-ARCH/build'
Makefile:15: recipe for target 'razer_drv_verbose' failed
make: *** [razer_drv_verbose] Error 2
```

terrycain, 2016-03-06:
in the makefile change 

```
razer_drv_verbose:
        @echo "\n::\033[32m COMPILING razer kernel modules\033[0m"
        @echo "========================================"
        make -C $(KERNELDIR) SUBDIRS=$(DRIVERDIR) modules
```
to
```
razer_drv_verbose:
        @echo "\n::\033[32m COMPILING razer kernel modules\033[0m"
        @echo "========================================"
        make -C $(KERNELDIR) SUBDIRS=$(DRIVERDIR)
```

Then rerun `make razer_drv_verbose`

Rain288, 2016-03-06:
Same-y error:
```
\n::\033[32m COMPILING razer kernel modules\033[0m
========================================
make -C /lib/modules/4.4.3-1-ARCH/build SUBDIRS=/home/rain/Documents/rzr/driver
make[1]: Entering directory '/usr/lib/modules/4.4.3-1-ARCH/build'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.4.3-1-ARCH/build'
Makefile:15: recipe for target 'razer_drv_verbose' failed
make: *** [razer_drv_verbose] Error 2
```

terrycain, 2016-03-06:
are you sure you have the kernel headers installed?

Rain288, 2016-03-06:
I did have it installed, but to make sure I reinstalled it, and make worked fine!
Running into another issue now however...
```
\n::\033[32m INSTALLING razer daemon init.d file\033[0m
=====================================================
chown: cannot access '/etc/init.d/razer_bcd': Not a directory
Makefile:98: recipe for target 'install' failed
make: *** [install] Error 1
```
Apparently there was a github issue about this before too but when looking it up I don't really see what the fix for it is.

terrycain, 2016-03-06:
Yeah this is a commonly known issue, you need to use the systemd scripts. Try `make fedora_install`

Rain288, 2016-03-06:
Alright, works fine-ish, but then I get this
```
/usr/lib/dkms/common.postinst razer_chroma_driver 1.0.0 /usr/share/razer_chroma_driver-dkms
make: /usr/lib/dkms/common.postinst: Command not found
Makefile:45: recipe for target 'fedora_install' failed
make: *** [fedora_install] Error 127
```

terrycain, 2016-03-06:
Look at installing the DKMS package

Rain288, 2016-03-06:
That did the trick, thank you so much! I am really greatful for your assistance! I know I can be annoying and sometimes want to be spoonfed but I am pretty new to linux (thus should have probably not moved to Arch so fast...) so your help was very much appreciated!
Again, can't thank you enough for both your work on this driver and for this specific issue!

closed 2016-03-06 by Rain288

terrycain, 2016-03-06:
Not a problem at all, it's just nice to see the driver working and being
used. Yeah I wouldn't recommend switching to arch whilst your still new to
all things command line but it's not the end of the world. The main issue
is that we don't officially support it. When I get some more time ill look
into writing a guide with how to do it as we've got a few arch users now