pez2001/razer_chroma_drivers #98: WebKit issue

GovanifY, 2016-03-19:
The following error happens on fresh install of deb based distros: 
ValueError: Namespace WebKit not available
This is due to a dependancie not beeing told to the deb package, beeing the gtk+3 package for ubuntu(guess that's the same for debian)

terrycain, 2016-03-19:
Yeah sounds about right. The settings file for the deb's were made ages ago and haven't been kept up to date. Over the next few months (I'm quite preoccupied with other stuff you see) I'm redoing all of the packaging so it should in theory work much nicer and then there should be a PPA for people to use instead.

terrycain, 2016-05-22:
The applications to control the driver/daemon have been moved out to a separate repository, so marking this as closed/

closed 2016-05-22 by terrycain