pez2001/razer_chroma_drivers #101: Cannot connect to razer_bcd

GovanifY, 2016-03-28:
This is an issue I've got for quite a while but basically in the daemon_dbus.py this line
`self.dbus_daemon_object = system_bus.get_object("org.voyagerproject.razer.daemon", "/")`
is crashing, going to say each time: "Failed to connect to the dbus service. Is the razer_bcd service running?"
I tried running razer_bcd directly from the compiled version or anything(I have very few experience with daemons) and restarted countless of times without any clues.

The less clueless hints I succeeded to get was that:
`sudo bash razer_bcd_ubuntu force-reload
 * Restarting Razer Keyboard daemon razer_bcd                                                                        start-stop-daemon: warning: failed to kill 31008: No such process
Starting razer blackwidow chroma daemon as a daemon`
While it says the daemon is in the work, it also says it cannot find latest instance, and this goes each time I do a force-reload.

A lot might've gone wrong and I know that but having really few experience w/ linux daemons is not helping on my hand, and what I can say for sure is that there's something wrong with the way the daemon is running on my system
(For who cares modded version of Kubuntu by myself)
So sorry if I really made an obvious mistake or anything, but I'm beeing clueless this time beeing.

terrycain, 2016-03-28:
Ubuntu Version?
What device have you got?
Is the kernel driver loaded?

GovanifY, 2016-03-28:
Razer mamba, driver seems to load, latest Kubuntu(updated from dev, should be on wily now. Custom build is this: http://govanify.com/kubuntu/ list of changes http://govanify.com/kubuntu/fulllist.txt

GovanifY, 2016-03-28:
Things were changed from the list btw, such as wine etc but should be unimportant whatsoever

terrycain, 2016-03-28:
The changed shouldn't matter. The reason is the Mamba isn't supported by the daemon yet. I'm currently busy with Uni stuff at the moment but this is on my list of things to do as I too have a Mamba :wink:

GovanifY, 2016-03-28:
it's not really the issue, if I just want to open the GUI, even without any device it just never connects to the daemon, this is the real issue

terrycain, 2016-03-29:
Ahh yeah I also know about that. There is an exit statement in the code which does that, the exit shoudn't be there its was just added until we have an error page saying the daemon could not be contacted. I've also planned a refactor of the UI and accompanying libraries. So that will be fixed in time. Also just to note the app only does the Chroma keyboard at the moment. Once I get time I will add a section for the firefly and mamba

GovanifY, 2016-03-30:
Ok thanks then! Noted for the GUI but just wanted to know the exact reason that wasn't working. I just thougt the project was interesting and wanted to start checking it from time to time.

terrycain, 2016-03-30:
Ahh fair enough. Yeah keep an eye on the repo like after May as I hope to be pushing it forward quickly then when I have time to work on it.

I'm going to mark this as closed for now as the same issue about the daemon and mamba is open elsewhere, and I've fixed the not opening issue so that will be pushed when I come back to working on this properly.

closed 2016-03-30 by terrycain