pez2001/razer_chroma_drivers #104: Getting SyntaxError on chroma_controller:130 for Blackwidow Tournament Chroma

ganiserb, 2016-04-01:
Hello and thank you for doing this!

I'm using Kubuntu 14.04 and managed to install the Ubuntu 14.04 version just fine.
The Desktop entries did not get registered properly but that does not bother me.

When I try to manually run ``python /usr/share/razer_chroma_controller/chroma_controller.py`` to open the GUI it plops the following error:
```
 ~  python /usr/share/razer_chroma_controller/chroma_controller.py 
  File "/usr/share/razer_chroma_controller/chroma_controller.py", line 130
    js_exec.exec()
               ^
SyntaxError: invalid syntax
```

I have no clue about where to start debugging this. Any ideas? I think I'm missing a dependency or maybe it is really a bug.

lah7, 2016-04-01:
The GUI and Tray are written for Python 3. Try again with this command:
`python3 /usr/share/razer_chroma_controller/chroma_controller.py`

ganiserb, 2016-04-01:
Oh! Did not think about that.

I created a virtualenv with Python 3, activated and then got a 
```
(razer) ~  python /usr/share/razer_chroma_controller/chroma_controller.py 
Traceback (most recent call last):
  File "/usr/share/razer_chroma_controller/chroma_controller.py", line 20, in <module>
    import gi
ImportError: No module named 'gi'

```

So I ``pip install gi`` and got
```
(razer) ~  python /usr/share/razer_chroma_controller/chroma_controller.py
Traceback (most recent call last):
  File "/usr/share/razer_chroma_controller/chroma_controller.py", line 20, in <module>
    import gi
  File "/work/gabriel/dev/.virtualenvs/razer/lib/python3.4/site-packages/gi/__init__.py", line 39
    print url
            ^
SyntaxError: Missing parentheses in call to 'print'
```

Is ``gi`` a Python2 package? Or does it have to be installed from somewhere else?

lah7, 2016-04-01:
As you're on Ubuntu, you can install the `python3-gi` package for your system.

    sudo apt-get install python3-gi

ganiserb, 2016-04-01:
Apparently I already have this installed but I keep getting the same output

```
(razer) ~  sudo apt-get install python3-gi                        
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-gi is already the newest version.
The following packages were automatically installed and are no longer required:
  linux-headers-3.16.0-50 linux-headers-3.16.0-50-generic
  linux-image-3.16.0-50-generic linux-image-extra-3.16.0-50-generic
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 155 not upgraded.
(razer) ~  python3 /usr/share/razer_chroma_controller/chroma_controller.py
Traceback (most recent call last):
  File "/usr/share/razer_chroma_controller/chroma_controller.py", line 20, in <module>
    import gi
ImportError: No module named 'gi'
```

terrycain, 2016-04-01:
Run `python3` then run `import gi` if it doesnt work then python3-gi still isnt installed

ganiserb, 2016-04-01:
Definitely not installed then. Will look into that.

Thanks for your help so far!

terrycain, 2016-04-01:
If you made a virtualenv make sure you are not still inside it. Also have you confirmed the daemon is running as that always helps ;)

ganiserb, 2016-04-01:
Allright. I made it work by installing ``sudo apt-get install libwebkitgtk-3.0-dev`` and getting rid of the virtualenv and calling the python3 bin directly like this:

``~  /usr/bin/python3.4 /usr/share/razer_chroma_controller/chroma_controller.py``

Now the problem is that when I create a new custom profile the GUI shows me a Full Razer Blackwidow Chroma instead of the Tournament edition that has no numpad, and also the GUI is for the one that has the big Enter key and not the thin US keyboard Enter key, so I cannot set a color for the ``\`` key.

terrycain, 2016-04-01:
Ahh I didn't even know they made the no-numpad versions. What's the output of `setxkbmap -query`

ganiserb, 2016-04-01:
This is the output
```
rules:      evdev
model:      pc101
layout:     us
variant:    altgr-intl
```
The keyboard in question is this one: http://www.razerzone.com/store/razer-blackwidow-tournament-edition-chroma
I think it's been available only recently.

Jost so you know I could set custom colors to the keyboard, except for the ``\````|`` key between backspace and Enter: 
[image]

terrycain, 2016-04-01:
Numpad is the best, your missing out. Ok I'm busy with Uni at the moment and it will take a while to make a new keyboard with no numkeys (I might just disable them instead). But i've got the information now so I can look into it.

ganiserb, 2016-04-01:
Oh, I just don't like to have the mouse and alpha keys so far apart and I don't care about the macro keys. Anyways... You are awesome and thank you for your help!

Maybe if I have nothing else to do I can help with a PR. Will ping you before I start.

closed 2016-04-01 by ganiserb

terrycain, 2016-04-01:
Cool PR's are always welcome. I'll look into the keyboard issue, as we have a us layout but its not being triggered.