pez2001/razer_chroma_drivers #106: Unable to connect to razer_bcd on fresh install of Ubuntu 15.10 on Razer Blade Stealth

centaur2048, 2016-04-02:
Follow steps for creating and installing an Ubuntu .deb file. Rebooted.

Running `/usr/share/razer_chroma_controller/chroma_controller.py` results in:

```
Loading preferences from "/home/andrew/.config/razer_chroma/preferences.json'...
Successfully loaded preferences.
Failed to connect to the dbus service. Is the razer_bcd service running?
```

Checking the status with `sudo service razer_bcd status` has an outcome of

```
razer_bcd.service - LSB: start/stops razer_bcd
   Loaded: loaded (/etc/init.d/razer_bcd)
   Active: active (exited) since Fri 2016-04-01 22:15:51 EDT; 2min 46s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 857 ExecStart=/etc/init.d/razer_bcd start (code=exited, status=0/SUCCESS)

Apr 01 22:15:51 deathblade systemd[1]: Starting LSB: start/stops razer_bcd...
Apr 01 22:15:51 deathblade razer_bcd[857]: * Starting Razer Keyboard daemon ...d
Apr 01 22:15:51 deathblade razer_bcd[857]: Starting razer blackwidow chroma ...n
Apr 01 22:15:51 deathblade razer_bcd[857]: ...done.
Apr 01 22:15:51 deathblade systemd[1]: Started LSB: start/stops razer_bcd.
Hint: Some lines were ellipsized, use -l to show in full.
```

If I run `sudo service razer_bcd start` there is no output.

Executing `sudo apt-get install -f` results in

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
```

There's barely anything installed on this, as I just got the laptop a few hours ago. Let me know if there's a log file I can get at and I'll grab its contents.

centaur2048, 2016-04-02:
A `service --status-all` shows `[ + ] razer_bcd`

terrycain, 2016-04-02:
`service --status-all` can be useless at times, this one included. The
daemon hasn't been modified for the stealth yet. Can you confirm that it's
bound to the driver

centaur2048, 2016-04-02:
What's the best way to do that?

terrycain, 2016-04-02:
That would mean its bound :wink:, there is some details on the wiki about the driver and binding to it. The issue about the stealth not connecting is also on another thread so closing this one as there is no point in having 2 open. Glad the driver is working though ;)

closed 2016-04-02 by terrycain