pez2001/razer_chroma_drivers #111: Backlight Brightness

songandsilence, 2016-04-09:
Fn + F11/12 used to gradually change the backlight brightness level, by 5%. Now, brightness is changed much more, I think 25%. I miss the slower change.

Adding the extra options (5% increments) to the brightness submenu on the applet works around the bug, but only when using the applet. I'm not sure which lines to change to make the key combo also work like this.

terrycain, 2016-04-09:
What device? What os?
And is is highly unlikely we can change how the fn brightness keys work

lah7, 2016-04-09:
I'm not using the latest daemon/driver here, but AFAIK for the BlackWidow, 5% is default. Until the macro keys are enabled, the brightness changes to 25%.

songandsilence, 2016-04-09:
Sorry, it's a Chroma, running on Ubuntu 15.10.

An older version of the driver would stay at 5%, even with the macro keys enabled.

terrycain, 2016-04-10:
Guys can we confirm this is the case with specific versions of the driver. If it's 5% even with macros on an old driver and 25% now then we can compare differences and see if anything can be done about it

ganiserb, 2016-04-11:
@terrycain I was just going to post a new issue about something related to this:

If I have my keyboard set to full brightness and then I run a script with the following code
```
#!/usr/bin/env python3
# coding=utf-8
import razer.daemon_dbus

daemon = razer.daemon_dbus.DaemonInterface()

daemon.set_brightness(50)  # 50 out of 255
```

The keyboard brightness does not change at all. It stays at 100%, BUT when I click ``Fn + F12`` to increase brightness it seems that it goes to 50 and increases a little.

Could it be related?

terrycain, 2016-04-11:
@ganiserb I think that's a different issue, one I've also never come across. Does it do that if you echo 50 into the driver file? What keyboard are you using and are you using the latests version from the repo.