pez2001/razer_chroma_drivers #114: Razer Abyssus 2014 support

claell, 2016-04-20:
I post this as an issue to express my demand for support of Razer Abyssus 2014. Maybe it can get supported one day.

terrycain, 2016-04-20:
Are you willing to help us get support? As it's hard to code for it without owning one.

claell, 2016-04-20:
Yes, I could help with it, of course.

terrycain, 2016-04-21:
OK, well what I need is packet captures. Read this issue #82 from about commend 17 onwards. There is a link to the wiki page there and some examples of what you should be getting. [Wiki Page](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol)

claell, 2016-04-21:
Alright. I will do my best :)

terrycain, 2016-04-21:
That's all we ask ;). I have decided to go through the wiki page and finally add the images it should of had ages ago.

claell, 2016-04-21:
I guess a virtual box is essential to capture things, right?

terrycain, 2016-04-21:
Yep, or vmware but virtualbox is easier in my opinion

terrycain, 2016-04-21:
And installing windows in a VM

claell, 2016-04-21:
Alright.

terrycain, 2016-04-21:
Its a bit involved but is all that can be done, short of you sending me said device that is lol.

claell, 2016-04-21:
I don't understand the sentence completely :)

terrycain, 2016-04-21:
I'm just saying its a lot of work just to get a mouse working :wink:. The only other thing you could do is mail me your mouse and I do all the hard work.

claell, 2016-04-21:
Ah, alright. But I guess it is rather easy compared with other devices, as it does not have many functions. I am currently downloading a Windows image.

terrycain, 2016-04-21:
Well it should take near no effort once I have some packet captures as my
Mamba Chroma should have all the features and more so shouldn't even have
to code much.

Once you have the windows installed, install synapse. Once synapse is
installed pass the USB device to the VM (there is some discussion in that
bug report I sent you about that). Synapse should recognise the mouse and
you "should" be able to configure it. If you get this far, take screenshots
of all the synapse options, zip them up and send them to me.

As your giving the VM full control over your mouse, the mouse wont work on
linux whilst the VM is on, so plug in a spare mouse and use that whilst
capturing packets.

claell, 2016-04-21:
Alright. Good that I have a spare mouse

claell, 2016-04-21:
Installed synapse now. Will continue later when I have time again.

claell, 2016-04-21:
After installing the extension pack from Virtualbox I managed to pass the mouse through to Windows. It seems it got recognized. Shell I upload the screenshot zip here?

terrycain, 2016-04-21:
yeah here is fine

claell, 2016-04-21:
Ok. It just installed the driver.

claell, 2016-04-21:
Alright. Took screenshots from every page of the driver I could access.
[Razer Abyssus 2014.zip](https://github.com/pez2001/razer_chroma_drivers/files/230370/Razer.Abyssus.2014.zip)

terrycain, 2016-04-21:
Ok whats the output from `lsusb`

terrycain, 2016-04-21:
Ok not being able to speak or read german this is what I think we've got.

* DPI Slider
* Checkbox to toggle X and Y DPI
* Possibly an acceleration slider
* frequency dropdown list in Hz
* Toggle for the light on the mouse

Am I right?

claell, 2016-04-21:
Correct (Checkbox to toggle X and Y DPI is translated: activate sensibility for x- and y- axis, don't know, whether this is the same).

terrycain, 2016-04-21:
OK now you need to run `sudo modprobe usbmon` then load up wireshark as root

also output of `lsusb`

terrycain, 2016-04-21:
Follow the rest of the Wiki to result in some packet captures

claell, 2016-04-21:
relevant output of lsusb: Bus 001 Device 003: ID 1532:0042 Razer USA, Ltd

claell, 2016-04-21:
Started wireshark and selected all usb entries. Applied filter usb.bus_id == 1 && usb.device_address == 3

claell, 2016-04-21:
What do I need to do now? Just send some commands to the mouse from synapse and log with wireshark? Do I need to protocoll which commands I sent at which time?

terrycain, 2016-04-21:
Cool now for each option I listed toggle or slide the slider a few times (preferably make a note of what values you slide/toggled to). A separate capture per option would be good.

claell, 2016-04-21:
Ok.

terrycain, 2016-04-21:
Do one feature at a time, for example the logo on the mouse a packet capture should contain it being toggled on and off like 3 times.

For the sliders slide it from minimum to maximum 3 times.

claell, 2016-04-21:
Will do.

claell, 2016-04-21:
I have collected all data. Just need to find out, how to send them to you. I think I will upload them to a file hoster and post the link here.

terrycain, 2016-04-21:
If you zip them you can just drag the zip into the comment box and GitHub will upload it

claell, 2016-04-21:
Problem is a single file is larger than 15 MB sometimes :)

claell, 2016-04-21:
Complete zip file is around 70MB.

terrycain, 2016-04-21:
Ahh I didnt know if there was a limit. If you have google drive that isnt a bad place to put them

claell, 2016-04-21:
I upload them to jumpshare as I don't want to mess up Google drive. Should be fine as well.

claell, 2016-04-21:
BTW: There is also an option in the software to automatically deactivate lighting when screen turns off. Is this a driver specific or only software feature?

terrycain, 2016-04-21:
That's part of the Synapse software

claell, 2016-04-21:
Ok

claell, 2016-04-21:
Here is the link: http://jmp.sh/bvn13aQ

I did a capture of just ticking the X Y checkbox, probably not needed. Also note that I first slided the Y slider and after that the X slider.

claell, 2016-04-21:
Also with "frequeny checkbox" (filename) I mean "frequency dropdown".

terrycain, 2016-04-21:
Ok I've got the logo and the refresh interval drop down. But the other captures didn't contain anything useful. The filter `(frame.len == 154) && (usb.endpoint_number == 0x00)` also will limit the output alot to exactly whats needed.

Can you redo all of them but the logo. and for the refresh rate ones can you note down what option you selected as I have the refresh rate but I dont know what lines refer to what option.

terrycain, 2016-04-21:
Im specifically after the packets of 154 bytes of length.

claell, 2016-04-21:
In the dropdown menu I started with the lowest rate (I think 125) and went up to the highest.

I will redo all captures with the new filter.

claell, 2016-04-21:
Dropdown List: 
[Framerate 125-500-1000.pcapng.zip](https://github.com/pez2001/razer_chroma_drivers/files/230599/Framerate.125-500-1000.pcapng.zip)

I didn't get data for the sliders with your filter applied.

claell, 2016-04-21:
I also tried usb.bus_id == 1 && usb.device_address == 3 && (usb.endpoint_number == 0x00)
and usb.bus_id == 1 && usb.device_address == 3 && (frame.len == 154) as filter but without any data.

claell, 2016-04-21:
Here is a short sample of me recording a short random change of clicking on the DPI slider. Maybe you can identify some characteristic patterns leading to a good filter.
[DPI random change.pcapng.zip](https://github.com/pez2001/razer_chroma_drivers/files/230602/DPI.random.change.pcapng.zip)

claell, 2016-04-21:
Tomorrow I can continue testing.

terrycain, 2016-04-21:
Nothing in the DPI. Got the Refresh Rate done though. Try the DPI again without any filters, it will collect a mass of data but it might get something. I am starting to think that you mouse might not actually change DPI in hardware but it will take more packet captures to prove that.

claell, 2016-04-22:
I also thought that it might only change it in the software, but I think I noticed a change of behaviour when switching back to the Linux system.

I will collect some data without a filter.

claell, 2016-04-22:
After some testing you might be right that it is not applied in the hardware. It seems to have no effect when changing to Linux. Although I thought that DPI has increased after connecting to synapse for the first time and switching to Linux after that.

claell, 2016-04-22:
Here is a capture without filters. I also closed the synapse software at the end. After closing it the applied settings persist. However when making the mouse available to Linux and then again to the VM the settings seem to be reset until I restart the synapse software.
[DPI changed and closed synapse.pcapng.zip](https://github.com/pez2001/razer_chroma_drivers/files/231256/DPI.changed.and.closed.synapse.pcapng.zip)

terrycain, 2016-04-22:
Nothing again well there was a logo change and refresh rate change but that's it. From what you've said it could quite possibly be that synapse is just changing the windows mouse speed, or doing something clever in software.

The driver is currently undergoing a massive overhaul, ill add this to it at some point soonish.

claell, 2016-04-22:
Strange. I did not change the logo nor the refresh rate. I just changed DPI and closed the software after it.

I also opened the Windows Mouse speed window and it does not appear to be changed by the software. So it is probably some hidden thing, if it is only software specific.

claell, 2016-04-22:
I guess it is pretty clear that the changes are only software changes, right?

claell, 2016-04-22:
So I just think that I noticed a DPI change on Linux: There is a difference when deactivating pass through with synapse open and synapse closed before. On the Linux system is a change in DPI for me, so a hint that there is something stored on the mouse probably.

terrycain, 2016-04-22:
I know synapse sets all settings to the last settings it knows which is probably why it set the refresh rate and the logo. There could be, I've emailed my razer contact to see if they are willing to shed any light on the issue

claell, 2016-04-22:
Ok, this is probably the reason.

Hopefully your contact can help.

[4 more replies]

closed 2016-05-05 by terrycain