pez2001/razer_chroma_drivers #117: [FC23] Unable to compile Driver: make[1]: *** No rule to make target 'modules'.

es02, 2016-05-03:
```
======================
\n::\033[32m Compiling Razer kernel modules\033[0m
========================================
make[1]: *** No rule to make target 'modules'.  Stop.
Makefile:24: recipe for target 'driver_verbose' failed
make: *** [driver_verbose] Error 2
```

As far as I can tell all dependencies are installed, running stock kernel.

I don't get any additional useful output from running the verbose version of the driver compile.

Full command & output: http://pastebin.com/bi1fKxEH

Any ideas?

terrycain, 2016-05-03:
Hmmm, what Linux distribution you using?

terrycain, 2016-05-03:
Full path to where the repository is located as well.

es02, 2016-05-03:
Fedora Core 23 (x64)
I'm compiling from /home/<myusername>/razer

I compiled around 6 months ago under xubuntu and had no issues >.>

terrycain, 2016-05-04:
Hmm, I thought I fixed it on f22 so it _should_ work ha.
Output of `ls -l /lib/modules/$(uname -r)/build/`

es02, 2016-05-04:
$ ls -l /lib/modules/$(uname -r)/build/
total 0

What should be in that dir?

terrycain, 2016-05-05:
Kernel headers. Install the relevant kernel headers for fedora.

This might do it `yum -y install kernel-devel kernel-headers`

es02, 2016-05-05:
Package kernel-devel-4.4.6-300.fc23.x86_64 is already installed, skipping.
Package kernel-headers-4.4.6-300.fc23.x86_64 is already installed, skipping.

Looks like thats installed to /usr/include/linux

Pointing the kernel dir at that path gets me:
========================================
make -C /usr/include/linux SUBDIRS=/home/myusername/razer/driver modules
make[1]: Entering directory '/usr/include/linux'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/include/linux'
Makefile:24: recipe for target 'driver_verbose' failed
make: *** [driver_verbose] Error 2

terrycain, 2016-05-06:
Try the command again without modules like
`make -C /usr/include/linux SUBDIRS=/home/myusername/razer/driver`

Also whats in /usr/src/ ?? As we ultimatly want to get to the kernel sources

terrycain, 2016-05-07:
OK I just installed F23, updated and it works. 

Basically the path `/lib/modules/$(uname -r)/build/` is fine. But when you install `kernel-devel` it will only install kernel sources for the latest kernel installed (which might be one with a higher version than one the system is currently running). 

So reboot, use the latest kernel available, ensure the directory `/lib/modules/$(uname -r)/` contains a symlink `build` which points to a valid target. Also for F23 you need to install `gcc` as clang or whatever is installed won't do it.

es02, 2016-05-08:
/usr/src contains a kernel folder with a bunch of .patch files.

```
\n::\033[32m Compiling Razer kernel modules\033[0m
========================================
make -C /usr/include/linux SUBDIRS=/home/thobson/razer/driver 
make[1]: Entering directory '/usr/include/linux'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/usr/include/linux'
Makefile:24: recipe for target 'driver_verbose' failed
make: *** [driver_verbose] Error 2
```

terrycain, 2016-05-09:
OK lets start from the beginning. Make sure kernel-devel is installed and reboot. Then

What is the output from `uname -a`
What is the output from `ls -l /lib/modules`

es02, 2016-05-13:
As previously noted kernel-devel is installed. I have stock kernel installed which should match the kernel sources installed.
```
[username@hostname razer]$ uname -a
Linux hostname 4.2.3-300.fc23.x86_64 #1 SMP Mon Oct 5 15:42:54 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
[username@hostname razer]$ ls -l /lib/modules
total 4
drwxr-xr-x. 6 root root 4096 Apr 29 13:02 4.2.3-300.fc23.x86_64
```

terrycain, 2016-05-13:
`ls /lib/modules/4.2.3-300.fc23.x86_64`

es02, 2016-05-15:
```
$ ls /lib/modules/4.2.3-300.fc23.x86_64
build   extra   modules.alias      modules.block    modules.builtin.bin  modules.dep.bin  modules.drm          modules.networking  modules.softdep  modules.symbols.bin  System.map  vdso
config  kernel  modules.alias.bin  modules.builtin  modules.dep          modules.devname  modules.modesetting  modules.order       modules.symbols  source               updates     vmlinuz
```

terrycain, 2016-05-16:
`ls -l /lib/modules/4.2.3-300.fc23.x86_64` (i wanted the long listing :wink:)
and
`ls -l /lib/modules/4.2.3-300.fc23.x86_64/build`

es02, 2016-05-16:
Apologies :)

> $ ls -l /lib/modules/4.2.3-300.fc23.x86_64
> total 12292
> lrwxrwxrwx.  1 root root      38 Apr 29 13:02 build -> /usr/src/kernel/4.2.3-300.fc23.x86_64/
> -rw-r--r--.  1 root root  160270 Oct  6  2015 config
> drwxr-xr-x.  3 root root    4096 Apr  8 21:49 extra
> drwxr-xr-x. 12 root root    4096 Oct 30  2015 kernel
> -rw-r--r--.  1 root root  840445 Apr  8 21:49 modules.alias
> -rw-r--r--.  1 root root  822329 Apr  8 21:49 modules.alias.bin
> -rw-r--r--.  1 root root    1778 Oct  6  2015 modules.block
> -rw-r--r--.  1 root root    6986 Oct  6  2015 modules.builtin
> -rw-r--r--.  1 root root    9288 Apr  8 21:49 modules.builtin.bin
> -rw-r--r--.  1 root root  284980 Apr  8 21:49 modules.dep
> -rw-r--r--.  1 root root  406072 Apr  8 21:49 modules.dep.bin
> -rw-r--r--.  1 root root     331 Apr  8 21:49 modules.devname
> -rw-r--r--.  1 root root     147 Oct  6  2015 modules.drm
> -rw-r--r--.  1 root root     110 Oct  6  2015 modules.modesetting
> -rw-r--r--.  1 root root    2562 Oct  6  2015 modules.networking
> -rw-r--r--.  1 root root  111580 Oct  6  2015 modules.order
> -rw-r--r--.  1 root root      55 Apr  8 21:49 modules.softdep
> -rw-r--r--.  1 root root  333616 Apr  8 21:49 modules.symbols
> -rw-r--r--.  1 root root  413586 Apr  8 21:49 modules.symbols.bin
> lrwxrwxrwx.  1 root root       5 Oct  6  2015 source -> build
> -rw-------.  1 root root 3152021 Oct  6  2015 System.map
> drwxr-xr-x.  2 root root    4096 Oct  6  2015 updates
> drwxr-xr-x.  2 root root    4096 Oct 30  2015 vdso
> -rwxr-xr-x.  1 root root 5977368 Oct  6  2015 vmlinuz

> $ ls -l /lib/modules/4.2.3-300.fc23.x86_64/build
> lrwxrwxrwx. 1 root root 38 Apr 29 13:02 /lib/modules/4.2.3-300.fc23.x86_64/build -> /usr/src/kernel/4.2.3-300.fc23.x86_64/

terrycain, 2016-05-17:
Sorry this one meant to end in a slash (as its a symlink)
`ls -l /lib/modules/4.2.3-300.fc23.x86_64/build/`

Also try running the `make driver_verbose` in the repo again.

es02, 2016-05-17:
$ ls -l /lib/modules/4.2.3-300.fc23.x86_64/build/
total 0

terrycain, 2016-05-19:
Ok this is very strange as `/usr/src/kernel/4.2.3-300.fc23.x86_64/` shouldn't be empty. `kernel-devel` should of populated that.

Whats the output of
```
find /usr/src/kernels -maxdepth 2 -name 'Kconfig'

rpm -qa | grep kernel
```

es02, 2016-07-06:
I gave up and moved back to Ubuntu for a variety of reasons.

closed 2016-07-06 by es02