pez2001/razer_chroma_drivers #119: Installation tutorial

claell, 2016-05-04:
On github there is a wrong path in step one of the installation tutorial:

Download the source code:

```
git clone --depth=1 https://github.com/pez2001/razer_blackwidow_chroma_driver.git
cd razer_blackwidow_chroma_drive
```

Also I saw that there is a different git link on http://pez2001.github.io/razer_chroma_drivers/

`git clone --depth=1 https://github.com/pez2001/razer_chroma_drivers.git`

terrycain, 2016-05-04:
Ahh good spot, we had a different repo name a while ago. Shall go and fix now.

closed 2016-05-04 by terrycain

claell, 2016-05-04:
There is still the wrong path (missing s):

`cd razer_blackwidow_chroma_drive`

terrycain, 2016-05-04:
Oh yeah, good shout. Fixed

claell, 2016-05-04:
I meant this section on https://github.com/pez2001/razer_chroma_drivers

```
git clone --depth=1 https://github.com/pez2001/razer_chroma_drivers.git
cd razer_blackwidow_chroma_drive
```

terrycain, 2016-05-05:
There should be fixed, in future I shouldn't rush things :wink:

claell, 2016-05-05:
Corrected but now there is two times cd :smiley: :

`cd cd razer_chroma_drivers`

terrycain, 2016-05-05:
Haha, one day it will be perfect. There fixed :wink: