pez2001/razer_chroma_drivers #120: razer_bcd.service: Main process exited, code=exited, status=1/FAILURE

denysvitali, 2016-05-11:
Hi everyone,
today I received my new Razer BlackWidow X Chroma and after I installed it I looked for a cool controller to use my keyboard on Arch Linux when I found this awesome project.

Excited I tried it, `git clone`d the repo as per instructions, compiled with `sudo make -s all install` and rebooted the pc.
After the reboot I looked at the service (razer_bcd) to see if it was running, but apparently it was in a state of failure:

```
● razer_bcd.service - Razer Keyboard Daemon
   Loaded: loaded (/etc/systemd/system/razer_bcd.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since mer 2016-05-11 20:29:31 CEST; 6min ago
  Process: 1774 ExecStopPost=/usr/share/razer_bcd/systemd_helpers.sh unbind (code=exited, status=0/SUCCESS)
  Process: 1770 ExecStart=/usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid (code=exited, status=0/SUCCESS)
  Process: 1757 ExecStartPre=/usr/share/razer_bcd/systemd_helpers.sh bind (code=exited, status=0/SUCCESS)
 Main PID: 1772 (code=exited, status=1/FAILURE)

mag 11 20:29:31 den-arch systemd[1]: Starting Razer Keyboard Daemon...
mag 11 20:29:31 den-arch razer_bcd[1770]: Starting razer blackwidow chroma daemon as a daemon
mag 11 20:29:31 den-arch systemd[1]: Started Razer Keyboard Daemon.
mag 11 20:29:31 den-arch systemd[1]: razer_bcd.service: Main process exited, code=exited, status=1/FAILURE
mag 11 20:29:31 den-arch systemd[1]: razer_bcd.service: Unit entered failed state.
mag 11 20:29:31 den-arch systemd[1]: razer_bcd.service: Failed with result 'exit-code'.
```
My Razer Keyboard on `lsusb`:
```
Bus 006 Device 002: ID 1532:0216 Razer USA, Ltd 
```
I also had to add the following string to `/etc/udev/rules.d/95-razerkbd.rules` because my keyboard has a different Vendor ID:Device ID (and then I had to `udevadm trigger` it or reboot the system):
```
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0216", ENV{is_razer_kbd}="yes"
```

```
dvitali / cat /run/udev/data/+hid:0003:1532:0216.0001 
I:92659260
E:is_razer_kbd=yes
dvitali / cat /run/udev/data/+hid:0003:1532:0216.0002
I:92659334
E:is_razer_kbd=yes
dvitali / cat /run/udev/data/+hid:0003:1532:0216.0003
I:92659174
E:is_razer_kbd=yes
```

strace result of `sudo strace razer_bcd`:
```
execve("/usr/bin/razer_bcd", ["razer_bcd"], [/* 22 vars */]) = 0
brk(NULL)                               = 0x557a29f32000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=322753, ...}) = 0
mmap(NULL, 322753, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fa277bdf000
close(3)                                = 0
open("/usr/lib/tls/x86_64/librazer_chroma.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/tls/x86_64", 0x7ffeadfa33c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/tls/librazer_chroma.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/tls", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
open("/usr/lib/x86_64/librazer_chroma.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/lib/x86_64", 0x7ffeadfa33c0) = -1 ENOENT (No such file or directory)
open("/usr/lib/librazer_chroma.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\340K\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=68288, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa277c35000
mmap(NULL, 2163952, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa2777f7000
mprotect(0x7fa277807000, 2093056, PROT_NONE) = 0
mmap(0x7fa277a06000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xf000) = 0x7fa277a06000
close(3)                                = 0
open("/usr/lib/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0pU\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1063296, ...}) = 0
mmap(NULL, 3158248, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fa2774ef000
[... 109 lines cut ...]
mprotect(0x7fa2772dd000, 4096, PROT_READ) = 0
mprotect(0x7fa2774e9000, 4096, PROT_READ) = 0
mprotect(0x7fa277a06000, 4096, PROT_READ) = 0
mprotect(0x557a29eec000, 4096, PROT_READ) = 0
mprotect(0x7fa277c32000, 4096, PROT_READ) = 0
munmap(0x7fa277bdf000, 322753)          = 0
set_tid_address(0x7fa277b55a50)         = 1998
set_robust_list(0x7fa277b55a60, 24)     = 0
rt_sigaction(SIGRTMIN, {0x7fa276accb90, [], SA_RESTORER|SA_SIGINFO, 0x7fa276ad7ef0}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {0x7fa276accc20, [], SA_RESTORER|SA_RESTART|SA_SIGINFO, 0x7fa276ad7ef0}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
getrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
brk(NULL)                               = 0x557a29f32000
brk(0x557a29f53000)                     = 0x557a29f53000
fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
write(1, "Starting razer blackwidow chroma"..., 52Starting razer blackwidow chroma daemon as a daemon
) = 52
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fa277b55a50) = 1999
exit_group(0)                           = ?
+++ exited with 0 +++
```

Thank you for the awesome project and for your future support,

Denys

terrycain, 2016-05-11:
In future don't paste large comments, use pastebin.

The kernel driver will need modifying to support your device, as will the daemon. I'm finishing off at Uni at the moment hence I'm rather busy.

I'm planning on replacing the daemon eventually with a python one which should make adding new devices easy. 

Good news is that theoretically the driver supports all the features of your keyboard so should take little effort modify. After the end of the month i'll have more than enough time to fix all this project.

terrycain, 2016-05-11:
Also what keyboard layout have you got? Picture would be useful.

denysvitali, 2016-05-11:
@terrycain Thanks for the answer,
I've got a German keyboard layout, and [this](https://goo.gl/photos/RBNhBXVenpFbQWjG8) is the keyboard picture

terrycain, 2016-05-11:
Nice the controller app will need editing to support the layout but it would work but just shows UK layout.

Yeah if you wait till the end of the month i'll crack on with support

denysvitali, 2016-05-11:
No problem, thank you very much!

denysvitali, 2016-06-13:
Any update on this?

terrycain, 2016-06-14:
Not yet sorry, have been packing as am moving house, will get on it once ive moved.

denysvitali, 2016-06-14:
No problem :+1:

denysvitali, 2016-08-17:
Maybe this should be tagged "enhancement" too

terrycain, 2016-08-18:
@denysvitali Sorry mate I completely forgot about this. Basically I've moved to doing all the development on my own fork. I've completely rewriten the daemon so it works with multiple devices properly. Once @lah7 updates his app we can then work again on the German layout.