pez2001/razer_chroma_drivers #121: Ubuntu 16.04 and Razer blackwidow chroma issue

nizamsp, 2016-05-12:
The keyboard configuration app is not working.
When I did the status command, I get the following error.

Any help is much appreciated!

**nizam@wimt-nizam:~$ sudo service razer_bcd status**

● razer_bcd.service - LSB: start/stops razer_bcd
   Loaded: loaded (/etc/init.d/razer_bcd; bad; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2016-05-12 23:16:26 IST; 2min 2s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 1734 ExecStart=/etc/init.d/razer_bcd start (code=exited, status=2)

systemd[1]: Starting LSB: start/stops razer_bcd...
razer_bcd[1734]:  * Starting Razer Keyboard daemon razer_bcd
razer_bcd[1734]: Unbinding device (0003:1532:0203.0003) from hid-generic
razer_bcd[1734]: Binding device (0003:1532:0203.0003) to razerkbd
razer_bcd[1734]: /etc/init.d/razer_bcd: 22: /etc/init.d/razer_bcd: cannot create /sys/bus/hid/drivers/razerkbd/bind: Directory nonexistent
systemd[1]: razer_bcd.service: Control process exited, code=exited status=2
systemd[1]: Failed to start LSB: start/stops razer_bcd.
systemd[1]: razer_bcd.service: Unit entered failed state. 
systemd[1]: razer_bcd.service: Failed with result 'exit-code'.

terrycain, 2016-05-13:
Is the razer kernel driver loaded? Output of `lsmod` (put it on pastebin.com and link to it)

nizamsp, 2016-05-13:
http://pastebin.com/ETyG6xw5

Thanks for your time.

terrycain, 2016-05-13:
The razerkbd driver is not loaded. How did you install the project?

Try running 
```
sudo modprobe razerkbd
sudo modprobe razerfirefly
sudo modprobe razermouse
```
That should load the modules and then they should appear in the output of `lsmod`

If that works then this _should_ get the system to load the modules on boot
```
sudo -i
mkdir -p /etc/modules-load.d/
cat <<EOF > /etc/modules-load.d/razer.conf
# Load Razer Drivers
razerkbd
razermouse
razerfirefly
EOF
```

nizamsp, 2016-05-13:
nizam@wimt-nizam:~$ sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Required key not available

I get the above error.

terrycain, 2016-05-13:
Hmm, that is very weird. Never seen that error before. How did you go about installing the driver?

Run `readelf -S path_to_razerkbd.ko` and put that in pastebin.

nizamsp, 2016-05-13:
Thanks a lot for the prompt response. Even official Razer wouldn't give this much support :)

There are two files actually. Attached the pastebin of both

/lib/modules/4.4.0-22-generic/kernel/drivers/usb/misc/razerkbd.ko
http://pastebin.com/BgaSVPJZ

/var/lib/dkms/razer_chroma_driver/1.0.0/4.4.0-22-generic/x86_64/module/razerkbd.ko
http://pastebin.com/Vtnbhpsy

terrycain, 2016-05-13:
Hmm there should only be one instance, did you install any .deb's to make that `/var/lib/dkms` one appear?

Easiest way would be to remove all traces of razer from your system and then install again using `./package_for_debian.sh`

nizamsp, 2016-05-13:
I removed all the traces of razer and then installed again using the 7 step process given in home page.

I used ./package_for_ubuntu.sh instead of debian.sh 

http://pastebin.com/NwXpnbxf
http://pastebin.com/FEMRXVV9

Same error. 

Now also I got two instances of that file.

/lib/modules/4.4.0-22-generic/updates/dkms/razerkbd.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.4.0-22-generic/x86_64/module/razerkbd.ko

While installing the .deb, I was asked to enter the password to disable secure boot.
Is that normal?

terrycain, 2016-05-13:
Ok we need to get the driver signed to keep secure boot, that requires far too much work for now. Ubuntu 16.04 has kernel module signing on default.

That should be alright. Is secureboot off now? as it the `sudo modprobe razerkbd` command _should_ work

You might have to try these commands
```
sudo apt install mokutil
sudo sudo mokutil --disable-validation
```

nizamsp, 2016-05-13:
I tried the commands you gave. No luck. Sorry.

```
nizam@wimt-nizam:~$ sudo apt install mokutil
Reading package lists... Done
Building dependency tree       
Reading state information... Done
mokutil is already the newest version (0.3.0-0ubuntu3).
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
nizam@wimt-nizam:~$ sudo sudo mokutil --disable-validation
password length: 8~16
input password: 
input password again: 
nizam@wimt-nizam:~$ sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Required key not available

```

nizamsp, 2016-05-13:
Hurray!!

Thank you so much! It worked :)

Actually your software is very clean.
It asked me for 
1. Continue boot
2. Change Secure Boot State.

My bad. I was clicking on the Continue boot. That was the issue.
I will send a donation for the continued support :)

terrycain, 2016-05-13:
Wicked you can either contribute to pez or my paypal either is good :wink: 

Yeah it's possible to fix the secure boot thing. Will add that to the todo list when things actually work how I want them too ;)

nizamsp, 2016-05-13:
Please tell me your paypal account or send it to nizam.sp@gmail.com

terrycain, 2016-05-13:
terry@terrys-home.co.uk, should really add it to the main page ;)

terrycain, 2016-05-13:
Cheers for that :+1:

closed 2016-05-13 by terrycain