pez2001/razer_chroma_drivers #122: Macro Keys enabled on startup

Deus-Rex, 2016-05-14:
I use the macros for daily desktop use, so would prefer them being active on startup.
Is there a script I could use or maybe a setting that I have missed?

terrycain, 2016-05-15:
I think this _might_ have been asked before. I think @lah7 knows something, if not I can whip you up a script.

lah7, 2016-05-15:
Similar questions have been asked before, but got buried somewhere. There's two ways:

* Setting the option in the GUI Chroma App's Preferences, this assumes the Tray Applet is loading at start-up.
* Write a Python script using the libraries, like so:

```
#!/usr/bin/env python3
import razer.daemon_dbus
daemon = razer.daemon_dbus.DaemonInterface()
daemon.marco_keys(True)
```

Original, but similar comments:
* https://github.com/pez2001/razer_chroma_drivers/issues/95#issuecomment-190609047
* https://github.com/pez2001/razer_chroma_drivers/issues/113#issuecomment-212817603

Deus-Rex, 2016-05-15:
Sorry, I had a look but seems I wasn't thorough enough.

To be honest, and this is embarrassing, I only just realized that the preferences in the tray applet could be scrolled down to reveal more options. Never noticed the slim scrollbar that appears on hover. 

So yup, found the setting. Thanks!

I tried creating a Python script a few weeks ago, but couldn't get it to work. I think I was typing 'macro_keys' instead. I should have looked closer at that, too.

Thanks, again!

closed 2016-05-15 by Deus-Rex