pez2001/razer_chroma_drivers #126: Razer Deathadder support

DanielVoogsgerd, 2016-05-22:
I noticed that the Razer Deathadder mouse isn't supported. The mouse is quite popular and I have one if you'd like to try some things.

terrycain, 2016-05-22:
Sure, same info as last time :wink:

DanielVoogsgerd, 2016-05-22:
It's a 2012 model again

terrycain, 2016-05-23:
Ok a list of features would be useful. 

Output of `lsusb -v -d 1532:0016` in pastebin would also help

DanielVoogsgerd, 2016-05-23:
https://gist.github.com/DanielVoogsgerd/0912506f1d5fc8dc27ebaedc99d243cc.
Not many features, but here we go:
- DPI switch button
- Back button
- Forward button
- Pulsating Logo
- Pulsating scrollwheel

terrycain, 2016-05-23:
Back and forward buttons should work as normal.
Logo and scrollwheel should be dooable.
Does it have adjustable DPI?
Is it wireless?

DanielVoogsgerd, 2016-05-23:
@terrycain No it's not wireless, but it does have adjustable DPI

terrycain, 2016-05-24:
Ok we should be able to support most of it (there are some DPI settings I need to analyse further like setting up DPI stages but can definitly set current DPI). Do you know if the logo and scroll wheel can be modified seperatly?

DanielVoogsgerd, 2016-05-24:
Not that I'm aware of

terrycain, 2016-05-24:
Ok, cool though we might be able to control them separately :wink:. I'll get to work on this later in the week, have one last bit of work due for Uni tomorrow.

nonverbally, 2016-05-26:
DeathAdder Chroma support would also be great. It is listed as 1532:0043. Here is the lsusb output:

http://pastebin.com/DHV22GkW

This is the product listing: http://www.razerzone.com/store/razer-deathadder-chroma

The features are fairly basic. It is wired and has back and forward buttons and lighting of the logo and mouse wheel.

terrycain, 2016-05-26:
@nonverbally would you mind creating a new issue, as then I can keep the death adder and death adder chroma separate as will be a fair bit of differences.

terrycain, 2016-05-27:
@DanielVoogsgerd I'm gonna work on the tartarus, then the deathadder choma (as nonverbally can capture packets) then I should be able to convert the deathadder chroma to a normal deathadder driver

presianbg, 2016-08-26:
I poses both - Chroma and the normal death adder. Just say if you need some help with dumps or other things.

terrycain, 2016-08-26:
Yeah dumps please :smile: If you could create a chroma deathadder issue [here](https://github.com/terrycain/razer-drivers) and attach some dumps (preferably with a readme saying what was done in each dump) that would be great. If you could start with the chroma then the normal should be the same without lighting. Look [here](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol) for info about getting dumps etc... And if I could get screenshots of each page of razer synapse that'll help too as then It'll be easier to see what features the mice have and what options etc...

presianbg, 2016-08-26:
Ok :) I will do that when have the time. 

Meanwhile... are you aware of this [project](https://github.com/mbuesch/razer) ? I'm using this tool and Michael has reverse engineered most of the drivers for the Razer mouse.

terrycain, 2016-08-26:
Yeah the difference with that project I believe is that they dont use kernel drivers, they use libusb which is fine I just found during my initial testing kernel drivers work a little nicer + having a kernel module is cooler yeah.

mirh, 2016-08-26:
Speaking of kernel drivers, probably dumb question: why isn't this being mainlined?

terrycain, 2016-08-26:
Short story. Wanted to make big code changes, and use some features of github that you need to be owner for

mirh, 2016-08-26:
Well, assuming big code changes are senseful, I don't see why they shouldn't be accepted in the end. 
And.. development could still proceed here then, you should only need to push stable releases to linux-next every now and then.

claell, 2016-08-26:
@terrycain I think that you explained why you continued development of this on your own fork (https://github.com/terrycain/razer-drivers).

As far as I understand @mirh wanted to know, why the drivers aren't included in the Kernel.

terrycain, 2016-08-26:
As getting drives in the kernel is a mission in its self, especially getting it into the older kernels. The main reason is I'd rather focus on getting it working with as many devices as possible and getting the supporting software working well rather than bothering with linus' whineing when the driver doesnt look as pretty as he wants it.

mirh, 2016-08-26:
I don't see why even only getting in newest kernels couldn't still be seen as an accomplishment. 
And if the bar is set high, then that might not be the first objective, but still, looking into it.

terrycain, 2016-08-26:
Yeah I do eventually want it to be in a mainline kernel but I have other things I want to get working first. And if I can get all of the tooling in place working nicely then when the driver does make it into the kernel then it will work out of the box so to speak.

m4ng0squ4sh, 2016-08-26:
Hi,
you are always welcome to join the [OpenRazer](https://github.com/openrazer/openrazer-drivers) project, joining our efforts into a single project :) The kernel code has reached its first stable and clean release and the patches are already send to the linux mailing lists. Hope this is the beginning of a great Razer expierence under Linux.