pez2001/razer_chroma_drivers #128: Is the Razer Blackwidow Ultimate 2014 supported?

pedroadame, 2016-05-24:
As it.

terrycain, 2016-05-24:
Read through #89 and provide me with the details I asked for in that issue.

Short answer, not yet. Long answer, if you help then probably soon.

pedroadame, 2016-05-24:
Ok!
`lsusb` outputs:
`Bus 003 Device 006: ID 1532:011a Razer USA, Ltd`

`lsusb -d 1532:011a -vvv` outputs:
[lsusb-verbose-output.txt](https://github.com/pez2001/razer_chroma_drivers/files/279957/lsusb-verbose-output.txt)

Razer Synapse allows:
* Adjust the brightness of the Razer BW Ultimate 2014
* Turn on / off game mode
* Record and manage macros
* Create application profiles (including macros and brightness config)
* Create non-application profiles (same includes)

Non pulsating/animated lightining. But I can't check Synapse currently on so i'll post an updated comment later.

terrycain, 2016-05-24:
My 2013 can do pulsating so I would assume the 2014 can but know knows. I think the chroma was the first animated one but the 2016 can do that now as well (though only in green).

terrycain, 2016-05-24:
Just looking at it. 0x011A is the same as my 2013, In theory you should be able to bind it to the current driver. Read the rest of #89 and move onto the binding and testing part as it might already be supported. On that page there is a link to the wiki explaining driver binding etc...

closed 2016-06-17 by terrycain

Freso, 2016-07-30:
I'm trying to do the binding for a BW Ultimate 2014, and I don't see a mouse interface for it:
```
freso@din ~> ls -ld /sys/bus/hid/devices/*1532*
lrwxrwxrwx 1 root root 0 Jul 30 22:49 /sys/bus/hid/devices/0003:1532:011A.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1532:011A.0003/
lrwxrwxrwx 1 root root 0 Jul 30 22:49 /sys/bus/hid/devices/0003:1532:011A.0004 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:1532:011A.0004/
freso@din ~> udevadm info -a /sys/bus/hid/devices/0003:1532:011A.0003|grep bInterfaceNumber
    ATTRS{bInterfaceNumber}=="00"
freso@din ~> udevadm info -a /sys/bus/hid/devices/0003:1532:011A.0004|grep bInterfaceNumber
    ATTRS{bInterfaceNumber}=="01"
```
It does show up in `lsusb` though:
```
freso@din ~> sudo lsusb -v -d 1532:011a|grep bInterfaceProtocol
      bInterfaceProtocol      1 Keyboard
      bInterfaceProtocol      1 Keyboard
      bInterfaceProtocol      2 Mouse
```

Any clues?

[Dump of `lsusb -vvv -d 1532:011a`](https://github.com/pez2001/razer_chroma_drivers/files/392090/RazerBlackwidowUltimate2014-lsusb-dump.txt)

terrycain, 2016-07-31:
Unplug keyboard, then `tail -f /var/log/dmesg` then plug in keyboard. You
should see some rubbish about usb plugged in etc... Put that in a pastebin