pez2001/razer_chroma_drivers #129: Driver seems not loaded on Ubuntu 16.04

claell, 2016-05-24:
I just wanted to use the driver with my Abyssus, but it does not seem to be loaded correctly. At least the polychromatic controller does not start. https://github.com/lah7/polychromatic/issues/9

claell, 2016-05-24:
Output of `clel@clel-ubuntu-ThinkPad-T400:~$ sudo service razer_bcd status`:
```
● razer_bcd.service - Razer Keyboard Daemon
   Loaded: loaded (/lib/systemd/system/razer_bcd.service; enabled; vendor preset: enabled)
   Active: failed (Result: resources) since Di 2016-05-24 19:37:39 CEST; 4min 13s ago
  Process: 867 ExecStart=/usr/sbin/razer_bcd --pid-file /var/run/razer_bcd.pid (code=exited, status=0/SUCCESS)
  Process: 839 ExecStartPre=/usr/share/razer_bcd/systemd_helpers.sh bind (code=exited, status=0/SUCCESS)

Mai 24 19:37:39 clel-ubuntu-ThinkPad-T400 systemd[1]: Starting Razer Keyboard Daemon...
Mai 24 19:37:39 clel-ubuntu-ThinkPad-T400 razer_bcd[867]: Starting razer blackwidow chroma daemon as a daemon
Mai 24 19:37:39 clel-ubuntu-ThinkPad-T400 systemd[1]: razer_bcd.service: PID 874 read from file /var/run/razer_bcd.pid d
Mai 24 19:37:39 clel-ubuntu-ThinkPad-T400 systemd[1]: Failed to start Razer Keyboard Daemon.
Mai 24 19:37:39 clel-ubuntu-ThinkPad-T400 systemd[1]: razer_bcd.service: Unit entered failed state.
Mai 24 19:37:39 clel-ubuntu-ThinkPad-T400 systemd[1]: razer_bcd.service: Failed with result 'resources'.
```

terrycain, 2016-05-24:
Yeah I'm working on a replacement daemon but its not ready yet (should be soon). Adding the mouse to the current driver would take more effort than its worth.

claell, 2016-05-24:
Ah, alright. This is probably the reason why I did not install it earlier. Today I saw you merged 20 days ago and wondered, why I haven't installed it earlier.

closed 2016-05-24 by claell