pez2001/razer_chroma_drivers #131: Razer Tartarus Chroma Support

nonverbally, 2016-05-26:
Razer Tartarus Chroma support would be a great addition. 
ID 1532:0208

lsusb output:
http://pastebin.com/eEieqsLQ

Razer product page:
http://www.razerzone.com/gaming-keyboards-keypads/razer-tartarus-chroma

It features rgb lighting, a single color for the whole keypad and many programmable keys (similar to M keys on the black widow) that default to WASD configuration unless programmed.   There is an existing project that has some support for the Tartarus in Linux: https://bitbucket.org/conzar/keyboarding-master

terrycain, 2016-05-26:
@nonverbally does it emit any keycodes when the keys are pressed?

If you don't know, run `xev` and hit some keys. If stuff appears then create me a mapping between each key/button and the keycode `xev` reports.

nonverbally, 2016-05-26:
They do emit keycodes. Here are the mappings to the keys.

Key	keycode
1	23
2	24
3	25
4	26
5	27
6	66
7	38
8	39
9	40
10	41
11	50
12	52
13	58
14	54
15	55
Mode Switch	64
Up (hat switch)	111
Down (hat switch)		116
Left (hat switch)	113
Right (hat switch)	114
Bottom Thumb Button (below hat switch)		65

terrycain, 2016-05-26:
Ok we'll need to do some reverse engineering to get things perfect as it presents itself as a keyboard but is inherently different than one. Look [here](https://github.com/pez2001/razer_chroma_drivers/wiki/Reverse-Engineering-USB-Protocol), are you able to help reverse engineering the protocol?

nonverbally, 2016-05-26:
Sure. I'll review the procedure and setup the VM/wireshark environment.

nonverbally, 2016-05-27:
Here are three pcaps of the three different lighting modes, Static (Green), Breathing (Green, Red), and Spectrum Cycling (Green). I followed the procedure and alternated between none and each lighting mode. There is a separate pcap file for each lighting mode in archive.
[RazerTartarus-lighting-pcaps.tar.gz](https://github.com/pez2001/razer_chroma_drivers/files/285572/RazerTartarus-lighting-pcaps.tar.gz)

terrycain, 2016-05-27:
Can I get screenshots of synapse too :wink:

terrycain, 2016-05-27:
pcaps are looking good. Me being lazy im going to make a script to extract the data I want and format it nicely ;)

Does it only do static, breathing and spectrum? Can it do custom colours assigned to different keys?

On the right on the image there is 3 LEDs yellow, green, blue. What do they do (im assuming profiles). And can they change colour too?

terrycain, 2016-05-27:
Ok I have a driver for you to test, familiarise youself with [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver) page.

Checkout my [fork](https://github.com/terrycain/razer_drivers/) and your after the `RazerTartarusChroma_0x0208` branch, [this](https://github.com/terrycain/razer_drivers/tree/RazerTartarusChroma_0x0208) should link directly.

1. Make the driver with `make driver_verbose`
1. Load the driver with `sudo insmod ./driver/razerkbd.ko`
1. Goto the wiki page, start at the Binding Driver section [here](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver#binding-the-chroma-to-the-keyboard-driver).
1. Work your way down to the Using the driver section, and then paste the output of the `ls -l` when your inside the device driver thing. (Aka I want to see files like `device_type`, `get_serial` etc...)

Make sense?

nonverbally, 2016-05-31:
I've attached screenshots of the synapse interface. There are only those three lighting choices. The lights on the side only display the current selected keymap. You can toggle through different maps and the light will indicate which (1 through 8) by lighting combinations of the three colors. You can see this in the screenshot. Otherwise, only one color of the keys themselves can by dynamically changed. I will test out your new driver and report back.

[image]
[image]
[image]

terrycain, 2016-06-18:
Have you had a chance to test the driver I made? (Re the comment above)

Denlior, 2016-08-18:
I've tried to test the driver you made for Razer Tartarus Chroma.
Unfortunately I failed - I can't bind it to the driver.
> bash: echo: write error: No such device
(I still can bind it back to hid-generic)

terrycain, 2016-08-18:
Did you get the one from the correct branch? As it "should" work. What did you run to bind it?

Denlior, 2016-08-18:
Cloned and made the driver from your RazerTartarusChroma_0x0208 branch. 
I failed at `# echo -n "0003:1532:0208.0018" > /sys/bus/hid/drivers/razerkbd/bind` step.
Since then I purged all versions of Chroma Linux Drivers to avoid confusion and redownloaded your branch again. This time all worked as supposed to.
ls -l output I get:
`
total 0
-r--r--r-- 1 root root 4096 авг 19 01:32 country
-r--r--r-- 1 root root 4096 авг 19 01:32 device_type
lrwxrwxrwx 1 root root    0 авг 19 01:32 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r--r-- 1 root root 4096 авг 19 01:32 get_firmware_version
-r--r--r-- 1 root root 4096 авг 19 01:32 get_serial
drwxr-xr-x 3 root root    0 авг 19 01:32 hidraw
drwxr-xr-x 3 root root    0 авг 19 01:32 input
-r--r--r-- 1 root root 4096 авг 19 01:32 modalias
--w--w---- 1 root root 4096 авг 19 01:32 mode_breath
--w--w---- 1 root root 4096 авг 19 01:32 mode_spectrum
--w--w---- 1 root root 4096 авг 19 01:32 mode_static
drwxr-xr-x 2 root root    0 авг 19 01:32 power
-r--r--r-- 1 root root 4096 авг 19 01:32 report_descriptor
--w--w---- 1 root root 4096 авг 19 01:32 reset
-rw-rw---- 1 root root 4096 авг 19 01:32 set_brightness
lrwxrwxrwx 1 root root    0 авг 19 01:32 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root root 4096 авг 19 01:32 test
-rw-r--r-- 1 root root 4096 авг 19 01:32 uevent
`

terrycain, 2016-08-19:
Now thats what I want to see :).
At work at the moment but ill send some commands to test when I get home

terrycain, 2016-08-19:
**Do this as root**

Ok from memory this should set it to red
```
echo -ne "\xFF\x00\x00" > mode_static
```
--------
This to green
```
echo -ne "\x00\xFF\x00" > mode_static
```
------------
This should enable spectrum mode
```
echo -n "1" > mode_spectrum
```
-------------
Breathing effect
```
echo -n "1" > mode_breath                              # Random Mode
echo -n -e "\xFF\x00\x00" > mode_breath                # Single mode (red)
echo -n -e "\xFF\x00\x00\x00\x00\xFF" > mode_breath    # Dual mode (red & blue)
```
------
Setting brightness to 50% (100% is 255)
```
echo -n "128" > set_brightness
```

Should be able to `cat set_brightness` to get current brightness

Also these should output stuff. (get_serial might take a few tries)
```
cat device_type
cat get_firmware_version
cat get_serial
```

If I remember correctly, theres more I can add to the driver I think but we'll wait till these work first.

terrycain, 2016-08-19:
Only just noticed your not the original person I was talking to. Shows how much attention I pay. Also dont worry once we get all the features were after, theres deb files on the horizon, (and arch ones too I believe) and probably RPM's after that.

GabMus, 2016-08-19:
Actually Arch ones are already available. The beauty of Arch is that you
can just make a packaging script (PKGBUILD) and that's all a user needs. No
need to distribute binaries :)

Denlior, 2016-08-19:
Yep, I figured colour mode and brightness from driver wiki and /sys/bus/hid/drivers/razerkbd/ contents already. Everything present works fine (commands you posted other RGB colors)
cat device_type gets `Razer Tartarus Chroma`
cat get_firmware_version gets `v1.0`
cat get_serial gets `PM1609151701453`
More functionality would be great, I'm ready to test or help any other way I can.
Personally I would like to be able to have keys producing other keycodes, not overlapping with normal keyboard.

terrycain, 2016-08-19:
@Denlior Yeah about the keys producing keycodes, in the new daemon I have an ingenious solution for that :) :) :). But first ill go back through the packet captures and decode some other stuff (all my work for this device is backed up to my old pc and cba to turn it on to get it :wink:). What functionality is missing in your eyes, (more the h/w functionality not software bits).

Denlior, 2016-08-19:
Well, all keys are producing keycodes and backlight works ok. Backlight keeps it's pattern, brightness and colour on reboot, so writing to internal device memory is good too.
As far as I know, only hardware bit "not working" is three led indicator. All else is software functionality.

terrycain, 2016-08-20:
I thought as much. 

Ok Have pushed a new driver to the `RazerTartarusChroma_0x0208` branch. It "should" contain some new endpoints called something like `profile_led_red`, `profile_led_green` and `profile_led_blue`

You should be able to write from them with
```
echo -n "1" > profile_led_red
echo -n "1" > profile_led_green
echo -n "0" > profile_led_red
```
And read if they are set by `cat`ing them.

terrycain, 2016-08-20:
Once that works, ill merge it with the main code and have a wonderful conflict with the driver as I have a feeling its changed a fair bit since I started this driver. Then well I guess you can test out the new daemon :smile:.

Denlior, 2016-08-20:
Well, it works, but with some "issues".
Distal led answers to profile_led_blue and is greenish yellow in colour actually.
Middle one answers to profile_led_red and is green.
Proximal one is profile_led_green and it's indigo blue.
cat reads their state fine.

terrycain, 2016-08-21:
Ok, yeah I know red is really a yellow colour, just named it so it can be red,green,blue (as its named that in the spec I have). 

Have pushed a new driver, can you confirm the files map to the correct LEDs this time.

Denlior, 2016-08-21:
Yes, now blue is blue, green is green and rad is yellow.

terrycain, 2016-08-21:
:smile: I deem this issue resolved :). Can now add that to the main codebase :). Keep an eye on my fork as soon we'll have a version 1 release. I'll need some help with you testing the dameon with the tartarus. If you want make an issue on my fork for daemon support for tartarus (and copy the contents of the 3rd post in this issue to the new issue :wink:).

closed 2016-08-21 by terrycain